arendst/Tasmota #1328: Sonoff Basic max amount of WS2812B leds without external power supply?

Looooopy, 2017-12-09:
Hi

Just wanted to know if someone know or have tried how many leds the Sonoff Basic can supply power to without an external power supply?

Genosse274, 2017-12-09:
The AMS1117 voltage regulator has a minimum current limit of 900mA. There are different informations about the current draw of the esp. The itead wiki says 215mA, but people have measured spikes of almost 400mA. Also powering from usb (usually 500mA) seems sometimes a problem. Considering the current draw of other components, there is popaply **not** much left for stable operation.
Also the minimal voltage rating of ws2812B is 3.5V. That could be a problem.

Looooopy, 2017-12-09:
I am running 30 leds right now if all is set to emit White it should be 30 x  30 ma = 900mA which should  not work if i did not missunderstand you or whould it stress the unit to fail faster?

I can't see any problems now but i was thinking and try to doubling the amount.

Genosse274, 2017-12-09:
900mA is the minimum current limit of the voltage regulator. The maximum current limit is 1.5A ([Datasheet](http://www.advanced-monolithic.com/pdf/ds1117.pdf)). It can work, but it's not designed for that. Maybe you hit the right spot of possible available current for your LEDs. A failure of the voltage regulator causes a voltage drop, but maybe the capacitor has enough current stored. How long do you run it this way? I personal had problems operating the esp on a breadboard because of voltage drops, causing problems like freezing.

Looooopy, 2017-12-10:
1 hour on the testbench... so i need to stress it more i suppose before i celebrate 😁

xeroiv, 2017-12-29:
@Looooopy How do you have your LED's hooked up to the sonoff?  Do you have a diagram you could share?

closed 2018-05-12 by stale[bot]