arendst/Tasmota #1331: one of my Sonoff T1 2Ch problem

ahzazou, 2017-12-09:
When i restart the sonoff, the MQTT connection is immediately ok.

But from time to time i often see these messages and lights turns off :

MQTT: Attempting connection...
MQTT: CONNECT FAILED, rc -2. Retry in 10 seconds

i have other Sonoff T1 which was the first versions working perfectly but this one is the updated hardware with ROHS starndard i am facing

ahzazou, 2017-12-11:
23:33:39 MQT: tele/sonoff102/STATE = {"Time":"2017-12-11T23:33:39", "Uptime":7, "Vcc":3.208, "POWER1":"ON", "POWER2":"ON", "POWER3":"OFF", "Wifi":{"AP":1, "SSId":"ZOKA", "RSSI":80, "APMac":"34:CE:00:4A:7A:E8"}}
23:37:57 MQT: Attempting connection...
23:37:58 MQT: Connect failed to 192.168.8.146:1883, rc -2. Retry in 10 sec
23:38:09 MQT: Attempting connection...
23:38:10 MQT: Connected
23:38:12 MQT: tele/sonoff102/LWT = Online (retained)
23:38:12 MQT: cmnd/sonoff102/POWER = 
23:38:12 MQT: stat/sonoff102/RESULT = {"POWER1":"OFF"}
23:38:12 MQT: stat/sonoff102/POWER1 = OFF (retained)
23:38:12 MQT: stat/sonoff102/RESULT = {"POWER2":"OFF"}
23:38:12 MQT: stat/sonoff102/POWER2 = OFF (retained)
23:38:13 MQT: stat/sonoff102/RESULT = {"POWER1":"OFF"}
23:38:13 MQT: stat/sonoff102/POWER1 = OFF (retained)
23:38:13 MQT: stat/sonoff102/RESULT = {"POWER2":"OFF"}
23:38:13 MQT: stat/sonoff102/POWER2 = OFF (retained)

mikeS7, 2017-12-12:
For me the issue is gone after "fixing" wifi channel on the router

ahzazou, 2017-12-12:
I have other sonoffs with better signal and working fine , I read through the troubleshooting I changed client id and I will wait and see and update

ahzazou, 2017-12-17:
after disabling mDNS problem is solved .

closed 2017-12-17 by ahzazou