arendst/Tasmota #1333: Sonoff with HA not updating status

dedynobre, 2017-12-09:
Folks

I have three sonoff installed and configured with HA and it is working very well.
But one thing I realized is that when a manual firing of the sonoff is done physically the status is updated in the web interface but does not update the status in the HA.
This also happens when HA is restarted.

Someone would have figured out how to solve this condition.

Thank you

davidelang, 2017-12-09:
in your home automation software you need to watch for status/ and tele/ 
messages to know what has happened that wasn't commanded by your home automation 
software.

In prior discussions, I've had people tell me that HA doesn't support doing this 
so if this is still true, you are out of luck. I have done this on OpenHAB.

Odianosen25, 2017-12-10:
@dedynobre , in HA when you configure your mqtt switch, you have to put in both command and status topic. The status topic helps you to ensure it is always in sync if pressed physically.

As for when HA reboots, I also have come across that issue and as @davidelang said, the option is to subscribe to the tele/ topic which updates I think every 5 mins or so. I considered this too slow for me, so I deviced another means to achieve my sync.

Regards

davidelang, 2017-12-10:
On Sun, 10 Dec 2017, Odianosen25 wrote:

> As for when HA reboots, I also have come across that issue and as @davidelang 
> said, the option is to subscribe to the tele/ topic which updates I think 
> every 5 mins or so. I considered this too slow for me, so I deviced another 
> means to achieve my sync.

I setmy tele/ period to 60 seconds, I consider getting everything accurate 
within a minute fast enough.

dedynobre, 2017-12-10:
Would you like to provide an example configuration?

moskovskiy82, 2017-12-12:
Also interested in thisto keep track across restarts. And is there a way to publish tele topic as soon as device switches on|off?

zibous, 2017-12-20:
@dedynobre 

This is working for me with Home Assistant 0.60.0.
Used Sonoff Basic Module with Software Program Version	5.10.0a
```
MQT: stat/M1120N/STATUS = {"Status":
     {"Module":1,"FriendlyName":"Drucker",
	  "Topic":"M1120N","ButtonTopic":"0",
	  "Power":0,"PowerOnState":3,"LedState":1,
	  "SaveData":1,"SaveState":1,
	  "ButtonRetain":0,"PowerRetain":0}}
```
HA Switch settings
-------------------
```
- platform: mqtt
  name: "HP Printer"
  state_topic: "stat/M1120N/POWER"
  command_topic: "cmnd/M1120N/POWER"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
  retain: false
 ```

lobradov, 2017-12-20:
Peter, are you sure this works?

Like, when you restart HA, does it ends up showing correct switch status
after restart?
Or when you restart your Sonoff, does it set the relay status to same
status as in HA?

zibous, 2017-12-20:
@dedynobre 

Restarting HA shows the initial status OFF - not working  !

This, it try to fix this with the tele message:

```
state_topic: "tele/M1120N/STATE"

18:52:11 MQT: tele/M1120N/STATE = {"Time":"2017-12-20T18:52:11","Uptime":0,"Vcc":3.170,"POWER":"ON","Wifi":{"AP":2,"SSId":"McNessi","RSSI":84,"APMac":"88:1F:A1:34:3E:82"}}
```
Restarting Sonoff works, because in this case the stat message will be send.
```
00:00:06 MQT: tele/M1120N/INFO3 = {"RestartReason":"Software/System restart"}
00:00:06 UPP: Multicast (re)joined
18:46:04 MQT: stat/M1120N/RESULT = {"POWER":"ON"}
18:46:04 MQT: stat/M1120N/POWER = ON
```

lobradov, 2017-12-20:
Just reading this thread from beginning - I'm using HA and I don't have the
issue as OP reported. Status is synced between HA and Sonoff no matter what
way I use to change it.
I tried switching the relay ON from Sonoff web interface and it's
immediately visible in HA. Obviously, switching it from HA works and is
reflected right away on sonoff, but I also tried switching the relay from
MQTT client, which right away gets reflected in both sonoff and in HA.

What I think is critical is to:
- Make sure you are sending a command to right cmnd/ topic and receiving
status (not tele) from a right stat/sonoff/POWER topic (not results).
- Make sure Sonoff is sending retained statuses (PowerRetain On) so HA can
pick up the status after reboot.
- Make sure HA is set to send retained commands (retain: true) so Sonoff
can pick up the desired status after its own reboot

https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant describes it
pretty well and 100% accurate for switch configuration.

I have issues with Lights (using colors / dimmers and so on) but I guess
that's not what OP was complaining about.

zibous, 2017-12-20:
Negativ, if you use the konfiguration as shown in the wiki and you will:
 **restart the HA** 
than the status is wrong and not in sync with the sonoff device state.

This is because HA do not check the "state_topic: "stat/M1120N/POWER" when restarting.
I think this is only possible with the "tele/M1120N/STATE" topic, because Sonoff will send
this as defined by the reporting period.

So it is only based on the  "state_topic:"  and only when HA restarts,  rest ist working...

lobradov, 2017-12-20:
Hm, I just tested it and it still works.

Maybe you're restarting your MQTT together with HA, and maybe your MQTT is
not configured for persistence, so it looses all states upon reboot?

zibous, 2017-12-22:
```
- platform: mqtt
  name: "HP Printer"
  state_topic: "stat/M1120N/POWER"
  command_topic: "cmnd/M1120N/POWER"
  qos: 1
  payload_on: "ON"
  payload_off: "OFF"
  optimistic: false
  retain: true
```

You must program the sonoff to send the state with the retain flag set. This tells the broker to send the last message on that topic to any new clients that subscribe, such as when HA restarts.

see: https://community.home-assistant.io/t/ha-sonoff-switch-does-not-work-properly/36500/2