arendst/Tasmota #1334: Smoking Boards

nconder, 2017-12-10:
Im not sure what Im doing wrong I have flashed 8 sonoff devices successfully then all of a sudden I've had two devices(s20 and SV) start smoking exactly at the point the code tries to upload. Have tripled checked that I'm using 3v, tx to rx, ground is good, flashmode is dout, baud is 115200, usb ftdi, is good and working with other devices, and I'm only uncommenting the env_default. 

Any ideas or pointers here I'm not sure what I'm missing.

Genosse274, 2017-12-10:
The only situation i know the sonoff smokes is if you have connected main voltage and the programmer. If only the programmer is connected correct nothing should happen, no matter what you upload.

Genosse274, 2017-12-10:
Make sure to have this line in the user_config.h uncommented (should be by default, i think):
`#define SERIAL_LOG_LEVEL       LOG_LEVEL_INFO    // [SerialLog]`
In arduino ide there is a button in the right upper corner for serial monitor. You can also use putty if you have that.

lobradov, 2017-12-10:
Are you sure your programmer is 3v3 and not 5v?
Check not only Vcc line, but also Tx to make sure.

nconder, 2017-12-10:
Well sure enough it is 5v on the Vcc and 3.3 on TX. I don't know how every board I flashed did not fry. So don't take the word of the seller that your ftdi is 3.3v always verify with you volt meter!

closed 2017-12-10 by nconder