arendst/Tasmota #1335: Suggestions

ionciubotaru, 2017-12-10:
Hi @arendst 

Tasmota is a great tool. In the last period you add more and more sensors, which is good for users, but a big time consumer for you. Here are some suggestions:
- add a userSetup() function at the end of setup
- add a userLoop() function at the end of loop
- add a empty string variable to be append at teleperiod messages - the user will put it's own messages
- add `status 99` to get the value of the previous variable
With this tools any user can add it's own sensors, libraries, routines saving your precious time.

In the end, please don't forget that Tasmota will be the best choice for ESP32 chips.
Here are some great tutorials about ESP32 features and secrets https://www.youtube.com/watch?v=jhjZZkKupk8&list=PL3XBzmAj53RnZPeWe799F-uoXERBldhn9

Keep up your good work

davidelang, 2017-12-10:
We want the sensor support in the main package, not something that everyone has 
to re-create themselves (people who want to insert their own code can do it 
easily enough without user*() functions)

There are lots of example sensors tht people can use, and no special variable is 
needed for the tele/ messages, they can just hook into the process for 
generating messages like all other sensors

ESP32 is a very different beast, simply because it is so much more powerful that 
a lot of things are MUCH easier than on the esp8266

but what switches/outlets are out there shipping with an ESP32 in them?

ionciubotaru, 2017-12-10:
Hi @davidelang 
that's true, the more sensors in Tasmota, the better. But even if @arendst add 100 new more sensors, there will be thousands not added.

Regarding ESP32 you cannot find it in outlets because for 1 output and 1 input esp8266 is cheaper, but 90% of Tasmota users are developers, they try to develop their own projects (that's why so many sensors are  added to Tasmota recently) and I am sure that all those users will be happy to use Tasmota's great features on their ESP32 projects.

Never try to stop progress - Microsoft didn't write a kernel form ARM, that's why windows is missing on phones and tablets

davidelang, 2017-12-10:
I'm not trying to stop progress, but I'm also not volunteering someone else's 
time and directing what they must work on.

The architecture of tasmota like software for the ESP32 will be very different 
with all the new capabilities.

But Tasmota development is driven by available hardware, not theoretical 
hardware. So I wouldn't expect to see Tasmota move to ESP32 until there is some 
ESP32 based systems to run it on available to purchase.

arendst, 2017-12-13:
@ionciubotaru most of your suggestions are already available through the function call pointers as can be seen at the bottom of all xsns files. 
```
boolean Xsns06(byte function)
{
  boolean result = false;

  if (dht_flg) {
    switch (function) {
      case FUNC_XSNS_INIT:
        DhtInit();
        break;
      case FUNC_XSNS_PREP:
        DhtReadPrep();
        break;
      case FUNC_XSNS_JSON_APPEND:
        DhtShow(1);
        break;
#ifdef USE_WEBSERVER
      case FUNC_XSNS_WEB:
        DhtShow(0);
        break;
#endif  // USE_WEBSERVER
    }
  }
  return result;
}
```
FUNC_XSNS_INIT is called at the end of setup().
FUNC_XSNS_PREP is called a second before Teleperiod to allow sensors to prepare their data
FUNC_XSNS_JSON_APPEND is called at Teleperiod to let the sensor append it's data to the Teleperiod string
FUNC_XSNS_WEB is called every 2.34 second if a root web page is openend to let the sensor append it's data to the web page string.

``<Start monkey beat on chest>``I must say I did a fine job here``</End monkey beat on chest>``.

I still feel I have to check additions for completeness and code optimizations regarding redundancy and size but if a user wants to add his local device/sensor it's alright with me.

ionciubotaru, 2017-12-13:
Hi @arendst 
Thank you for function description - it is very useful

> 
> Start monkey beat on chest> I must say I did a fine job here</End monkey beat on chest>.
>

You did a great job and you have all reasons to be proud. There are only few people in the world who are able to do a similar job.

There are enough good programmers in Tasmota's community who are able to add sensors and peripherals to main core, but none of them are able to port Tasmota to ESP32

closed 2017-12-13 by ionciubotaru

rulshid, 2018-09-18:
hi, i'm using tasmota in my esp32(with ardino ide), i have downloaded it from 
https://github.com/JacekDob/Sonoff-Tasmota
and moved(copy) all the files inside the /lib into my ardino/library folder but compiling the program shows the error as : **'const char HTTP_HEAD []' redeclared as different kind of symbol**
and i don't know how to solve the issue please help me ..