arendst/Tasmota #1336: Sonoff Bridge not working after upload

Odianosen25, 2017-12-10:
Hello @arendst ,

Please I just got a sonoff Bridge which I have uploaded my config and it seems not to work again and it doesn't connect to the wifi. In short only the power LED comes up.

I have flashed several sonoff basics which I use around my house, but this is my first bridge. Could there be anything I am missing? Need help urgently please.

Regards

Genosse274, 2017-12-10:
Did you upload your config or did you reflash it?
You could check the log via serial.

Odianosen25, 2017-12-10:
@Genosse274,

I just send the Arduino software to upload my Tasmota file. Pls can you point me how I can get the log over serial? As I haven't done it before

Thanks

Genosse274, 2017-12-10:
Make sure to have this line in the user_config.h uncommented (should be by default, i think):
`#define SERIAL_LOG_LEVEL LOG_LEVEL_INFO // [SerialLog]`
In arduino ide there is a button in the right upper corner for serial monitor.

ChukXP, 2017-12-10:
@Odianosen25 
I faced with same problem with last version of firmware. 2 sonoff basic which I flashed can't access to network at all. With WIFI_MANAGER option there is no AP too. Reset not helps. I've spent whole day to find a solution, but no luck. Try to flash bridge with earlier version of firmware. Seems some bug in the last version ...

ChukXP, 2017-12-10:
And one strange thing: few days ago I modified Sonoff Bridge hardware and files (added some commands) xdrv_snfbridge.ino, sonoff.ino, en-GB.h, user_config.h in the last version 5.10.0a. And this firmware works well in Bridge and Basic. 
Yesterday I deceided to compile firmware without changes for Basic module (just added some chages in user_config.h ). And this version can't access to network.

Odianosen25, 2017-12-10:
Hello @ChukXP,

Thanks for everything. I have tried uploading a previous version, but that still didn't bring it up. I don't know where the issue is, but will reload the latest and try an earlier suggestion by @Genosse274  so I can go via the serial logs.

Regards

ChukXP, 2017-12-10:
@Odianosen25 don't forget that Bridge uses RX and TX for communicate with RF controller, so You need to leave switch in same position after flashing.

Odianosen25, 2017-12-10:
@ChukXP,

Thanks but I am plugged  in but can't figure out for the life of me what to do. I sent a test string but nothing happened. Have my baud rate set to 115200 (hope that's right?).

If not can you kindly outline what I need to do? Sorry if these questions are basic, but will really like to get this thing to work; before thinking of buying anew.

Regards

ChukXP, 2017-12-10:
I can advice to flash Your bridge with 5.8.0 or 5.8.0m version. What software You're using for upload?

Odianosen25, 2017-12-10:
@ChukXP ,

I really don't have access to that. I just have 5.10.0a (latest) and 5.5.2b which I think is old. If you have it pls can you hep send over?

Regards

ChukXP, 2017-12-10:
I think if You flashed 5.5.2 unsuccesfully, then problem not in firmware version in Your case ...

Odianosen25, 2017-12-10:
@ChukXP ,

Well was afraid of that. Anyways will get a new one and try again. Kind regards

ChukXP, 2017-12-10:
Find in internet FLASH_DOWNLOAD_TOOLS_V3.4.2_Win.zip and ReadFlash_ESP82XX.zip (need to edit .bat file for choosing right COM port) These utilites for dumping and flashing bin files from devices. Using them You can flash back backup.

Odianosen25, 2017-12-10:
@ChukXP ,

Thanks for that. Will have to find the originals first. Will do and let you know. Kind regards

reloxx13, 2017-12-11:
Did you try to start the WIFI Manager by pressing the button?
Default WIFI Config is WPS Mode, that could be the reason why their is no WIFI AP to connect.
=> https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

Did you change the CFG_HOLDER in the user_config.h to be sure your new config is loaded?
=>https://github.com/arendst/Sonoff-Tasmota/wiki/Troubleshooting#config-problems-can-cause-boot-loops-items-set-in-user_config-to-not-appear-etc