arendst/Tasmota #1337: Magic Home board not flashing

anthonyangel, 2017-12-10:
I'm trying to flash an Arilux  LED controller v1.4 board, but I can't get it to boot in flash mode. I have it connected up as follows (as per [this wiki page):](https://github.com/arendst/Sonoff-Tasmota/wiki/MagicHome-LED-strip-controller)

FTDI               Board
Rx         -->    Tx    
Tx          -->    Rx   
GND     -->    GND
GND     -->    GPIO0 

Using ESPtool I'm getting the error: `A fatal error occurred: Invalid head of packet ('\x00')`, when I use a serial debugger through PIO I can only see a NULL character and no other output.

Not sure if anyone has had any issues with this board before or if something has changed, but I'm pretty much out of ideas.

lobradov, 2017-12-14:
I'd start by checking following things:
- check your FTDI to see if it's true 3v3 (or actually a 5v board that would fry your esp)
- can your FTDI flash another type of board 
above two would make sure that your FTDI is working good and that your PC is properly configured to use it.

- are you supplying 12v via barrel connector?
- is your MagicHome LED controller working with its stock firmware? 
if all above is good, it would imply that board is good and working, so it's only about communication between PC and a board... 

so, final check/attempt: 
- try swapping Rx and Tx on MagicHome LED controller.

anthonyangel, 2017-12-18:
Thanks, I'll double-check and give this another shot, but I'm not so confident:
- I've used the same FTDI before to successfully flash a Sonoff,
- I'm powering the magic-home from the 12V barrel connector, not the FTDI
- Stock firmware works

I'll try with swapping the tx & rx lines to make a straight-though connection, rather than a crossover.

The stock firmware doesn't have update functionality, so I doubt there's any chance of an OTA update

lobradov, 2017-12-18:
You know that GPIO0 needs to be tied to GND to go into flash mode, right?
Hope you didn't forget that ;)