arendst/Tasmota #1338: Arilux-LC11 Pairing a RF remote control?

timtimsson, 2017-12-10:
Hi @all,
how can i pair the remote with the board?
Any ideas?
Thanks!

arendst, 2017-12-10:
If it's the remote delivered with the lc11 proceed as follows:

- Select a different module than lc11
- let  it reboot and select module lc11 (this resets the rf home code)
- press a key on the rf remote (this learns the rf home code)

That should do the trick.

timtimsson, 2017-12-10:
->sonoffBasic->blueLedoff>poweroff/on->arilc11->blueLedOn->poweroff/on->RFcontrolkey On/Off
nothing.. console also nothing, when i pressed a key on/off
Im not sure that the controller is ok - i dont tested it before i flash tasmota..

timtimsson, 2017-12-11:
> press a key on the rf remote (this learns the rf home code)

what do you mean with learns??
(Unfortunately, i don't receive anything...)

--
... i found this on:
https://github.com/mertenats/Arilux_AL-LC0X/blob/master/Arilux.h#L120
*
   RF Remote
   Encoding: Chinese Protocol 1
   Codes provided by KmanOz (https://github.com/KmanOz)
   +--------+--------+--------+
   |   ON   | Toggle |   OFF  |
   +--------+--------+--------+
   | Speed+ | Mode+  | Bright+|
   +--------+--------+--------+
   | Speed- | Mode-  | Bright-|
   +--------+--------+--------+
   |  RED   | GREEN  |  BLUE  |
   +--------+--------+--------+
   | ORANGE | LT GRN | LT BLUE|
   +--------+--------+--------+
   | AMBER  |  CYAN  | PURPLE |
   +--------+--------+--------+
   | YELLOW |  PINK  | WHITE  |
   +--------+--------+--------+
*/

#ifdef RF_REMOTE
#define ARILUX_RF_CODE_KEY_ON           7808513
#define ARILUX_RF_CODE_KEY_TOGGLE       7808514
#define ARILUX_RF_CODE_KEY_OFF          7808515
#define ARILUX_RF_CODE_KEY_SPEED_PLUS   7808516
#define ARILUX_RF_CODE_KEY_MODE_PLUS    7808517
#define ARILUX_RF_CODE_KEY_BRIGHT_PLUS  7808518
#define ARILUX_RF_CODE_KEY_SPEED_MINUS  7808519
#define ARILUX_RF_CODE_KEY_MODE_MINUS   7808520
#define ARILUX_RF_CODE_KEY_BRIGHT_MINUS 7808521
#define ARILUX_RF_CODE_KEY_RED          7808522
#define ARILUX_RF_CODE_KEY_GREEN        7808523
#define ARILUX_RF_CODE_KEY_BLUE         7808524
#define ARILUX_RF_CODE_KEY_ORANGE       7808525
#define ARILUX_RF_CODE_KEY_LTGRN        7808526
#define ARILUX_RF_CODE_KEY_LTBLUE       7808527
#define ARILUX_RF_CODE_KEY_AMBER        7808528
#define ARILUX_RF_CODE_KEY_CYAN         7808529
#define ARILUX_RF_CODE_KEY_PURPLE       7808530
#define ARILUX_RF_CODE_KEY_YELLOW       7808531
#define ARILUX_RF_CODE_KEY_PINK         7808532
#define ARILUX_RF_CODE_KEY_WHITE        7808533
#endif

arendst, 2017-12-12:
That 78085xx is part of the host code and changes with every remote.

You could enable more logging and see if a rf code from your remote is recognised.

timtimsson, 2017-12-12:
... my remote is sluggish, also with new batteries..
only directly nearbynear i received only this (i pressed different keys) - whats wrong?

```
18:38:13 RFR: Hostcode 0x8000, Received 0x800000
18:38:14 RFR: Hostcode 0x8000, Received 0x800000
18:38:19 WIF: Checking connection...
18:38:19 WIF: Connected
18:38:20 RFR: Hostcode 0x8000, Received 0x800000
18:38:22 RFR: Hostcode 0x8000, Received 0x800000
18:38:24 RFR: Hostcode 0x8000, Received 0x800000
18:38:25 RFR: Hostcode 0x8000, Received 0x800000
18:38:32 RFR: Hostcode 0x8000, Received 0x800000
18:38:39 WIF: Checking connection...
18:38:39 WIF: Connected
18:38:59 WIF: Checking connection...
18:38:59 WIF: Connected
```

arendst, 2017-12-13:
If you have a Sonoff Bridge you might want to use it to test your remote
```
12:34:14 RFR: Hostcode 0x361F, Received 0x361F13
12:34:14 RSL: Group 0, Index 1, Command COLOR, Data 10
12:34:14 MQT: stat/led4/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FFFF000000"}
12:34:17 RFR: Hostcode 0x361F, Received 0x361F0A
12:34:17 RSL: Group 0, Index 1, Command COLOR, Data 1
12:34:17 MQT: stat/led4/RESULT = {"POWER":"ON","Dimmer":100,"Color":"FF00000000"}
```
And the bridge sees this:
```
12:34:15 BRG: Received A4 30 2A 01 B8 04 F6 36 1F 13 55 
12:34:15 MQT: tele/bridge1/RFRECEIVED = {"RfReceived":{"Sync":12330,"Low":440,"High":1270,"Data":"361F13","RfKey":"None"}}
12:34:18 BRG: Received A4 30 20 01 B8 04 EC 36 1F 0A 55 
12:34:18 MQT: tele/bridge1/RFRECEIVED = {"RfReceived":{"Sync":12320,"Low":440,"High":1260,"Data":"361F0A","RfKey":4}}
```

johny-mnemonic, 2018-02-10:
Hi @arendst , thanks for all the work. 
I am not sure, whether it was the last update, or just pure luck while trying different GPIO combinations, but my RF receiver is finally working.
I got similar board to the board mentioned above. It is RGBW Magic Home with RF remote, i.e. Arilux AL-LC10.
[image]
It flashed fine and after some fiddling, I made LED lighting working properly, but RF receiver resisted for more then a week.
I finally ended with following template:
```
  { "MagicHome RF",    // Magic Home RF (ESP8266) - (Arilux LC10)
     GPIO_USER,        // GPIO00 Optional Button
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     GPIO_LED1_INV,    // GPIO02 Blue onboard LED (optional)
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor0
     GPIO_ARIRFRCV,    // GPIO04 RF receiver input 
     GPIO_PWM2,        // GPIO05 RGB LED Green
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_PWM3,        // GPIO12 RGB LED Blue
     GPIO_PWM4,        // GPIO13 RGBW LED White
     GPIO_PWM1,        // GPIO14 RGB LED Red
     GPIO_LED2_INV,    // GPIO15 RF receiver control
     0, 0
  },
```
I tried lot of hacks and GPIO combinations, but there were no messages in web console when i pushed buttons on RF (with weblog level 4).
I then tried to find how RF receiver is enabled in Tasmota and I found function AriluxRfInit() which enables RF receiver only if this interesting condition is met:
`if ((pin[GPIO_ARIRFRCV] < 99) && (pin[GPIO_LED2] < 99)) {`
Do I understand it correctly, that for RF receiver to work, I need to have also GPIO_LED2[_INV] defined?
If yes, I think it would be useful for others to mention it somewhere in the wiki. Should I create PR, or would you put it there?

So, right now it works with GPIO4 as RF pin and GPIO15 as LED2 pin, but the reliability of pushes is subpar. Lot of that goes to that cheap flimsy remote, but even when it sends the key press, it is not always decoded correctly.
Here is my table of codes for few keys to make it clear:
ON - 0x775301 (0xEEA60C, 0xEEA61C, 0xEEA6FC, 0x775201, 0xEC08FC, 0xE008FC, 0xEEA0FC)
OFF - 0x775303 (0xEEA600, 0xEEA606, 0xEEA618, 0xEEA638, 0xEC08F8, 0xE008F8)
M+ - 0x775305 (0xEEA600)
M- - 0x775308 (0xC008E7, 0xEEA667)
Those codes in brackets are not doing anything and my perception is, that whenever I press the button for longer time (than some default key resolution time) it produces these different codes. Sometimes it produces only one of these, sometimes, whole bunch of them, often together with the correct key code (i.e. 0x775301).
Any idea how to improve this? Is there some configuration for communication frequency with that SYN480R chip, or some keypress delays, etc. ?

My final goal is, to be able to make these buttons only send MQTT messages, not directly doing anything on the LED controller itself, so I am definitely not finished with this...

arendst, 2018-02-11:
1) I see no need to specifically mention the use of led2 as it is virtually "under the hood" and needed for the LC-11 to power up it's RF reception hardware. Does your module need this control signal too or does it also work without the RF enabling?

2) The RF signal contains a kind of home code which needs to be learned to the software first. This is done once when the first time you select a module type with both GPIO_ARIRFRCV and GPIO_LED2 defined. So to reset the learned home code you will need to switch to a module like Sonoff Basic and then back to your config module.

3) Your final goal will need to be services by a Sonoff bridge and not the available RF decoding as it is not designed for it. It's a very stripped version from a full RF reception library and I won't support full RF in software as it burdens the rest of the functions too much.

johny-mnemonic, 2018-02-12:
1. Yes, finding, that I need to configure GPIO_LED2 to GPIO15 was the last piece of puzzle to make it working. As soon as I remove the definition of GPIO_LED2 from GPIO15, RF remote stops working.

2. That's exactly what I was doing when I was trying to find working config.

3. I am not sure, whether I understand you here. I checked Sonoff Bridge and it looked interesting at first, but then I tried to search for remote controls... Seems like it's pretty tough to find any. At least if you don't search for garage door openers with four buttons, these are plenty. Any advice on good remote control for LED lighting or window blinds, which could be used with Sonoff Bridge? Ideally with 16+ buttons if you support that amount in Tasmota. 

Thanks

BTW: Would it be possible to use my current RF remote with Sonoff Bridge?

arendst, 2018-02-12:
My LC-11 remote works with Sonoff bridge. Other than that I abandoned 433MHz (RF) signals a while ago in favor of wifi.

johny-mnemonic, 2018-02-13:
Great news, thanks for info!
Could you recommend some wifi remote control you have good experience with?