arendst/Tasmota #1339: Echo Show doesn't detect Sonoff in Hue or WeMo mode

elylv, 2017-12-10:
I have an Echo Show, and a Sonoff TH16 flashed to Tasmota.  I have tried setting to WeMo and HUE emulation, and searching for new devices, but nothing is ever found.

I have checked with the gen1 Hue app on my phone, and it finds the device as a bridge, and I can control it.

I have set Debug level to 3 and checked what happens when I scan for new devices with Alexa, but nothing shows.  When I searched with the gen1 app, I get:
UPP: Hue 3 response packets sent to 192.168.0.27:1901

I'm at a loss as to what could be the problem.  Everything's on the one network.  Does the Echo Show (UK) support this?  I can't see why it wouldn't.  Anything else I can try to troubleshoot?

jimjimminy, 2017-12-11:
I am having the same issue it seems to be related to a recent Echo update. There are some additional details here https://bitbucket.org/xoseperez/fauxmoesp/issues/25/echo-doesnt-discover-device

moose51789, 2017-12-11:
Good and bad news, i had a similiar issue and walked through it with an alexa team support person only to find out that the shows don't support hue devices that aren't through the hue skill.... which means they will likely be doing the same to newer devices.  If i added a dot to my account it worked no problem but on its own its a no go.

erik-wendt, 2017-12-13:
I had the same problem **but now it works**. I connected to my wifi router (via browser) and saw that the echo dot was connected to my wifi using 5GHz and the Sonoff used 2.4GHz. So I deactivated the 5GHz frequency and set the emulation to Hue and now everything is working