arendst/Tasmota #1340: PWM- vs. Dimmer/Color-Mode (SetOption15) - why are they xor'ed?

mirko, 2017-12-11:
Why can I only use *either* the `Pwm`- *or* the `Dimmer`/`Color` control mode?

Use case: I'm having a 230AC dimmer controlled via PWM. If PWM exceeds a value above 900+something, the dimmer loses sync with the 50Hz main line phase and does things it shouldn't do.
When being in PWM-mode I can just set `PwmRange` - and voilá! - I don't have to worry about stinky smoke originating from the dimmer.

However, now I also wanna use features only available in `Dimmer`/`Color` control mode, such as fading.
No sooner said than done: Setting `SetOption15=1`, happily playing around with the new `Dimmer`- and `Fade`-commands and.. telling from the smoke coming up it looks like we got a new pope!

Lesson learned: apparently `PwmRange` is not taken into account when being in "`Option15`-Mode".

So I started digging around in the code, trying to find the bit where - at least I assumed so - the `Dimmer`/`Color`-commands would actually get translated into PWM values, so I might add a check for the eventual PWM value being within `PwmRange` or not.

It seems however that setting `light_current_color[i]` to something between 0 and 100 is enough to magically change the PWM duty cycle. Obviously I'm missing the part where `light_current_color` translates into the PWM value - any pointer would be appreciated.

Maybe instead of just checking for the `Dimmer` value being within `PwmRange` we could even create a two-way binding between the `Dimmer`/`Color`- and the `Pwm*`-values to keep both values in sync(?).

However I'm afraid there's probably a reason why `Pwm*` control and `Dimmer`/`Color` control can only be used exclusively.

I also admit, my use case is probably a rather unique and personal one, however it actually *is* a use-case - and even if it's not worth the effort, I keep scratching my head about the PWM-translation part.

arendst, 2017-12-13:
PWM is the legacy result of Color/Dimmer. Originally PWM allowed values up 1023. 

Then Color/Dimmer came along and a downplay from 1023 to 255 was needed to have color control within 8 bits. The PWM value is just re-calculated up by multiplying 255 with the value of 1023 divided by 255. 

This 1023 was then made user configurable on demand with command PwmRange allowing more PWM control for reasons unknown to me. At the same time PwmFrequency was made user configurable.

As said the color/dimmer value is always re-calculated from a value of 0-255 to 0-PwmRange.

You might want to consider lowering PwmFrequency to keep the current pope alive.

jacobkuriakose, 2018-07-14:
Could you please tell me how to control that AC dimmer via tasmota using PWM..?