arendst/Tasmota #1341: IP Address, etc not visible or changeable

netcompsys, 2017-12-11:
I fixed my IP address etc in config.h but I note that

the "Configure Wifi" / "Wifi Paramters" does not display nor allow you to change the IP Address, Mask,,gateway or DNS

jetema, 2017-12-11:
If you can reach your sonoff try it on cosole, see IP commands at 
 
 https://github.com/arendst/Sonoff-Tasmota/wiki/Commands#wifi

for instance

IPAddress1 |   | Show current IP address
IPAddress1 | 0.0.0.0 | Use dynamic IP addresses (DHCP) and restart
IPAddress1 | x.x.x.x | Set (decimal) static IP address and restart
IPAddress2 |   | Show current IP address of Gateway
IPAddress2 | x.x.x.x | Set (decimal) IP address of Gateway and restart
IPAddress3 |   | Show current subnet mask
IPAddress3 | x.x.x.x | Set (decimal) subnet mask and restart
IPAddress4 |   | Show current IP address of DNS server
IPAddress4 | x.x.x.x | Set (decimal) IP address of DNS server and restart

netcompsys, 2017-12-11:
Yes indeed, and I know what the settings are currently as I have just flashed the devices

However at some point in the future I might want to review the settings as part of faultfinding or network reconfiguration, and given all the other information that is available on the Sonoffs web configuration pages, I was surprised to find that this information was not displayed and that it was not possible to alter these parameters through the web interface

ChukXP, 2017-12-12:
IP, Gateway, SubNet Mask, Host Name, MAC info are available in "Information" menu.

arendst, 2017-12-13:
As code space is tight I have decided to keep webpage data as limited as possible. You can always use the command line in the web console to get to all status information like ``status 0``. This also shos your network parameters.

shayshtern, 2018-01-24:
opss I send command : IPAddress1 | 192.168.1.21
now I can't reach my sonoff.... opss 
what can I do ?

davidelang, 2018-01-24:
reset by holding down the button for 4 seconds

closed 2018-06-08 by stale[bot]