arendst/Tasmota #1343: Version 5.10.0a execption on invalid POWERx command

mikelh, 2017-12-11:
using V 5.10.0a
not defining relais 4, only relais 1. 
sending MQTT: cmnd/sonoff/POWER4 ON
device restarts after exception instead of ignoring the command and returning an error
same works on the web console with POWER4 ON

arendst, 2017-12-13:
Can't reproduce. Pls provide output of command ``status 0``

mikelh, 2017-12-13:
using web console window.
configured only one relais and 4 switches.

13:45:34 CMD: status 0
13:45:34 MQT: stat/sonoff_7513/STATUS = {"Status":{"Module":18,"FriendlyName":"Sonoff","Topic":"sonoff_7513","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":4,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
13:45:34 MQT: stat/sonoff_7513/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Uptime":0,"Sleep":0,"BootCount":49,"SaveCount":226,"SaveAddress":"3FA000"}}
13:45:34 MQT: stat/sonoff_7513/STATUS2 = {"StatusFWR":{"Version":"5.10.0a","BuildDateTime":"2017-12-10T15:28:41","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
13:45:34 MQT: stat/sonoff_7513/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"xyzzy","SSId2":"indebuurt2","TelePeriod":30,"SetOption":"00000009"}}
13:45:34 MQT: stat/sonoff_7513/STATUS4 = {"StatusMEM":{"ProgramSize":457,"Free":2612,"Heap":24,"ProgramFlashSize":4096,"FlashSize":4096,"FlashMode":3}}
13:45:34 MQT: stat/sonoff_7513/STATUS5 = {"StatusNET":{"Hostname":"sonoff_7513-7513","IPAddress":"192.168.1.XXX","Gateway":"192.168.1.XXX","Subnetmask":"255.255.255.0","DNSServer":"192.168.1.XXX","Mac":"XX:XX:XX:XX:XX:XX","Webserver":2,"WifiConfig":2}}
13:45:34 MQT: stat/sonoff_7513/STATUS6 = {"StatusMQT":{"MqttHost":"XXXXX.home.lan","MqttPort":1883,"MqttClientMask":"DVES_X","MqttClient":"DVES_29DD59","MqttUser":"DVES_USER","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
13:45:34 MQT: stat/sonoff_7513/STATUS7 = {"StatusTIM":{"UTC":"Wed Dec 13 12:45:34 2017","Local":"Wed Dec 13 13:45:34 2017","StartDST":"Sun Mar 26 02:00:00 2017","EndDST":"Sun Oct 29 03:00:00 2017","Timezone":99}}
13:45:34 MQT: stat/sonoff_7513/STATUS10 = {"StatusSNS":{"Time":"2017-12-13T13:45:34","Switch1":"OFF","Switch2":"ON","Switch3":"ON","Switch4":"ON"}}
13:45:34 MQT: stat/sonoff_7513/STATUS11 = {"StatusSTS":{"Time":"2017-12-13T13:45:34","Uptime":0,"Vcc":2.788,"POWER":"OFF","Wifi":{"AP":1,"SSId":"xxxxx","RSSI":82,"APMac":"xx:xx:xx:xx:xx:xx"}}}

13:49:12 CMD: switchtopic
13:49:12 MQT: stat/sonoff_7513/RESULT = {"SwitchTopic":"gf_floor"}

13:52:07 CMD: switchmode1
13:52:07 MQT: stat/sonoff_7513/RESULT = {"SwitchMode1":2}
13:52:15 CMD: switchmode2
13:52:15 MQT: stat/sonoff_7513/RESULT = {"SwitchMode2":1}
13:52:21 CMD: switchmode3
13:52:21 MQT: stat/sonoff_7513/RESULT = {"SwitchMode3":1}
13:52:29 CMD: switchmode4
13:52:29 MQT: stat/sonoff_7513/RESULT = {"SwitchMode4":2}

configured GPIO3 as Relais1

13:55:16 CMD: power1 on
13:55:16 MQT: stat/sonoff_7513/RESULT = {"POWER":"ON"}
13:55:16 MQT: stat/sonoff_7513/POWER = ON
13:55:21 CMD: power1 off
13:55:21 MQT: stat/sonoff_7513/RESULT = {"POWER":"OFF"}
13:55:21 MQT: stat/sonoff_7513/POWER = OFF

sending "power2 on" reboots:
13:56:25 MQT: tele/sonoff_7513/INFO3 = {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x4000bf64 epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000001 depc:0x00000000"}

mikelh, 2017-12-13:
please disregard this issue.
was using wrong project config on my system while doing regression tests in own
development fork.
Confirm this will not happen in stock firmware.
Thanks much for your attention and support.

closed 2017-12-13 by mikelh