arendst/Tasmota #1344: Improve Domoticz integration

emontnemery, 2017-12-11:
The documented integration with Domoticz using dummy devices working OK, but is not perfect because:
- Domoticz does not automatically discover new Tasmota devices, instead it requires a fairly involved process of manually adding a device in Domoticz, finding its "idx", entering this "idx" in Tasmota
- There is no synchronization, the dummy switch status will update in Domoticz regardless of if the Tasmota device received the message or not.

What could be done to improve, for example:
- Implement a Tasmota plugin for Domoticz using Tasmota's own MQTT protocol
- Add support to Domoticz' MQTT implementation to bind hardware and send commands
Both of these could be implemented based on the MySensors MQTT gateway plugin already included in Domoticz

Any suggestions?

davidelang, 2017-12-11:
On Mon, 11 Dec 2017, emontnemery wrote:

> - Implement a Tasmota plugin for Domoticz using Tasmota's own MQTT protocol
> - Add support to Domoticz' MQTT implementation to bind hardware and send commands

what is the difference between the mqtt messages Tasmota currently processes and 
what is needed for Domoticz?

emontnemery, 2017-12-11:
With the suggested (Tasmota Wiki) way of using Domoticz, you create "virtual sensors" which can be any type, including a switch (so you might want to use "virtual switch" for controlling a Tasmota relay, or a "virtual dimmer" for a Tasmota LED dimmer).
When you change state of the virtual switch in Domoticz, for example through Domoticz WebUI, Domoticz will publish an MQTT message informing about the state change. This means there is no synchronization, if the Tasmota switch is down and the command is not received, this is not reflected in the UI.
Also, the "virtual sensor" must be manually created in Domoticz.

I'm thinking it makes sense to either:
- Modify the Domoticz MQTT implementation so it is possible to create devices directly under the MQTT HW instead of relying on "virtual sensors".
In this way Domoticz will first request a state change and then, if it succeeds inform the world about the state change. This is fairly simple to implement, I have something more or less working. Tasmota will also need some small updates to react to an order from Domoticz to change state instead of reacting to Domoticz informing about the state change.
-- Pro: Simple to implement, the plumbing is already there in Domoticz
-- Con: I'm not sure if it's possible to add automatic discovery support
- Add a new Hardware type "Tasmota MQTT" to Domoticz speaking Tasmot native MQTT protocol. This is a little bit more work, but there is already an implementation for "MySensors Gateway with MQTT interface" which is essentially the same thing but speaking to a MySensors gateway.
-- Pro: Less / no change to Tasmota needed, automatic discovery will work (if the Tasmota is set up with a suitable MQTT topic, for example tasmota/%topic%/%prefix%/)
-- Con: A bit more involved changes to Domoticz

davidelang, 2017-12-12:
both suggestions you have are for changes to Domoticz, so you should probably be 
talking to the developers who maintain it.

But the first thing they will ask you is what the difference is between 
"Domoticz MQTT" and "Tasmota MQTT". Unless you can specify what the difference 
is that needs to be supported, you are unlikely to make much progress.

I suspect that when you dig into it to define the difference, you will find that 
the difference is not what you are thinking it needs to be.

emontnemery, 2017-12-12:
Maybe my initial question was unclear, let my try again :)
I'm using a number of Tasmota flashed devices together with Domoticz. This works pretty well, but there are a couple of missing things:

1. No error detection due to the use of "dummy device" in Domoticz
   This can (for example) be fixed by updating the existing Domoticz MQTT integration to send actuator commands instead of just sending status updates or by a "real" Tasmota plugin for Domoticz.
2.  Auto discovery, several manual steps are needed in both Domoticz and Tasmota when a new Tasmota device is added
  For this a "real" Tasmota plugin would be necessary?

I'm thinking about implementing either of these solutions and submitting patches to Tasmota/Domoticz.
I hope to get some suggestions here from others who use Tasmota together with Domoticz, but I'll bring it up in the Domoticz forums too!

> I suspect that when you dig into it to define the difference, you will find that
> the difference is not what you are thinking it needs to be.

I'm not sure what you'r getting at, can you explain?

rlust, 2017-12-12:
I see the value in what your describing, would be very helpful to have a domoticz plugin for Tasmota!

closed 2018-04-02 by emontnemery