arendst/Tasmota #1345: Sonoff 4CH and Switchmode

saundersrg, 2017-12-11:
I'm trying to wire my Sonoff 4CH in what I think is the same way as #1194. Now I've got the Switchmode working exactly as expected for a normal household wallswitch ("switchmode1 2" in console), but I want to do this for a 4 switch panel in the wall. The GPIOs on the Sonoff and in Tasmota only offer 3x Switches that I can configure (IO2, TX, RX).

Is there a way for me to change the pushbuttons to switches? I don't mind soldering to the pushbuttons (except GPIO0 with the whole factory reset thing).

Ideally I'd like to use IO2 as a switch to bypass factory reset from holding in IO0, and I'd like to solder a wire to IO9, 10, and 14 and use these as switches instead of pushbuttons.

Any pointers?

saundersrg, 2017-12-12:
Not an urgent request, figured it's easier and more elegant to just replace the wall switches with bell press switches...

Would still like to know if the buttons can be changed to switches though.

davidelang, 2017-12-12:
edit the module config or use the module type wemos and define all the GPOIs as 
you wish

saundersrg, 2017-12-12:
Makes sense, thanks! Will try that out.

closed 2017-12-12 by saundersrg