arendst/Tasmota #1346: Connect Sonoff 4CH Pro or RF Bridge to 433Mhz weather station

ericbun1984, 2017-12-11:
Hi all,

I was wondering of it is possible, or if anyone has succeeded in interfacing Sonoff HW with 433mhz receivers with one of the very available cheap weater stations such as described here:
- http://lucsmall.com/2012/04/30/weather-station-hacking-part-3/
- https://github.com/hassebjork/WeatherReader
- http://forum.arduino.cc/index.php?topic=142871.0

Can anyone tell if this works on 4CH Pro, where 433mhz is handled by a separate chip?

Thanks in advance

RoganDawes, 2018-02-06:
It is my understanding that the RF receiver is connected to the STM32 microcontroller, which does all the setup and decoding of what is received. Any support for alternative transmissions would therefore require updates to the STM32 firmware as well as Tasmota.

app41, 2018-02-06:
nice idea to expand the flexibility to use the 433 receiver
likey sonoff made a custom software to manage a minimum of local automation for common people who don't want or don't know how to program

I never saw any way to program the STM32, and I would have competencies neither the compiler
why not to try to bypass the STM32?
just a jump soldered connecting D0 output of SYN470 to any free input pin of ESP8285, then there are plenty of lib to manage and decode anything

maybe to read Oregon scientific remote sensor, but I don't know if they work on the same frequency and if the SYN470 can detect the manchester coding that I know oregon uses

RoganDawes, 2018-02-06:
There is a header on J6 and J7 which looks to be connected to the STM32, and should be able to reprogram it with suitable firmware. 

It will ultimately depend on the connections between the ESP8285 and the STM32, as to what can be done. 

To my understanding, the STM32 is acting as an interface between the dipswitches, the buttons, the RF module, and the ESP. It then emits "button presses" to the ESP depending on how the dipswitches are configured. If this is the case, it may be more difficult to change how this module works. If there are any lines that can be repurposed as e.g SPI or I2C, then one could establish an alternative protocol for communications between the two CPU's. Then something like the above might be possible.

First step would be to trace out which pins of the STM32 are connected to which pins of the ESP8285. Then we can see if there is a possibility of remapping e.g. the UART or SPI on each uC to allow them to talk to each other.

arendst, 2018-02-06:
No need to trace. Just look at the wiki at https://www.itead.cc/wiki/Sonoff_4CH_Pro which also contains the schematics of the 4CHPro and have great fun.

RoganDawes, 2018-02-06:
Good point. So yes, J6 and J7 are for monitoring and reprogramming the STM32. And I see how the STM32 intermediates all the access to the relays, dipswitches and buttons. I guess this is how they make sure that the interlocks work, etc

And the RF circuit is connected to DIN (PA12) on the STM32. I'll note that the receiver is a fixed purpose ASK/OOK receiver, which may exclude reception of alternative encodings. See https://www.birdandgua.net/bird/wp-content/uploads/2016/09/SYN470R-Synoxo.pdf

As a proof of concept, it would be easy enough to connect TX-RX on each controller, and let them interface to each other via the UART. Obviously, one would have to disable any other logging on the ESP8285 UART (and on the STM32, potentially).

pilot1981, 2020-04-25:
@RoganDawes can you share result of your POC?
Have you connected SYN470 directly to ESP?
Which result?
Some photos?

RoganDawes, 2020-04-25:
I made no progress on this, my 4CH-PRO is in active use, I'm afraid.

RoganDawes, 2020-04-25:
Having taken another quick look at the schematic and datasheet of the STM32F030C6, it seems that PB4,5,6,7 are connected directly from the STM to the ESP, and could be repurposed if desired. PB6 and 7 have alternate functions of USART1 RX and TX, so USART 1 could be remapped from the existing pin header to talk to the ESP8285 using a serial protocol. Alternatively, they can also operate as I2C SCL and SDA. If the ESP8285 can select pins 0 and 14 as I2C pins, this may also be a good way to communicate between the ESP and the STM.
This would allow for a much more powerful and flexible command protocol to be used between the two chips. Of course, the firmware in the ESP and the STM would have to be kept in sync, complicating things a lot.