arendst/Tasmota #1347: Connectivity issues when using Mosquitto / connection overloaded?

GAZ082, 2017-12-12:
Hi there. When i use Mosquitto as broker takes a lot of time to access the web interface of the Sonoff Basic switches, that's not the case with the standard Home Assistant broker. I dont know if it's so much a "connectivity issue" or more like Mosquitto is spamming the connection and making the switches respond much slower (or even provoking restarts) because of this.

I'm using 5.10.0.

davidelang, 2017-12-12:
what do the logs show?

I'll bet that you are getting disconnected from the mqtt broker (see the 
troubleshooting page for the problems this causes)

GAZ082, 2017-12-12:
No worries @davidelang , i isolated the issue to the ui_switch component of the dashboard package.

closed 2017-12-12 by GAZ082