arendst/Tasmota #1348: Support for loxone

leschNUX, 2017-12-12:
Hello arendst,

Sonoff-Tasmota supports a lot of Home Automation Integration Platforms.
Do you plan support loxone, too?
API LOXONE:
https://www.loxone.com/enen/kb/web-services/

br

wunderbaum, 2017-12-12:
This could help:
[https://github.com/alladdin/node-lox-mqtt-gateway](https://github.com/alladdin/node-lox-mqtt-gateway
)

elpaso666, 2018-02-25:
Hello,
you can start using it via SYSLOG and UDP port. I did not test exact functions, because I am only having a power socket right now. But even the ON/OFF trigger is creating correct SYSLOG message which can be captured on Loxone.

Only potential problem is missing the ability for FIXing the source port of Tasmota.

Jakub