arendst/Tasmota #1349: RFE: New device support

macemoneta, 2017-12-12:
There are in-wall switches which are marketed under several names: GRESATEK Smart WiFi Light, Laduo, Joso, Lyasi, DLX Control Solutions, DLX-KS602. Uses esp8255.

- The blue WIFI led is controlled by GPIO 13. The LED is on when the IO is LOW.
- The relay and the bulb LED are controlled by GPIO 12 – they are on when IO is HIGH.
- The capacitive touch switch is connected to GPIO 0 and the IO goes low when touched.

They currently support eWeLink.  Details from:
https://makingsomestuff.wordpress.com/2017/11/07/flashing-a-gresatek-smart-wifi-light-switch-with-an-esp8285-chip-or-joso-or-lyasi-or-dlx-control-solutions-dlx-ks602/

arendst, 2017-12-13:
So using the Sonoff Touch module selection should satisfy your request as it is the perfect match.

macemoneta, 2017-12-13:
Excellent, thanks!

closed 2017-12-13 by macemoneta