arendst/Tasmota #1350: Feature Request : POW power usage Last Month and This Month

TommySharpNZ, 2017-12-12:
We currently have the power today and yesterday. Would it be difficult to add this month and last month?

I'm sure I can get openhab to do it but having it calculated on device and sent out via MQTT would be more reliable...

MatzeJoerling, 2017-12-12:
the only reliable way, in my opinion, is to save the data in some sort of database. openhab sure has one? 
i use influxdb for saving power data and grafana for displaying.

stefanbode, 2017-12-13:
If you dont‘t Update your Firmware to often you are also fine with storing the values into the flash. Anyhow the is a risk to loose this data and afaik there is no way to reinitilize the values with some command. I cause of failure they start at 0. updating firmware always creates some risk that you don’t get the old configuration working.

arendst, 2017-12-13:
I Prefer an external tool for this.