arendst/Tasmota #1351: Help with custom sensor? :-/

gajotnt, 2017-12-12:
I Know this is not the correct "forum", but i tried contacting the person responsible for this PR and he doesnt reply, and just maybe one of you can help.

Im using this PR from tasmota:
[Tasmota with PZEM004T support](https://github.com/vicfergar/Sonoff-Tasmota/tree/develop
) to use the PZEM004T wattmeter (its really accurate and can handle really high current).
After changing the GPIO of the TX and RX in **sonoff_template.h** and using the 5v from the sonoff basic i got it sending readings to sonoff:

```
20:57:25 MQTT: tele/sonoff/STATE = {"Time":"2017-12-12T20:57:25", "Uptime":1, "Vcc":3.447, "Wifi":{"AP":1, "SSID":"TNT", "RSSI":40, "APMac":"9C:97:26:9F:A5:0F"}}
20:57:25 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T20:57:25", "Yesterday":0.000, "Today":10.534, "Period":9, "Power":551.0, "Voltage":236.5, "Current":2.470}
20:58:25 MQTT: tele/sonoff/STATE = {"Time":"2017-12-12T20:58:25", "Uptime":1, "Vcc":3.447, "Wifi":{"AP":1, "SSID":"TNT", "RSSI":40, "APMac":"9C:97:26:9F:A5:0F"}}
20:58:25 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T20:58:25", "Yesterday":0.000, "Today":10.544, "Period":10, "Power":555.0, "Voltage":237.5, "Current":2.480}
20:59:25 MQTT: tele/sonoff/STATE = {"Time":"2017-12-12T20:59:25", "Uptime":1, "Vcc":3.449, "Wifi":{"AP":1, "SSID":"TNT", "RSSI":40, "APMac":"9C:97:26:9F:A5:0F"}}
20:59:25 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T20:59:25", "Yesterday":0.000, "Today":10.553, "Period":9, "Power":556.0, "Voltage":235.9, "Current":2.510}
21:00:25 MQTT: tele/sonoff/STATE = {"Time":"2017-12-12T21:00:25", "Uptime":1, "Vcc":3.449, "Wifi":{"AP":1, "SSID":"TNT", "RSSI":42, "APMac":"9C:97:26:9F:A5:0F"}}
21:00:25 MQTT: tele/sonoff/ENERGY = {"Time":"2017-12-12T21:00:25", "Yesterday":0.000, "Today":10.562, "Period":9, "Power":551.0, "Voltage":236.2, "Current":2.480}
```

I just cant get it do publish to domoticz and my guess is that the **ixsns_pzem004t.ino** file doesnt have any reference to domoticz.

the xsns_hlw8012.ino file has this:

```
#ifdef USE_DOMOTICZ
  if (option) {  // Only send if telemetry
    dtostrf(pet * 1000, 1, 1, stemp1);
    domoticz_sensor4(pw, stemp1);
  }
#endif  // USE_DOMOTICZ
```

from what i can tell it's that command that sends information from the POW the defined idx in domoticz_sensor4.

i just dont have the knowhow to do this "minor" change, tried the paste and pray method but no luck :(

I really love the simplicity and power of this firmware and would like to use it in all of my devices.

Sorry for posting this here

gajotnt, 2017-12-12:
Got it working, after very long trial and error, in case anyone wants to use this here its goes:
add:
```
#ifdef USE_DOMOTICZ
    domoticz_sensor4(power, sTruePower);
#endif  // USE_DOMOTICZ
```
right after: 
```
  snprintf_P(svalue, ssvalue, PSTR("%s\"Yesterday\":%s, \"Today\":%s%s, \"Power\":%s, \"Voltage\":%s, \"Current\":%s}"),
             svalue, sKWHY, sKWHT, (withPeriod) ? sPeriod : "", sTruePower, sVoltage, sCurrent);
```
in **ixsns_pzem004t.ino**

gajotnt, 2017-12-12:
Cried victory too soon...
Pow:
`22:05:33 MQTT: domoticz/in = {"idx":49,"nvalue":0,"svalue":"754;32.4"}`

PZEM:
`22:06:02 MQTT: domoticz/in = {"idx":48,"nvalue":0,"svalue":"0;359.0"}`

The "359.0" value of the pzem should be where the "0" is. some more digging to do -.- xD

davidelang, 2017-12-12:
I don't think you can move the serial port to different GPIOs on the ESP8266

gajotnt, 2017-12-12:
```
  { "PZEM004T",        // Board with PZEM004T (ESP8266)
     GPIO_USER,        // GPIO00
     GPIO_USER,        // GPIO01 TX Serial RXD and Optional sensor
     0,                // GPIO02
     GPIO_USER,        // GPIO03 RX Serial TXD and Optional sensor
     GPIO_PZEM_RX,     // GPIO04 PZEM004T RX
     GPIO_PZEM_TX,     // GPIO05 PZEM004T TX
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER,        // GPIO12
     GPIO_LED1_INV,    // GPIO13 BLUE LED
     GPIO_USER,        // GPIO14 Optional sensor
     0, 0, 0
  },
```
In the original code it was using GPIO 4 and 5. Since the basic doesn't have those I set it to use GPIO 1 and 14. Works like a charm.

gajotnt, 2017-12-13:
In case anyone wants to use pzem004t the maker of the modified Tasmota FW replied with the correct code to insert :)
 here it goes:
```
#ifdef USE_DOMOTICZ
  if (withPeriod) {  // Only send if telemetry
    dtostrf((float)pzem_kWhtoday / 100000, 1, 1, sKWHT);
    domoticz_sensor4(pzem_truePower, sKWHT);
  }
#endif  // USE_DOMOTICZ
```

closed 2017-12-13 by gajotnt