arendst/Tasmota #1352: PulseTime Help

aferraz, 2017-12-12:
Hi @arendst,
 in the wiki I can see:
PulseTime<x> | 1..111 | Set PulseTime for relay<x> with 0.1 seconds increment
-- | -- | --
PulseTime<x> | 112..64900 | Set PulseTime for relay<x> with 1 seconds increment starting with 12 seconds (113 = 13 seconds etc.)

so, from what I understand, a PulseTime of 123 should be 23 seconds.
Is this correct ?

I use Electrodragons for covers, I timed 23 seconds to have cover open/close.
However if I set pulsetime to 123 the relay stay open longer, I need to set pulsetime to 43 for a correct opening/closing.
 
Thanks
 Alessandro

davidelang, 2017-12-12:
On Tue, 12 Dec 2017, Alessandro wrote:

> Hi @arendst,
> in the wiki I can see:
> PulseTime<x> | 1..111 | Set PulseTime for relay<x> with 0.1 seconds increment
> -- | -- | --
> PulseTime<x> | 112..64900 | Set PulseTime for relay<x> with 1 seconds [...]

yes

> I use Electrodragons for covers, I timed 23 seconds to have cover open/close.
> However if I set pulsetime to 123 the relay stay open longer, I need to set pulsetime to 43 for a correct opening/closing.

what shows up in the logs? do you have any communications problems with the mqtt 
server during this time? that can cause all processing to pause

aferraz, 2017-12-12:
@davidelang I've done the tests by web interface because at first I thought about a mqtt problem.
Before using the pulsetime I used nodered giving the off command to the relay after n seconds, never had these problems.

Thanks

davidelang, 2017-12-12:
if the device is trying to reconnect to the mqtt server, everything else pauses 
while it is trying. it doesn't matter if you are sending a command via serial, 
web, mqtt, or anything else, _everything_ stops

This is why I asked about the logs during this time.

aferraz, 2017-12-12:
Thanks @davidelang, I checked the logs in console and did not disconnect from mqtt.
However, in the past I had tried the pulsetime and it worked as written in the wiki, but I did not use it because there was not yet the interlock function. 
I do not remember what version I tried but it worked on the same devices I use now.

davidelang, 2017-12-12:
are there any logs of anything happening during the timeframe? a pulsetime of 43 
should only be a delays of 4.3 seconds, if it's delaying for 23 seconds 
something is going wrong, please set debugging to 'more detail' and see if there 
is anything showing up.

aferraz, 2017-12-12:
@davidelang I think I understand where the problem is. I have set sleep to 250ms.
So, with a sleep 250 a pulsetime of 43 become about 20 seconds

stefanbode, 2017-12-13:
Why did you use a sleep? I Do Not See any benefit from using this. And yes, this is likely to cause the issue.

aferraz, 2017-12-13:
to reduce electricity consumption

closed 2017-12-17 by aferraz