arendst/Tasmota #1353: magic home v2.3 issues

tarbax, 2017-12-12:
my device is an RGBW with IR magic ligic home light version 2.3
configured to magic home
GPIO4 ->IRrecv
GPIO13 Led1i -> PWM4
i also did the ''command 15 1''

colors are not correct and i have no rgb control on the main webpage of tasmota only brightness
commands:
color 1 (gives no light   dedicated white are off)
color 2 (green light   dedicated white are off)
color 3 (red light   dedicated white are off)
color 4 (dimmed grey light   dedicated white are off)
color 12 ( green/yellow light   dedicated white are off)
color  000000ff ( blue light   dedicated white are off)
color FFFFFFFF (rgb are all on white   dedicated white are off)

how to fix this and how to add this to home assistant ?

arendst, 2017-12-13:
- Tasmota does not provide RGB light control on web page.
- Your Magic light uses different PWM channels than supported by Tasmota Magic Light module. This is expected as every day the Chinese provide new GPIO configurations. No problem as you can easily provide the GPIO configuration yourself using the Wemos D1 module.

So verify your results against the current Magic Light module config parameters in sonoff_template.h , select the wemos module and enter the correct PWM channel information.

In Tasmota PWM1 is Red, PWM2 is Green, PWM3 is Blue and PWM4 is White when an RGBW device is connected.

As an example your Green channel seems to be OK so select for GPIO5 option 34 PWM2. As ``color 3`` should be Blue but your receive Red you know that GPIO12 must be option 33 PWM1. Experiement with the other two PWMs.

Success.

As for Home Assistant...

tarbax, 2017-12-13:
Thanks arendst!
my Colors are correct now but now i have an other issue
i need to set these extra that i dont heve/need
RRGGBBCCWW  CC for color temp but dont need how to disable this
17:03:36 CMD: color 000000FF
17:03:36 MQT: stat/magichome/RESULT = {"Color":"FFFFFFFFFF"}

how to fix this ?

arendst, 2017-12-13:
It seems you configured one PWM channel too much. Verify you only have 4 PWM channels configured.

tarbax, 2017-12-13:
Thanks Arendst for the fast reply! 
everything is now working

robotdad, 2017-12-23:
@tarbax this is what I have for the home assistant configuration. I'm seeing some oddities with the RGB values, but monitoring the messages that is on the home assistant side.

In this example the Living room light is a Sonoff B1, the other is an AiLight.
```
light:
  - platform: mqtt
    name: 'Living room lamp'
    state_topic: "stat/lr_lamp/POWER"
    command_topic: "cmnd/lr_lamp/POWER"
    rgb_state_topic: "stat/lr_lamp/COLOR"
    rgb_command_topic: "cmnd/lr_lamp/COLOR"
    rgb_command_template: "{{ '%02x%02x%02x0000' | format(red, green, blue) }}"
    brightness_state_topic: "stat/lr_lamp/DIMMER"
    brightness_value_template: "{{ value_json.brightness }}"
    brightness_command_topic: "cmnd/lr_lamp/DIMMER"
    brightness_scale: 100
    color_temp_command_topic: "cmnd/lr_lamp/CT"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true
  - platform: mqtt
    name: 'Family room lamp'
    state_topic: "stat/fr_lamp/POWER"
    command_topic: "cmnd/fr_lamp/POWER"
    brightness_state_topic: "stat/fr_lamp/DIMMER"
    brightness_command_topic: "cmnd/fr_lamp/DIMMER"
    brightness_value_template: "{{ value_json.brightness }}"
    brightness_scale: 100
    rgb_state_topic: "stat/fr_lamp/COLOR"
    rgb_command_topic: "cmnd/fr_lamp/COLOR"
    rgb_command_template: "{{ '%02x%02x%02x00' | format(red, green, blue) }}"
    qos: 1
    payload_on: "ON"
    payload_off: "OFF"
    retain: true
```

CZ1979, 2017-12-30:
@tarbax : Would you provide your configuration changes? I've got the same problem with my mgichome Device. Maybe your config would help me? 
Would be great! Thanks ;)

tarbax, 2018-01-03:
@CZ1979 i still did not figured it out now my rgb chart is also off and with every restart is different

Alveran81, 2018-01-06:
How do you flash Board Version 2.3. My is not booting after flashing.

diijkstra, 2018-01-19:
Mine v2.3 board flashed just fine using method for "Variant without Rx/Tx pads".

Here is working configuration for outputs on v2.3 board (works fine as Wemos):
- PWM1 -> GPIO12 (Red)
- PWM2 -> GPIO05 (Green)
- PWM3 -> GPIO13 (Blue)
- PWM4 -> GPIO15 (White)

I have a version without IR and IR Data pad is not connected to anything. It looks like to be missing resistors R9/R8 on bottom side of PCB. Probably this pad is connected via R9 to GPIO04 and pulled down to ground via R8.

I did not checked color codes because I have only white leds strips - anyway, it should be working.

Alveran81, 2018-01-20:
What do you mean with "Variant without Rx/Tx pads"
I flashed it like my other sonoff. Same settings in Atom but it doesn't work.

diijkstra, 2018-01-20:
I meant I've soldered cables directly to ESP module (GND, GPIO-0, rx, tx, see picture on wiki). It took couple of tries until successful flash. I think, I had to disconnect GPIO-0 from GND shortly after applying 12V power and it flashed fine.

Alveran81, 2018-01-27:
It flashes fine without error but afer flashing i got only this in log

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_9546E5, GroupTopic sonoffs) Version 5.11.1g
�␀�␀␀rd␀l��|␀�#␂�o␄␌␄�␄l�␌#|��␃�␓��|␛bl␄b��on�loN���␌B␜p�c$`␃rlp�o�␐␂␄␌�␌$␌��␄␌␄b␄n�|␃l�␌␄�p␄��no�␃d��␀l`␂�␒␛Nn␄l ␂␏␂n{���N␌␄#␄�␏l␏rےn␄␌"␌�␏l܂␌$␌$���{�`␃��n�␂�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_9546E5, GroupTopic sonoffs) Version 5.11.1g
�␀␀␀r$␀l���␂�#␂�o␌␌␌�␄l�␌B|��␃�␛��|␛cl␄c��on�lon���␌B␜p�c$`␃rlp�o�␐␃␄␌�␌$␌��␄␌␌"␌N�|␂l�␌␄�p␄��No�␂l��␀$`␂�␒␓on␌$`␂␎␃os���o␌␄b␌�␎$␎{��o␌␄c␌�␎$��␄l␄l�␜␏r� ␂��o�␃�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_9546E5, GroupTopic sonoffs) Version 5.11.1g
�␀␀␀rl␀l��|␀�l�|␃␄␌␄�␜�l�␄c|��␂�␒�r�Bl␌"��oN�␀$no���␄c␜p�l{l{lp�n�␐␃␌␄�␄l␄��␌␄␌#␌o�|␂$�␄␌�b��on�␂l��l ␃�␛␒no␌d`␃␏␂or�ےn␄␌"␌�␏l␎s���␂␄␌b␌�␎l܃␄l␌l���{�`␂��n�␂�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_9546E5, GroupTopic sonoffs) Version 5.11.1g

tarbax, 2018-02-04:
how can i have a dedicated slider for the white channel ?

johny-mnemonic, 2018-02-06:
I seem to have RGBW module version 2.2, which has same GPIO config for PWM channels as yours v2.3
It is also "Variant without Rx/Tx pads" as seen on attached image:
[image]
I am struggling to find which GPIOs are usable for connecting something else to the board though.
I tried to configure GPIO2 and 4 as buttons in Tasmota, but connecting these pins to GND does not produce anything on console (like MQTT message of pushed button). Am I doing it correctly?

I would like to connect two push buttons to it and in case there will still be another GPIO free, I would like to connect DHT22 as well.

markcarline, 2018-06-02:
Sorry I know this is going of topic but the example above from robot daddy is the only Tasmota/ha/Sonoff B1 & ailight I can find, would be a great addition to the help page:
https://github.com/arendst/Sonoff-Tasmota/wiki/Home-Assistant

ascillato, 2018-06-08:
@markcarline, Just added that to the wiki. Thanks :+1:

Stefan300381, 2018-08-20:
@diijkstra 

I flashed my MagicHome controller and colors were wrong. Now I set as module "generic" and assigned your values. Unfortunately still wrong (btw: I have RGB only, not RGBW). Can you point me into the right direction how to find out which PWM I need to assign to which GPIO? When I assign just PWM1 to one of the GPIOs then I see:

PWM1 -> GPIO05 = red
PWM1 -> GPIO12 = green
PWM1 -> GPIO13 = blue

But what do I need to do with PWM2 and 3? I don't even understand how PWM works ;-)

Any help is highly appreciated.
Stefan

johny-mnemonic, 2018-08-21:
@Stefan300381 Don't bother with how PWM works (that's just short for Phase Width Modulation - the way, how LED dimming actually works).
The only important thing for you is to try to assign three PWM outputs (as you only have RGB) to GPIOs.

So first try the same that @diijkstra did:
PWM1 -> GPIO12 (Red)
PWM2 -> GPIO05 (Green)
PWM3 -> GPIO13 (Blue)

If colors are messed, try different combinations of GPIOs, until it works for you.

Stefan300381, 2018-08-21:
Thanks. So it's not important if PWM1 or PWM2 is connected to GPIO"X" ?

niklaskr, 2018-10-25:
Mine works perfectly fine using premade device settings 37 Arilux LC01.
Found it because it uses same esp chip.