arendst/Tasmota #1354: mosquito and mqtt timeout problems

atlaware, 2017-12-13:
Hi,
first: thanks to all developpers for this good firmare!
second...: I have one problem with mqtt and mosquitto.
Every X minutes, (1 > X <10) sonoff go in timeout and disconnect from mosquitto.
Isn't a network problem, pinf -f sonoff doesn't have packet lost.

I have test with tasmota 5.9.0i and 5.10.0 and with mosquitto 1.4.14 and mosquitto-git.

Some one have some suggestion to resolv this?
It's a problem because when sonoff disconnect and reconnect MQTT doesn't work, and pubblication on broker is unheeded.

Log on sonoff:
`13:37:51 MQT: Attempting connection...
13:37:51 MQT: Connected`
OR
`13:37:34 MQT: Attempting connection...
13:37:40 MQT: Connect failed to 192.168.6.99:1122, rc -2. Retry in 10 sec`

Log on mosquitto:
`1513167940: Client SONOFF3D has exceeded timeout, disconnecting.
1513167940: Socket error on client SONOFF3D, disconnecting.
1513167946: New connection from 192.168.6.222 on port 1122.
1513167946: New client connected from 192.168.6.222 as SONOFF3D (c1, k15, u'sonoff3d').
1513168075: Client SONOFF3D has exceeded timeout, disconnecting.
1513168075: Socket error on client SONOFF3D, disconnecting.`

MatzeJoerling, 2017-12-13:
bad wifi? it is one thing to get 64byte ICMP ping packets to server, establishing a tcp connection is much more. have you tried ping with 1k packets (linux ping -s 1000 target) and faster than 1 pkg per sec ? what about a tcpdump and check it with wireshark?

atlaware, 2017-12-14:
So, ~5 minutes 1k ping flood (ping -f -s 1000 ip) to 2 differents sonoff base model, connected to 2 differents AP on the same wifi network bridged: (ping executed from mosquitto server)

42030 packets transmitted, 42028 received, 0% packet loss, time 274234ms
rtt min/avg/max/mdev = 1.622/6.787/100.340/5.850 ms, pipe 6, ipg/ewma 6.524/2.964 ms
96586 packets transmitted, 96585 received, 0% packet loss, time 283684ms
rtt min/avg/max/mdev = 1.274/2.704/55.213/1.678 ms, pipe 4, ipg/ewma 2.937/2.741 ms

I don't think that is a wifi problem.

The very strange things that I have noted is that when the ping flood was in execution no timeout disconnecting! In the moment that I stop the ping, disconnecting restart every 1/5 minutes :(

I understand that this is a very strange problems, but if someone have some suggestions... Or, someone can tell me that don't have any problems with mosquitto?

Tnx!

lobradov, 2017-12-14:
When you execute ```status 11``` on Tasmota console, what's the value of RSSI that you get? 

Can you check your Wifi router/ AP and see if you can find something in its logs - I had a similar issue with one of the old NetGear routers that simply wasn't compatible with ESP wifi... they could not agree on the radio parameters and esp was getting disconnected after few minutes of inactivity. It's a similar, but not the same as what you described - my setup wouldn't reconnect unless I power down esp, let it cool off for 10min or so, and then try again... I guess router was remembering something about client retries... I didn't play much with low-level radio params (can't say I understand them) but maybe there's something incompatible there... 

Anyway, good luck and let us know what you found - would be good to capture it in Wiki for future reference ;)

atlaware, 2017-12-14:
Hi, 
RSSI is good on one sonoff (very near to the wifi AP) from 90 to 100, in the other sonoff is 50/60.
The wifi routes have chip: Atheros AR9344 + AR9582 and AR9341. (tplink wdr3600 and tplink wr841)

But I don't think that is the same problems you descrives, I don't have to restart router or sonoff.

When I disable MQTT the http connection to sonoff is always very good and reactive.

When I enable MQTT, also the http sonoff interface go randomly slowly/not respond. Maybe the problems with the MQTT comunication slow the esp "loop".
But don't is so slowly as when you have a wrong MQTT server settings.

Thank you for your attention!
I try to investigate more...

lobradov, 2017-12-14:
Slow down of Tasmota while MQTT is retrying a connection is a known issue, caused by some of the MQTT library to still be synchronous (@arendst can elaborate more, I'm not sure I get all the details).

What's your Mosquitto config (authentication especially)? 
What's your Tasmota MQTT config (hostname, IP address or mdns name)?
Firewall maybe? :)

atlaware, 2017-12-14:
> Slow down of Tasmota while MQTT is retrying a connection is a known issue, caused by some of the MQTT library to still be synchronous (@arendst can elaborate more, I'm not sure I get all the details).

Yes, maybe i notice slowdown when tasmota lost mosquitto connection and reconnect, that is the problem (the continous reconnection) :(

> What's your Mosquitto config (authentication especially)?
> What's your Tasmota MQTT config (hostname, IP address or mdns name)?
> Firewall maybe? :)

So, authentication is with password file (password_file /etc/mosquitto/passwd) and work with other MQTT client (octoprint and mqtt dashboard (android)). No ssl or tls connection, plain;
Tasmota MQTT is set without mdns (that I have also tried to remove from user_config.h), so directly settings in tasmota with IP address and port;
No firewall problems, all the devices involved is in the same network subnet without firewall.

Tnx

atlaware, 2017-12-18:
Ok, after an "illumination" :) and I try with the last 5.10.0b that doesn't resolv... I think that my problem was related to my network high-availability configuration, with ucarp.
Long to explain, but the result is that my gateway has an ip (192.168.6.100) and the mosquitto server another ip (192.168.6.99) with the same MAC address.
Maybe two ip with the same mac address give some problem to the, I think tiny, wireless esp lib!
I have set the same IP on the MQTT server configuration as gateway IP and never one disconnection.

Urra :)

davidelang, 2017-12-18:
yes, two IP addresses with the same MAC address can be a problem, but only if 
they are not on the same machine.

atlaware, 2017-12-18:
`yes, two IP addresses with the same MAC address can be a problem, but only if
they are not on the same machine.`

Yes, is on the same machine and also on the same ethernet card and on a single fisical connection. Maybe the wireless lib have difficult to elaborate this, I don't know.
But before the ip change, the connection to mqtt server go up and down continously (only from sonota), other device on the same network doesn't have problems with mosquitto.
Now that I have tell to sonota tu use the same IP, no disconnection at all.

ptibubu, 2018-01-28:
I get the exact same problem, and same log that  atlaware decribed, on all my tasmota firmwares (5 sonoff 16th, 1 sonoff 4ch pro and 1 wemos d1).
No issue in the mosquitto log when using other mqtt clients (such as mqtt.fx, or home assistant).

Tried mosquitto1.3.?? and 1.4.14, and tasmota 5.11.1. 
The tasmota seems to succeeds in receiving mqtt commands and sending status. But is keeps disconnecting.
No strange IP/MAC configuration.
Any hint would be welcome.

savage872, 2018-02-02:
I'm getting 10-14% packet loss on both my sonoff basic and I'm not to sure what the problem is but is really annoying and also the tasmota firmware crashes often and I'm starting to dislike the whole sonoff-tasmota idea :(

jandrioli, 2018-03-06:
This is not related to WiFi signal quality or anything alike. 

My setup (all devices are on the same WiFi router):
- mosquitto v1.3.4 on orangepi 
- mosquitto v1.4.8 on laptop
- mqtt client on ESP32
- mqtt client on 8266 ESP-12E
- mqtt client on 8266 ESP-12F

ALL clients get these timeout and socket errors when they are connecting to the mosquitto server on the laptop. Files /etc/mosquitto/mosquitto.conf are identical for both servers.

savage872, 2018-03-06:
Since my last post I learned that the timeouts where because the ESP couldn't handle to many UDP requests when it had Emulation enabled. Hope it helps!