arendst/Tasmota #1355: No good wiki to setup sonoff-tasmota on a sonoff rf bridge 433

alcomys, 2017-12-13:
Can someone help me please ?
i have a recent version of the sonoff rf bridge 433.
on the wiki there is less information about how to pair and how to setup to domoticz.
i search google where i can but can not find specific info to my setup.

so my setup is : 
Domoticz on a raspberry pi with an mqtt broker server and client.
i have flashed a sonoff basic and that is working out of the box.
now i have received the sonoff rf bridge 433 (new version) (esp8285)
i have flashed the sonoff rf bridge also with the sonoff tasmota.
but now im stuck at settings things up with domoticz.
i have 16 buttons on my websetup of the rf bridge but how can i learn devices and make an connection to domoticz ?

Beste regards,

bleisoldat, 2017-12-18:
1. go to console
2. command "rfkey2 2" for LearningMode for rfkey2
3. press the key on your remote to learn. the rfbridge will beep, when ready

tukestolo, 2018-01-05:
Friend do you understand what need to do ?
I cannot do it to, I have the same problem. I can make my bridge learn the command, but I cannot connect it to Domoticz.

Can some one help ?

alcomys, 2018-01-06:
my bridge i now in the storage box because i can't find an answer

tukestolo, 2018-01-06:
:( I tried everything to, and I cannot figure it out...

tukestolo, 2018-01-06:
Friend I find a way that we can use it...
On Domoticz you can put a url in, state on and another one in state off... that url you can put the device that you what to turn on or off.

closed 2018-05-21 by stale[bot]