arendst/Tasmota #1359: How touch works of sonoff touch

sonukushwah221, 2017-12-14:
Hello 

I want to know how touch works with this wifi please help me to understand

ionciubotaru, 2017-12-14:
Sonoff touch has a separate IC for touch detection. It is connected to esp8266 to GPIO 0

sonukushwah221, 2017-12-14:
Thank you. ionciubotaru do you know anything about that IC because I am using at42qt2120 for that I am finding a chip solution this ic is very costly in India ,Do you have any schematic diagram of that complete circuit

ionciubotaru, 2017-12-14:
Sorry, i don't know anything about that IC.
If you look for a cheap solution - there are touch libraries for arduino, so you can detect touch with esp8266 only - https://github.com/PaulStoffregen/CapacitiveSensor

webtop, 2017-12-14:
On my version of touch, from a different manufacturer, they use a HD6001 Capacitive Touch IC, from Hab. Datasheet here: http://www.hab.com.tw/upload/2013879261133084.pdf