arendst/Tasmota #1365: Any possibility for ACS712 support?

martin072, 2017-12-14:
I noticed a very small and cheap current sensor, ACS712 that would possibly fit in the Sonoff modules. 
I believe it works with an Analoge Out.. Is there any option to support this in the Tasmota?

ionciubotaru, 2017-12-15:
Analog support is already implemented in Tasomota
comment  the following line in user_config.h

`#define USE_ADC_VCC                              // Display Vcc in Power status. Disable for use as Analog input on selected devices`

martin072, 2017-12-15:
Ok, but which command would readout the A0 pin? I Only see a reference in the code to a 'Witty cloud' device.. Mine is a regular Sonoff, and I would just like to use the A0 pin..

ionciubotaru, 2017-12-15:
read this and configure your module
https://github.com/arendst/Sonoff-Tasmota/wiki/Adding-new-Module-types

martin072, 2017-12-15:
Thanks ionciubotaru, still struggling a bit how to work this out.. I can see the ADC0 is disabled in mytmplt, so I can set it to GPIO_USER and I will assume it will popup in the webserver to configure.  But from there it ends.. Since none of the sensors in the list are related to ADC read out..

ionciubotaru, 2017-12-15:
select `wemod d1 mini`  it has ADC gipo selectable - you will find the adc values in tele messages 

  { "WeMos D1 mini",   // WeMos and NodeMCU hardware (ESP8266)
     GPIO_USER,        // GPIO00 D3 Wemos Button Shield
     GPIO_USER,        // GPIO01 TX Serial RXD
     GPIO_USER,        // GPIO02 D4 Wemos DHT Shield
     GPIO_USER,        // GPIO03 RX Serial TXD and Optional sensor
     GPIO_USER,        // GPIO04 D2 Wemos I2C SDA
     GPIO_USER,        // GPIO05 D1 Wemos I2C SCL / Wemos Relay Shield (0 = Off, 1 = On) / Wemos WS2812B RGB led Shield
     0, 0, 0, 0, 0, 0, // Flash connection
     GPIO_USER,        // GPIO12 D6
     GPIO_USER,        // GPIO13 D7
     GPIO_USER,        // GPIO14 D5
     GPIO_USER,        // GPIO15 D8
     GPIO_USER,        // GPIO16 D0 Wemos Wake
     GPIO_ADC0         // ADC0   A0 Analog input
  },

martin072, 2017-12-15:
Great, got it now, thanks!

closed 2017-12-15 by martin072

effelle, 2018-09-04:
Sorry to open again this thread, but is still a little confusing how to enable GPIO_ADC0 on generic module (old wemos one).
On sonoff_template.h I have:

> { "Generic",         // Any ESP8266/ESP8285 device like WeMos and NodeMCU hardware (ESP8266)
>      GPIO_USER,        // GPIO00 D3 Wemos Button Shield
>      GPIO_USER,        // GPIO01 TX Serial RXD
>      GPIO_USER,        // GPIO02 D4 Wemos DHT Shield
>      GPIO_USER,        // GPIO03 RX Serial TXD and Optional sensor
>      GPIO_USER,        // GPIO04 D2 Wemos I2C SDA
>      GPIO_USER,        // GPIO05 D1 Wemos I2C SCL / Wemos Relay Shield (0 = Off, 1 = On) / Wemos WS2812B RGB led Shield
>      0, 0, 0,          // Flash connection
>      GPIO_USER,        // Flash connection or GPIO09 on ESP8285 only!
>      GPIO_USER,        // Flash connection or GPIO10 on ESP8285 only!
>      0,                // Flash connection
>      GPIO_USER,        // GPIO12 D6
>      GPIO_USER,        // GPIO13 D7
>      GPIO_USER,        // GPIO14 D5
>      GPIO_USER,        // GPIO15 D8
>      GPIO_USER,        // GPIO16 D0 Wemos Wake
>      **GPIO_ADC0         // ADC0   A0 Analog** 

I have an ACS712 too and I'm trying to configure it on one of my Amica ESP-12E that has the A0 gpio, and I can't have it configured at all, since ADC0 pin doesn't show up on settings webpage. 
I' ve tryed with or without `define USE_ADC_VCC` (user_config.h) but with no luck. I wonder if I can use another type of module other than the generic one to accomplish what I need or if is simply an incompatible sensor.
Thanks.

ascillato, 2018-09-04:
You need to compile tasmota with #define USE_ADC_VCC commented out (file user_config.h)

effelle, 2018-09-04:
Is already commented out, with the ´#´ in front.

effelle, 2018-09-04:
The the correct way is `// #define USE_ADC_VCC`? 
Ok. Compiling right now.

effelle, 2018-09-04:
Still not working. Or at least is not here where I can find A0.
[image]

arendst, 2018-09-04:
As there is nothing to configure on a ADC channel you won't see it in the configuration webpage.

Just connect your analog signal and verify it's values on the main webpage.

effelle, 2018-09-04:
@arendst : never mind. After testing my acs712 under another sketch I realize is dead. 
Time to buy a pzem, I think is much more efficient e reliable.

effelle, 2018-09-14:
@arendst 
Sorry to bother you again but even with another  sensor (this time an LDR) I can't see any analog value on main webpage nor on tele.
On another sketch I tested if it's working:
```
void setup() {
  Serial.begin(9600);
}

void loop() {
  // read the input on analog pin 0:
  int sensorValue = analogRead(A0);
  
  // Convert the analog reading (which goes from 0 - 1023) to a voltage (0 - 5V):
  float voltage = sensorValue * (5.0 / 1023.0);
  
  // print out the value you read:
  Serial.println(voltage);
}
```
And all is fine, I got reading from 0.1 to 3.2.
On Tasmota (6.2.0) I've commented the line on `user_config.h` as suggested:
```
// -- Internal Analog input -----------------------
//#define USE_ADC_VCC                              // Display Vcc in Power status. Disable for use as Analog input on selected devices
```
and I've upload the sketch to my unit, configured as `generic module` since I'm using an AMICA NodeMCU V3 (ESP-12E), which has analog `A0` enabled.
On my main webpage I can see just my DS18B20, but nothing about the analog. 
[image]
Here's my ` status 0` command:
```
01:55:22 CMD: status 0
01:55:22 MQT: stat/sonoff2/STATUS = {"Status":{"Module":18,"FriendlyName":["Sonoff","Sonoff2"],"Topic":"sonoff2","ButtonTopic":"0","Power":0,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
01:55:22 MQT: stat/sonoff2/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://sonoff.maddox.co.uk/tasmota/sonoff.bin","RestartReason":"Power on","Uptime":"0T00:11:24","StartupUTC":"2018-09-14T00:43:58","Sleep":0,"BootCount":106,"SaveCount":545,"SaveAddress":"F7000"}}
01:55:22 MQT: stat/sonoff2/STATUS2 = {"StatusFWR":{"Version":"6.2.0","BuildDateTime":"2018-09-12T21:29:14","Boot":31,"Core":"2_4_2","SDK":"2.2.1(cfd48f3)"}}
01:55:22 MQT: stat/sonoff2/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":3,"LogHost":"","LogPort":514,"SSId":["EffE-Domus","EffE-WiFi-Sala"],"TelePeriod":300,"SetOption":["00008009","55818000","00000000"]}}
01:55:22 MQT: stat/sonoff2/STATUS4 = {"StatusMEM":{"ProgramSize":513,"Free":488,"Heap":15,"ProgramFlashSize":1024,"FlashSize":4096,"FlashMode":3,"Features":["00000809","0FDEE794","00000004","37FFBFCE","00000000"]}}
01:55:22 MQT: stat/sonoff2/STATUS5 = {"StatusNET":{"Hostname":"sonoff2-4272","IPAddress":"192.168.15.104","Gateway":"192.168.15.1","Subnetmask":"255.255.255.0","DNSServer":"192.168.15.100","Mac":"80:7D:3A:23:50:B0","Webserver":2,"WifiConfig":5}}
01:55:22 MQT: stat/sonoff2/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.15.102","MqttPort":1883,"MqttClientMask":"DVES_%06X","MqttClient":"DVES_2350B0","MqttUser":"admin","MqttType":1,"MAX_PACKET_SIZE":1000,"KEEPALIVE":15}}
01:55:22 MQT: stat/sonoff2/STATUS7 = {"StatusTIM":{"UTC":"Fri Sep 14 00:55:22 2018","Local":"Fri Sep 14 01:55:22 2018","StartDST":"Sun Mar 25 02:00:00 2018","EndDST":"Sun Oct 28 03:00:00 2018","Timezone":1,"Sunrise":"06:25","Sunset":"19:06"}}
01:55:22 MQT: stat/sonoff2/STATUS10 = {"StatusSNS":{"Time":"2018-09-14T01:55:22","Switch1":"ON","Switch2":"ON","DS18B20":{"Temperature":23.1},"TempUnit":"C"}}
01:55:22 MQT: stat/sonoff2/STATUS11 = {"StatusSTS":{"Time":"2018-09-14T01:55:22","Uptime":"0T00:11:24","Vcc":3.166,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"EffE-Domus","RSSI":54,"APMac":"A4:33:D7:3C:11:DC"}}}
```
On status 11 I can see a Vcc value, but it change slightly even in dark (minimum was 2.770) then I think is relative to the instant load of the board not the analog. But I may be wrong, obviously.
On `status 8` I can see just my digital sensor:
```
02:04:08 CMD: status  8
02:04:08 MQT: stat/sonoff2/STATUS8 = {"StatusSNS":{"Time":"2018-09-14T02:04:08","Switch1":"ON","Switch2":"ON","DS18B20":{"Temperature":23.1},"TempUnit":"C"}}
```
I do apologize if I'm hammering this thread in that way, but I'm just trying to understand how this works (newbe here), and I didn't find much hint on wiki. 
Since there is the possibility to use an analog device (or more?) I think someone with a good english should write a couple of line about, but not me, was already hard write down a post here in a foreign language.

martinezmp3, 2019-08-05:
did you ever got this working im in the same spot now

knopserl, 2020-01-04:
I have also the same use case where I wanted to sens the current with a ACS712. I have configured the device to use analog read and also to use analog values as CT (power monitor).
I use a ACS712/20A which igives 100mV per 1A. Since I use 230V AC and the ESP8266 modules have a 0-1V range, I have set it up with the trip-pot to 500mV (from the 2,5V output with no current).
So if there is a load on 230V AC, the analog input pulses aroun 500mV up and down. E.g. at 1A = 230Watt it has 400 to 600mV.
But tasmota shows nonenes values between 300 to 900Watt, when there is no load (should be 0).
I have set up the analog voltage with the trim-pot to 0mV, then it shows 0 A and 0Watt but that is wrong. 500mV = 512 digital value (from 1024 as max if 1V is reached = 20A).
I have attached a screen-shot.
Any ideas? I have a test program which works with the ESP8266 + ACS712. I will verify if the HW is working (I have tested it with anaother ESP8266 and it shows proper values at half voltage on A0 which is 0 Amp).
[image]

meingraham, 2020-01-04:
@knopserl 

A Tasmota [PR](https://github.com/arendst/Tasmota/pull/7100) was recently merged for using a current sensing device (e.g., a current transformer) on ADCO. The pin on the ESP82xx expects 1V however many devices provide a contact on the device for A0 which has a voltage divider built-in and expects voltages in the range on 0v-3.3v.

As this is a very old (ancient) issue, please refrain from posting as its not likely to get a lot of attention. In any case, Tasmota GitHub issues are intended for bugs and feature requests. Your question, at least at this time, likely falls into the category of configuration or setup question. 
Please address any further questions to the [Tasmota Support Discord Chat](https://discord.gg/Ks2Kzd4). The chat is a better and more dynamic channel for helping you. Github issues are best used for Tasmota **_software feature requests and bug reporting_**. Troubleshooting and setup assistance is more effective using an interactive forum.

Please check the [Contributing Guideline and Policy](https://github.com/arendst/Sonoff-Tasmota/blob/development/CONTRIBUTING.md) and the [Support Guide](https://github.com/arendst/Sonoff-Tasmota/blob/development/SUPPORT.md).

Thanks.

### Support Information
See [Wiki](https://tasmota.github.io/docs/#/Home) for more information.
See [Chat](https://discord.gg/Ks2Kzd4) for more user experience.
See [Community](https://groups.google.com/d/forum/sonoffusers) for forum.
See [Code of Conduct](https://github.com/arendst/Sonoff-Tasmota/blob/development/CODE_OF_CONDUCT.md)

knopserl, 2020-01-04:
@meingraham thanks for your fast response. I'll follow your recommendation.
Though, I thought it's either a feature request or bug and less a configuartion issue. At least I have not found any more configuration options than I have chose (analog feature switched on and CT option for power meter). And in the analog reading source I found a comment which points to 230V AC and a 20A/1V setting, which I found it fit't well to my setup.

mochoandre, 2020-11-21:
meingraham 

FY!!!!