arendst/Tasmota #1366: Unable to get device to connect to wifi

johntdyer, 2017-12-14:
S20 Sonoff ( Firmware 5.10.0a)

Serial Logs:

```logs
> Executing task: platformio device monitor <

--- Miniterm on /dev/cu.usbserial-00000000  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
�����rrR�00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Attempting connection...
00:00:12 WIF: Attempting connection...
00:00:12 CMD: reset 2
00:00:12 RSL: Group 0, Index 1, Command RESET, Data 2
00:00:12 RSL: stat/sonoff3/RESULT = {"Reset":"Erase, Reset and Restarting"}
00:00:13 APP: Erase 136 sectors
00:00:13 CFG: Use defaults
00:00:19 CFG: Saved to flash at FB, Count 1, Bytes 1648
00:00:19 WIF: Attempting connection...
00:00:20 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,1)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Boot Count 1
00:00:00 Project sonoff3 Sonoff (Topic sonoff3, Fallback DVES_07AA1D, GroupTopic sonoffs) Version 5.10.0a
00:00:00 CFG: Saved to flash at FA, Count 2, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 dyer-iot in mode 11N as sonoff3-2589...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Attempting connection...
00:00:12 WIF: Attempting connection...
00:00:13 WIF: Attempting connection...
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Patch issue 2186
00:00:15 WIF: Connecting to AP2 dyer in mode 11N as sonoff3-2589...
00:00:16 CFG: Saved to flash at F9, Count 3, Bytes 1648
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Attempting connection...
00:00:18 WIF: Attempting connection...
00:00:19 WIF: Attempting connection...
00:00:20 WIF: Attempting connection...
00:00:21 WIF: Attempting connection...
00:00:22 WIF: Attempting connection...
00:00:23 WIF: Attempting connection...
00:00:24 WIF: Attempting connection...
[... 489 lines cut ...]
00:00:15 WIF: Patch issue 2186
00:00:15 WIF: Connecting to AP2 dyer in mode 11N as sonoff3-2589...
00:00:16 CFG: Saved to flash at F5, Count 23, Bytes 1648
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Attempting connection...
00:00:18 WIF: Attempting connection...
00:00:19 WIF: Attempting connection...
00:00:20 WIF: Attempting connection...
00:00:21 WIF: Attempting connection...
00:00:22 WIF: Attempting connection...
00:00:23 WIF: Attempting connection...
00:00:24 WIF: Attempting connection...
00:00:25 WIF: Attempting connection...
00:00:26 WIF: Attempting connection...
00:00:27 WIF: Attempting connection...
00:00:28 WIF: Attempting connection...
00:00:29 WIF: Attempting connection...
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WPSConfigfailed to start
00:00:35 APP: Button 1 multi-press 1

--- exit ---
```

I jhave tried re-flashing many times but it doesnt seem to help. Appreciate any suggestions

johntdyer, 2017-12-14:
I tried a "reset 1" and it seemed to try a few more times to connect and then was able to fully connect...

```logs
> Executing task: platformio device monitor <

--- Miniterm on /dev/cu.usbserial-00000000  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀␀00:00:18 WIF: Attempting connection...
00:00:19 WIF: Attempting connection...
00:00:20 WIF: Attempting connection...
00:00:21 WIF: Attempting connection...
00:00:22 WIF: Attempting connection...
00:00:23 WIF: Attempting connection...
00:00:24 WIF: Attempting connection...
00:00:25 WIF: Attempting connection...
00:00:26 WIF: Attempting connection...
00:00:27 WIF: Attempting connection...
00:00:28 CMD:
00:00:28 RSL: Group 0, Index 1, Command , Data
00:00:28 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:28 WIF: Attempting connection...
00:00:28 CMD:
00:00:28 RSL: Group 0, Index 1, Command , Data
00:00:28 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:29 CMD:
00:00:29 RSL: Group 0, Index 1, Command , Data
00:00:29 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:29 CMD:
00:00:29 RSL: Group 0, Index 1, Command , Data
00:00:29 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:29 CMD:
00:00:29 RSL: Group 0, Index 1, Command , Data
00:00:29 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:29 WIF: Attempting connection...
00:00:30 CMD: help
00:00:30 RSL: Group 0, Index 1, Command HELP, Data
00:00:30 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WPSConfigfailed to start
00:00:35 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,3)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 APP: Boot Count 13
00:00:00 Project sonoff3 Sonoff (Topic sonoff3, Fallback DVES_07AA1D, GroupTopic sonoffs) Version 5.10.0a
00:00:00 CMD: re
00:00:00 RSL: Group 0, Index 1, Command RE, Data
00:00:00 RSL: stat/sonoff3/RESULT = {"Command":"Unknown"}
00:00:00 CFG: Saved to flash at FA, Count 26, Bytes 1648
00:00:00 WIF: Attempting connection...
00:00:00 WIF: Patch issue 2186
00:00:00 WIF: Connecting to AP1 dyer-iot in mode 11N as sonoff3-2589...
00:00:01 WIF: Attempting connection...
00:00:02 WIF: Attempting connection...
00:00:03 WIF: Attempting connection...
00:00:04 WIF: Attempting connection...
00:00:05 WIF: Attempting connection...
00:00:06 WIF: Attempting connection...
00:00:07 WIF: Attempting connection...
00:00:08 WIF: Attempting connection...
00:00:09 WIF: Attempting connection...
00:00:10 WIF: Attempting connection...
00:00:11 WIF: Attempting connection...
[... 414 lines cut ...]
00:00:14 WIF: Attempting connection...
00:00:15 WIF: Connect failed with AP timeout
00:00:15 WIF: Patch issue 2186
00:00:15 WIF: Connecting to AP1 dyer-iot in mode 11N as sonoff3-2589...
00:00:16 CFG: Saved to flash at F8, Count 20, Bytes 1648
00:00:16 WIF: Attempting connection...
00:00:17 WIF: Attempting connection...
00:00:18 WIF: Attempting connection...
00:00:19 WIF: Attempting connection...
00:00:20 WIF: Attempting connection...
00:00:21 WIF: Attempting connection...
00:00:22 WIF: Attempting connection...
00:00:23 WIF: Attempting connection...
00:00:24 WIF: Attempting connection...
00:00:25 WIF: Attempting connection...
00:00:26 WIF: Attempting connection...
00:00:27 WIF: Attempting connection...
00:00:28 WIF: Attempting connection...
00:00:29 WIF: Attempting connection...
00:00:30 WIF: Connect failed with AP timeout
00:00:30 WIF: WPSConfigfailed to start
00:00:35 APP: Restarting
```

Any idea why this seems so "temperamental" ?

johntdyer, 2017-12-15:
Argh, I have tried 5 Sonoff basics and I am getting the same behavior from each... Very frustrating..

reloxx13, 2017-12-15:
Did you enter your wifi config in the user_config.h before ?
=> than did you change the config holder in the top section of user_config.h?

Did u connect to the tasmota wifi manager and enter your wifi config?
=> 4 Short Press on button for wifi manager, see here: https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage
try a few times till the sonoff wifi shows up in your wifi list.

johntdyer, 2017-12-15:
@reloxx13  - Thanks for the response... so I am going to answer your questions below, inline

> Did you enter your wifi config in the user_config.h before ?

Yes, I put my wifi config in user_config.h before flashing.   

----

> => than did you change the config holder in the top section of user_config.h?

This part has never been entirely clear to me... I have this right now

```
#define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration parameters
```

and what would you have me change this to?  Also when is it necessary to change this value ? 

----

> Did u connect to the tasmota wifi manager and enter your wifi config?

I was under the impression that this was unnecessary when puting the credentials in **user_config.h**.  

> => 4 Short Press on button for wifi manager, see here: arendst/Sonoff-Tasmota/wiki/Button-usage

----

Thanks again
-John

stefanbode, 2017-12-15:
Changing the config_holder does not help because he already executes a "reset 2". This securely wipes out any strange config for sure.
Following scenarious are likely the issue; please check:

- Password is wrong or include some non ACSII characters (could be an issue)
- MAC Filter defined on the router
- IP-Address conflict (e.g. you assigned to the device a fixed IP and now already another device is using the IP (my best guess to be the problem). Please change back to dynamically assign in your router.

johntdyer, 2017-12-15:
@stefanbode  - 

>Password is wrong or include some non ACSII characters (could be an issue)

The SSID I setup for testing is  **dyer-iot**. I also specifically used a simplistic password that has no special characters ( **dyeriotsecret** ).

>MAC Filter defined on the router

There is no mac filter on the router.

>IP-Address conflict (e.g. you assigned to the device a fixed IP and now already another device is using the IP (my best guess to be the problem). Please change back to dynamically assign in your router.

I am using the default config, which AFAIK mean's it will do DHCP.  I have included the wifi section is below

```c
#define STA_SSID1              "dyer-iot"        // [Ssid1] Wifi SSID
#define STA_PASS1              "dyeriotsecret"   // [Password1] Wifi password
#define STA_SSID2              "indebuurt2"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "VnsqrtnrsddbrN"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect \
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```

stefanbode, 2017-12-15:
Please check on your router, if you have assigned an IP to the device. In the firmware you can, but with "reset 2" you're for sure back on DHCP

stefanbode, 2017-12-15:
If you go to WIFI_MANAGER as a failover configuration you should be able to connect to the AP of the ESP itself and discover for the Wifi networks the devices can see. Maybe this gives you more insight into the problem.

davidelang, 2017-12-15:
On Fri, 15 Dec 2017, John Dyer wrote:

> This part has never been entirely clear to me... I have this right now
>
> ```
> #define CFG_HOLDER             0x20161209        // [Reset 1] Change this value to load following default configuration [...]

When Tamota starts up, it looks in the existing config on the flash to see if 
the value of CFG_HOLDER is there. If it is, Tasmota assumes that the config 
there is valid and ignores whatever is compiled into user_config.h

If it's not there, it replaces the config in the flash with what's in 
user_config.h

so whenever you want to override whatever is in the existing config with what 
you have compiled into the image, you need to change the CFG_HOLDER value (or 
hold button0 for 4 seconds (if you haven't disabled it), or issue the reset 
command to tell it to replace the config with what was compiled into 
user_config.h)

David Lang

johntdyer, 2017-12-15:
@davidlang -  just increment the serial by 1 each time ?

reloxx13, 2017-12-15:
hey, yes, in- or decrement, doesnt matter. it just has te be not the same as before.

i usually just change the last number to  a diffrent number.

davidelang, 2017-12-15:
On Fri, 15 Dec 2017, John Dyer wrote:

> @davidlang -  just increment the serial by 1 each time ?

just making it different than before, it doesn't matter how it's different.

It looks like arendst is using the date there, and that's a good convention, but 
all that matters is that it's different.

johntdyer, 2017-12-17:
I think it might have something to do w/ the Meraki AP, though I have no idea what...  I tried having it connect to the AP on my iPhone ( Tethering mode ) and it worked first time..  I have my Meraki set to 2.4 and 5ghz, WPA2.... not really sure what the heck it could be...

PlumbumDN, 2018-01-04:
Hi @johntdyer did you find any solution on this problem? I have the exact same issues except, that my TH16 isn´t connecting to any AP, including AndoridAP and iOS-Hotspot...

#1461

niklasfink, 2018-01-23:
@johntdyer I also have a Meraki AP and was experiencing this problem. Before this happened, I set it to a fixed IP. Setting it back to dynamic solves the issue, however, it then get an IP address I might not want it to have. 
Did you already find a solution for this? Maybe by flashing a static IP instead of using DHCP?

niklasfink, 2018-01-23:
Strange thing is, that instead of getting the same IP after resetting it, it gets a different IP (even though it's recognized as the same device). 
Maybe this has to do with the dynamic/fixed IP problem.

johntdyer, 2018-01-23:
@niklasfink  - Honestly I gave up on Meraki for the house and switched everything to Unifi and it worked first try... I know thats probably not the answer you were looking for....

stefanbode, 2018-04-29:
Yes, this can happen. I requested a CRC check for the configuration to detect a defect. But this was declined. The problem is, that the device is not detecting the defect configuration at boot time and start with bullshit. Sometimes it start and just do not connect, sometimes it just reboot all the time.

senadaruc, 2018-06-28:
Gents,  I have the same issue with my Meraki MR53 and MR32 AP's after flashing my Sonoff devices with Tasmota FW. I tried all possible combinations and no success! I have more than 30 Sonoff devices at home with stock FW and they work perfectly with my Meraki AP's so is there any workaround to make this work?

Thanks,

stefanbode, 2018-06-28:
There seems to be a new issue with the 6.0.0a firmware. Ihad problem getting connected to my Fritzbox and with this change it now works again. No idea about side effects. I have deleted/comment out the command in support.ino

near line 1000

`  //WiFi.persistent(false);   // Solve possible wifi init errors`

senadaruc, 2018-06-28:
stefanbode, thanks for the feedback but searching inside the "support.ino" I found 3 lines with the same command.

`WiFi.persistent(false); // Solve possible wifi init errors
`
Do i need to comment(disable) the function in all 3 places or just at line 1012 ?

Thanks,

stefanbode, 2018-06-28:
Interesting to know. I was expecting it only once. Anyhow I would comment ALL of the lines out. The command itself seems to have some problems.

senadaruc, 2018-06-28:
Ok, testing now... I will let you know the results.

senadaruc, 2018-06-28:
Just tested by removing the "WiFi.persistent(false); // Solve possible wifi init errors" from "support.ino" but unfortunately this didn't work! same problem.....

> 00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_834616, GroupTopic sonoffs) Version 6.0.0a-2_3_0
> 00:00:00 WIF: Connecting to AP1 SmartMeraki in mode 11N as sonoff-1558...
> 00:00:17 WIF: Connect failed with AP timeout
> 00:00:17 WIF: Connecting to AP2 SmartMeraki in mode 11N as sonoff-1558...
> 00:00:33 WIF: Connect failed with AP timeout
> 00:00:33 WIF: WifiManager active for 3 minutes
> 00:00:34 HTP: Web server active on sonoff-1558 with IP address 192.168.4.1

stefanbode, 2018-06-28:
can you double check on the INFO. "status 2" that the build time is correct and you really have the NEW firmware up and running.

`12:42:08 MQT: hm/status/ESP_19FFA2/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-26T13:16:41","Boot":31,"Core":"2_4_0","SDK":"2.1.0(deb1901)"}}`

senadaruc, 2018-06-28:
```
pio device monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
SARUC-M-C1EA:Sonoff-Tasmota-development saruc$ pio device monitor --port /dev/cu.SLAB_USBtoUART --baud 115200
--- Miniterm on /dev/cu.SLAB_USBtoUART  115200,8,N,1 ---
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:33 CMD: status 2
00:00:33 RSL: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"6.0.0a","BuildDateTime":"2018-06-28T12:33:52","Boot":31,"Core":"2_3_0","SDK":"1.5.3(a
ec24ac9)"}}
00:00:34 WIF: Connect failed with AP timeout
00:00:34 WIF: WifiManager active for 3 minutes
00:00:34 HTP: Web server active on sonoff-1558 with IP address 192.168.4.1
```

JF62, 2018-06-28:
I also had to revert 20a5395 , because all of my Sonoffs get time out and no wifi connection - multiple different access points are in use here.
After reverting back everything is fine

senadaruc, 2018-06-28:
How to revert to 20a5395 ?

JF62, 2018-06-28:
I have my own repository so I reverted back all changes done by 20a5395
It looks for my, that the problem started with this commit.

senadaruc, 2018-06-28:
This is committed 4 hours ago, but I am strugling with the WiFi issue from yesterday morning !!! When I stated to flash my devices with Tasmota !

JF62, 2018-06-28:
You may have this problem longer (I also use AVM Routers) but after this fix the problems you described happend for me too. I had never seen this problem before. And after the revert the problems for me are gone.

ascillato, 2018-06-28:
Hi,

I have just tested the actual v6.0.0b and works for me without problems.

So, I think that your issue is not Tasmota, could be a config issue or a compilation issue.

I found that sometimes platformio compiles but the firmware have wifi issues. With just compiling twice, and then do the upload, the wifi issues are gone. If you compile with arduino ide, this weird behaviour is not there. There is an open issue at platformio with this at:
https://github.com/platformio/platform-espressif8266/issues/ 83

Hope this helps

senadaruc, 2018-06-28:
Ok, so I deleted all WiFi clients and DHCP reservations from my Meraki and then I flashed my Sonoff with `https://github.com/arendst/Sonoff-Tasmota/releases/download/v5.14.0/sonoff.bin` now works fine... Thanks

ascillato2, 2018-06-28:
Great!

Good to know that now it is working for you! :+1:

closed 2018-06-28 by ascillato2

JF62, 2018-06-28:
Thanks ascillato2 for the hint - twice compiling fix my problems

senadaruc, 2018-06-28:
Ascillato, from where you got the 6.0.0b version?

ascillato, 2018-06-28:
From this repository. In the readme say v6.0.0a but in the code say v6.0.0b

senadaruc, 2018-06-28:
Ok, and can you please compile a bin file for me so I can flash directly?

ascillato, 2018-06-28:
https://github.com/ascillato/Sonoff-Tasmota_KNX/releases

ascillato, 2018-06-28:
And

https://github.com/reloxx13/Sonoff-Tasmota-Modified/releases

Sergi0Martin, 2021-09-26:
I had same problem and I solved following these tasmota oficial steps (with termite) [using-serial-terminal](https://tasmota.github.io/docs/Getting-Started/#using-serial-terminal)

After send the command _Backlog ssid1 <yourssid>; password1 <your_password>; ssid2 <your_ssid2>; password2 <your_password>_  device connected automatically. IMPORTANT: put always ssid1 and ssid2 even if it is the same.