arendst/Tasmota #1368: extending tasmota, new command, web button color, command not found

MatzeJoerling, 2017-12-15:
Hi,

i am using tasmota in my home automation for some time now, great stuff. 
i want some extras and you programming guys answer is: fork, code!

i will try, but i am stuck! can someone take a look and send me a hint?
btw, i cant code, especailly not in C.

I want 2 new commands to set the color and highlight color of the menubuttons. I have that much devices that i need some attribute to recognize on which class of device i am.

i pretty much copied all from WebPassword to 2 new commands, compiles fine but gives me a "command not found" in the cosole. 

https://github.com/MatzeJoerling/Sonoff-Tasmota/commit/aaac8aec9c811dbb85eed12cc5dea26b38315e24

Any ideas? Only for function, i know that some of the code is not nice, especially in sonoff/settings.ino.

regards M.

reloxx13, 2017-12-15:
just to be sure, what did u enter as command into the console?

MatzeJoerling, 2017-12-16:
11:25:25 CMD: WebButtonColor1
11:25:25 MQT: stat/esp1/RESULT = {"Command":"Unknown"}
11:25:35 CMD: Web_bc1
11:25:35 MQT: stat/esp1/RESULT = {"Command":"Unknown"}

arendst, 2017-12-17:
The commands should work as long as you have the webserver enabled. To test see if command weblog still works.

Another problem is you messed up "compile time" and "run time". The defines used in the webserver.ino file will only be substituted at compile time. 
You want to be able to change colors at run time so there needs to be "something" programmed that takes care of changing the defines with your settings colors. To get run time control have a look at how I make use of page.replace like in ```page.replace(F("{ha"), my_module.name);```

MatzeJoerling, 2017-12-17:
thanks,

so runtime / compile time - right - i now get it. will look at page.replace.
if 2 of you say it should work, where did i mess that up? what should the command be???
weblog still works.

the tabs are nice, but colors are more eye-catching.
mainly this is for getting used to git/branches/push/pull and some easy code for beginners, and changing the color looks like easy stuff ( i thought).

M.

MatzeJoerling, 2017-12-17:
Hi,
i fixed the runtime/compiletime thing https://github.com/MatzeJoerling/Sonoff-Tasmota/commit/d51785c41b1d6b29b579755adef7db01effd9494
Working.

But i am lost with the command,
no matter what i type in the console (WebButtonColor1, webbc1, web_bc1), i always get an error.

_19:05:13 CMD: webbuttoncolor1 #ffffff
19:05:13 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
19:05:16 CMD: WebBC1
19:05:16 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
19:05:20 CMD: webbc1
19:05:20 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}
19:05:24 CMD: web_bc1
19:05:24 MQT: stat/sonoff/RESULT = {"Command":"Unknown"}_

any more ideas. would be really appreciated!

regards

MatzeJoerling, 2017-12-18:
Found it,

WebButtonColor1 , the command is WebButtonColor and the 1 is treated as index, like for switches ...
I have renamed the commands to names without numeric at the end, working.
https://github.com/MatzeJoerling/Sonoff-Tasmota/commit/1cd44015c9347db18490c7f39b3406c33e9e129b

regards

closed 2017-12-18 by MatzeJoerling