arendst/Tasmota #1369: 4Ch Relay Module, cannot found programming header

nicomda, 2017-12-15:
I've bought a bunch
 of 4ch tension free relay modules. Gonna attach a picture of my board. I tried to directly connect UART pins to the PSF-B04, without success in the flashing process. Drivers are ok, cause I've already flashed like 5 more sonoff switches. Do you know where is the header and its pinout?

[image]

nicomda, 2017-12-16:
I already got it working. I've to use the onchip TX/RX but it is working now.

closed 2017-12-16 by nicomda

tschaban, 2018-02-13:
hi, can you share where you connect tx,rx,3,3v + gnd? 
I'm also wondering if relays are controlled directly by GPIOs from this PSF, it does not look like that but maybe you knew?
thx

nicomda, 2018-02-15:
Yes, tasmota is working on it, I used the pins you just shared.

tschaban, 2018-02-15:
thx @nicomda

jpbarraca, 2018-03-05:
That board has a unknown chip (no markings) near the PSF, which intercepts connections to the relays. Do you have full control over the relays state?

tschaban, 2018-03-06:
you have a full control once you switch off additional function those chips brings like inching or selflocking.
I've uploaded to it my custom firmware compiled to ESP8285. From that video https://www.youtube.com/watch?v=i0SZLAJKloc&feature=youtu.be&t=105 you see it working  

Once you switch on inching/selflocing those chips impose additional features on a relay. You can see it from this moment at this video
https://www.youtube.com/watch?v=i0SZLAJKloc&feature=youtu.be&t=179

apology vides are in polish

jpbarraca, 2018-03-06:
Tks. I have a variant of this board with 4 buttons and 220v PSU.
Flashed ESPEasy (because of the Rules functionality) and it works perfectly. 
Unfortunately we have to solder the wires directly to the ESP as the headers do no provide the required connectivity.

h3lmutek, 2018-07-31:
i have 5V version, espurna for sonoff4CHPRO working perfect!