arendst/Tasmota #1370: Sleep through daylight (sleep until X)

AlessandroEmm, 2017-12-15:
Hey there

I know about the Sleep command but is there any way to sleep longer on purpose?

I have several Sonoff that switch light outside at night and thus don't have to be and up throughout the day.

Thanks a lot
Alessandro

cobra85ec, 2017-12-16:
U should use node-red to do ... its so good

Frogmore42, 2017-12-16:
There is currently no way to do this.  I assume your concern is power utilization of the device.  It is not clear that you would get much benefit from a longer sleep time, since the power supply used by Sonoffs is not really that efficient and at 250mS you are probably already hitting the low limit.

Frogmore42, 2017-12-16:
Looking at the code, sleep is stored in a byte so it would require adding more storage for a larger value and/or a redefinition to allow longer times.  But, I doubt this would be useful to many people.

AlessandroEmm, 2017-12-17:
@Frogmore42 thanks your assuming right I try to save as much energy as possible. I'm already at 250ms and I assumed there could be more in with a longer sleep as regular ESP utilize sub 0.1w when sleeping for a long time.

I guess it would make more sense to use another sonoff to shut them off completely through the daylight, what do you think?

ronnicek, 2017-12-17:
I would like to have actutally function for deepsleep.. so it wake up every 15 minutes to save battery.. (I think its the same as #1230)

davidelang, 2017-12-18:
Tasmota is designed to work with a MQTT server, and that requires being always 
connected. Deep sleep features may get added if they are trivial, but they go 
against the focus of this project enough that they will loose out to any 
conflict (including just code space)

AlessandroEmm, 2017-12-18:
@ronnicek question is whether deep sleep actually lowers the power consumption in this case.

@davidelang Why conflict? As in my case I _don't_ want it to listen to MQTT for a specific while?

ronnicek, 2017-12-18:
@AlessandroEmm actually.. the problem is, that sonoff-tasmota is designed for Sonoff devices (and they should be up and running 24/7). And yes, deepsleep for devices like Wemos are saving power. And also, you can have temp sensor near, because it's not heating up all the time from CPU :)

AlessandroEmm, 2017-12-19:
@ronnicek well the ESP obviously can save some power but the question is where the AC/DC Converter can step down the same...

webtop, 2017-12-20:
Why not just create a latching circuit (couple of transistors, couple of resistors, and an LDR), which only turns on the ESP when it gets dark? I have something very similar