arendst/Tasmota #1373: OTA upgrade not working - Upload buffer miscompare

tobang, 2017-12-16:
Hi, 
When i try to do an OTA upgrade, I get the following error in the web interface: Upload buffer miscompare

[image]

My current firmware version is 5.9.1 and the program size is 477kb

[image]

What am I doing wrong? Is there a way around it?

Thank you

mushudog, 2017-12-16:
I had the same issue, it has to do with the firmware already on the sonoff- you compiled using the wrong Spiff settings . It has to be 1M no spiff and probably you have different. For this reason there is not enough space. Solution is to flash via serial with the right settings (there are some other topics with same issue/solution)

tobang, 2017-12-16:
Thank you, that did the trick. However, I now have to flash the firmware again on all my devices :-(

closed 2017-12-16 by tobang

juan11perez, 2018-06-06:
You can still upgrade OTA. You need to first upload the sonoff.bin. minimal and then the full version. it works

tardar, 2018-07-14:
Uploading the minimal brings the same error on my S20 at Upgrading.
Could it be that the reason for this error is a first flash with wrong settings ?

Thanks for help :)

ascillato, 2018-07-14:
Yes, Sorry. you need to flash by cable. For avoiding any further issues, please do a full erase before flashing with `esptool.py --port COM5 erase_flash`

juan11perez, 2018-07-16:
@ascillato is there an erase tool with IDE (or Mac compatible)? Some of my sonoffs wont upgrade to 6.1.0 and only work with 5.14.0

ascillato, 2018-07-16:
No as far as I know.

Arduino IDE have an **erase all flash before upload** option but only with the esp8266 lib v2.4.1. It has nothing with esp8266 lib v2.3.0. So, for having that option in a stable esp core lib, inside Arduino, we will have to wait for next Esp8266 lib release (v2.4.2?)

But for just erasing you need `esptool.py --port COM5 erase_flash`. esptool.py is explained in the wiki.

Hope this helps

sumasage, 2019-01-04:
@mushudog if u flash with esptool would it automatically use the correct settings (1M no spiff)? I was able to update to 6.3.x but now can't update to latest 6.4.x

jcz1, 2019-01-04:
Same here. Dual R2, erased by "esptool.py --port /dev/ttyAMA0 erase_flash" and flashed via serial (via raspberry). I can upload sonoff.bin 6.3.0 via web, but not 6.4.1.
I had to upload 6.4.1 minimal and then normal sonoff.bin 6.4.1.

OderDochNicht, 2019-12-22:
Teckin SS42. Flashed with tuya-convert to 6.2.1.
Downloaded tasmota-DE.bin - attempt to flash - Upload Failed + Upload buffer miscompare
Downloaded tasmota.minimal.bin - attempt to flash - success
Downloaded tasmota.bin - attempt to flash - success
Again tried to flash tasmota-DE.bin - no luck - Upload Failed + Upload buffer miscompare

I'm pretty sure I flashed the right size, Shouldn't I not:
Flash Size	1024kB
Program Flash Size	1024kB
Program Size	567kB
Free Program Space	436kB
Free Memory	27kB

meingraham, 2019-12-22:
If updating via file upload OTA, you have to first update to tasmota-minimal. But, if you are just downloading a file from the web and then performing the file upload OTA... you should just point the URL to the variant binary you want and use web OTA. Then Tasmota will handle the minimal intermediate step automatically before loading the final firmware.

Please read - https://tasmota.github.io/docs/#/Upgrading?id=ota-upgrade

MHOMEASS, 2020-08-28:
> You can still upgrade OTA. You need to first upload the sonoff.bin. minimal and then the full version. it works

Still works perfect to upgrade

AlBrough, 2020-08-30:
Same issue 8.3.1 > 8.4.0
Updated to lite.bin first and then the full version. 
Thank goodness for TasmoAdmin, updating 17 of these is a pain

remy-burney, 2020-10-05:
same, used the minimal version first, and then the normal tasmota.bin

C0rn3j, 2021-01-04:
Ran into this issue on BW-SHP6 15A, but it seems that it was due to the fact I had old 8.0.x firmware from tuya-convert.

Did OTA web upgrade from 8.0.x->8.5.1->9.1.0->9.2.0(latest) and that worked fine.