arendst/Tasmota #1374: Sonoff Basic and Elecapital Push Button

rajkedda, 2017-12-16:
Hi,
 I successfully flashed the Sonoff Basic with Tasmota and works very well.  I need advise on how to connect the button to basic.  Based on articles and video, here is what I have done:

Sonoff (Header pins)      to        Push Button (pin)
GPIO 14                                      NO
GND                                            C
3.3V                                             Positive
???                                                Negative
With above connection, get the push button to turn on and off but LED is not lit.
Need to connect ground to negative to get LED lit on the push button.  Making a Y connection from header pin GND to C and negative does not work as it makes the LED lit, but the button to turn on the basic unit stops working.  So I am assuming Y connection is a bad idea.  Any other possibility or approach to get both aspects working LED and push button.

Frogmore42, 2017-12-16:
It sounds like you have a push button that has a built-in LED.  If you are trying to get the LED in the push button to reflect the state of the relay in the Sonoff Basic, that is likely going to be challenging.  How you can do this is going to depend on the push button and how it works.  I don't have a Basic, but the other ones I have do have LEDs on the Sonoff that indicate the state of the relay.  Is that not good enough?

Based on the 4 connections to the switch, I would expect that applying power to Positive and Negative would light the LED.  If that is the case (and there are no other internal connections in the switch) you should be able to connect those wires (with the correct polarity) to the relay coil (not the contacts).  Do not attempt this if you do not know exactly how to do this as it would be very dangerous if you make a mistake in the connections.

rajkedda, 2017-12-18:
Thank you.  LED on the Elcapital push button is purely to show the location of the light on the wall.  I am not using it for the status of the basic switch as the unit is mounted inside the wall box.  I have seen in the videos others do it, but they have not articulated the wiring.