arendst/Tasmota #1375: MagicHome OTA?

edit21, 2017-12-16:
Hi All, My MAgicHome device came with no rx/tx pads :( just wondering if anyone knows if this can be flashed OTA?
Kinda stuck otherwise!

Thanks for the epic code!

James

edit21, 2017-12-16:
Found the new pinout - will add to the Magic Home page :)

closed 2017-12-16 by edit21

ageorgios, 2018-01-05:
Is any OTA method available?
The pins on the chips are too risky for me to solder.

lobradov, 2018-01-05:
Nope, soldering only :/

davidelang, 2018-01-05:
well, wired anyway (you can use pogopins or just hold the wires in place while 
flashing, you don't _have_ to solder them.

Once Tasmota is installed, you will be able to upgrade over the network, it's 
the transition from the stock firmware to tasmota that will require wires