arendst/Tasmota #1376: magic home color blended color issue.

donster22, 2017-12-16:
Hi.

I have x 2 MagicHome LED controllers (1 is an ESP8285 chip and is configured as "Arilux" the other is an ESP8266 chip and configured as MagicHome). Both are flashed with Sonoff-Tasmota 5.9.1j.

Everything works extremely well except for the colors.

Red, Green and Blue work as expected. I can see this from injecting "Color 1", "Color 2", "Color 3" from the console. In fact all of the Colors (1-12) seem to work OK except 10 which shows green rather than yellow.

My problem generally arises when I try to blend the primary colours. First off I have tried to mimic the "warm" light from my Tradfri bulbs. The hex for this is FF6432 but I get a cold white light. Similarly when I aim for yellow (EEEE00) I get green. Taking any colours from an online generator and putting them into the console gives me completely different colors to the ones I'm after but I can get the primary colours individually.

Both MagicHomes behave exactly the same way. Please can you help me determine if there is a fault somewhere or is there a setting I have missed when flashing or configuring the devices perhaps? (most likely I imagine).

Thanks, Don

arendst, 2017-12-17:
As the illumination of your three led colors is different you cannot easily decide that using EEEE00 would be yellow; In your case the Green led has more lumen than the Red led. 

There are books written about color/hue/brightness and how to use them. You may find attaching another brand of led string will provide different colors too.

There is an issue here where someone suggested to provide some kind of filter to make all three colors having the same lumen. That has not been implemented (yet) as it will lower the overall max lumen available.

CZ1979, 2019-02-28:
@donster22 : Have you solved this issue for you? I've got the same problem with tasmota and a 12v LED Stripe. (yellow is green.)