arendst/Tasmota #1377: Can not connect to wifi

lampenschakelen, 2017-12-16:
Thanks i lot for making this software ;-) 

Can you help me a little bit?

It keeps connecting with wrong wifi credentials 
0:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_66706C, GroupTopic sonoffs) Versie 5.10.0a
00:00:00 WIF: Verbinden met AP2 indebuurt2 in stand 11N als sonoff-4204...
00:00:07 WIF: Verbinding mislukt omdat AP onbereikbaar is
00:00:07 WIF: Verbinden met AP1 indebuurt1 in stand 11N als sonoff-4204...
00:00:14 WIF: Verbinding mislukt omdat AP onbereikbaar is
00:00:14 WIF: WPSConfig1 minuut actief
00:00:33 APP: Herstarten

i changed the sonoff\user_config.h file 
With the follow setup below: 
#define WIFI_IP_ADDRESS        "192.168.2.197"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.2.254"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "8.8.8.8"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "wifilamp"      // [Ssid1] Wifi SSID
#define STA_PASS1              "1234567890"  // [Password1] Wifi password
#define STA_SSID2              ""      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              ""  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_RETRY    // [WifiConfig] Default tool if wifi fails to connect

davidelang, 2017-12-16:
see the troubleshooting page on the wiki

reloxx13, 2017-12-16:
change CFG__HOLDER value (eg ast digit + 1 oder - 1 to load new default config file

lampenschakelen, 2017-12-16:
Thanks, allot.

Just after the post, i found it.
I was not 100% sure about it.

This topic is fixed.

chris-008, 2017-12-17:
I have a similar issue after upgrading to 5.10.0, but changing the CFG_Holder value however didn't solve it. Any other ideas?

00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_BD2CCA, GroupTopic sonoffs) Version 5.10.0
00:00:00 WIF: Connecting to AP1 Wifi-008-2.4GHz in mode 11N as sonoff-3274...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP2 Wifi-008-2.4GHz in mode 11N as sonoff-3274...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute

lampenschakelen, 2017-12-17:
Please share you're code

chris-008, 2017-12-17:
thanks for the quick reply!
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_BD2CCA, GroupTopic sonoffs) Version 5.10.0
00:00:00 WIF: Connecting to AP2 Wifi-008-2.4GHz in mode 11N as sonoff-3274...
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 Wifi-008-2.4GHz in mode 11N as sonoff-3274...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfigactive for 1 minute
00:00:33 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,6)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld

lampenschakelen, 2017-12-17:
Please send you’re user config with you’re Wifi credentials.

chris-008, 2017-12-17:
#define CFG_HOLDER             0x20171217        // [Reset 1] Change this value to load following default configuration parameters

but did'nt help after refl
ashing it. I have this issue on two devices now after being upgraded to 5.10.0

chris-008, 2017-12-17:
// -- Project -------------------------------------
#define PROJECT                "sonoff"          // PROJECT is used as the default topic delimiter and OTA file name
                                                 //   As an IDE restriction it needs to be the same as the main .ino file

#define CFG_HOLDER             0x20171217        // [Reset 1] Change this value to load following default configuration parameters
#define SAVE_DATA              1                 // [SaveData] Save changed parameters to Flash (0 = disable, 1 - 3600 seconds)
#define SAVE_STATE             1                 // [SetOption0] Save changed power state to Flash (0 = disable, 1 = enable)

// -- Wifi ----------------------------------------
#define WIFI_IP_ADDRESS        "0.0.0.0"         // [IpAddress1] Set to 0.0.0.0 for using DHCP or IP address
#define WIFI_GATEWAY           "192.168.2.254"   // [IpAddress2] If not using DHCP set Gateway IP address
#define WIFI_SUBNETMASK        "255.255.255.0"   // [IpAddress3] If not using DHCP set Network mask
#define WIFI_DNS               "192.168.2.27"    // [IpAddress4] If not using DHCP set DNS IP address (might be equal to WIFI_GATEWAY)

#define STA_SSID1              "Wifi-008-2.4GHz"      // [Ssid1] Wifi SSID
#define STA_PASS1              "password"  // [Password1] Wifi password
#define STA_SSID2              "Wifi-008-2.4GHz"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "password"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)

chris-008, 2017-12-17:
[image]

lampenschakelen, 2017-12-17:
Change the CFG_holder on the end to 18
The ip settings are correct?

The WiFi credtials are correct ?

chris-008, 2017-12-17:
I haven't changed the IP settings, but  I have used the same as in the previous version which worked well.
The WIFI credentials are indeed correct.
I had changed the CFG holder from 20160912 already.
What about the Wifi config tool?
Not too sure about this I must say...
https://github.com/arendst/Sonoff-MQTT-OTA-Arduino/issues/271
https://github.com/arendst/Sonoff-Tasmota/issues/1252

reloxx13, 2017-12-17:
u can try to change it to wifi manager, if the sonoff cannot get into your wifi with the config from the user_config, it wil start the wifi_manager. than you can connect direct to your sonoff via wifi.
now you can open the web interface and enter your wifi credentials again.
if you change it in teh user_config.h, change the CFG_HOLDER again.

i would suggest you to remove the dot from your wifi ssid.

Wifi Manager:
> 4 short presses: Start Wifi manager providing an Access Point with IP address 192.168.4.1 and a web server allowing the configuration of both Wifi and MQTT parameters. The LED will blink during the config period. A single button press during this period will abort and restart sonoff.

chris-008, 2017-12-17:
That did the trick, thanks a lot to both for your kind help!

Spyky, 2018-03-04:
Hello, i'm having the same problems. See that it gets a webserver IP adres 192.168.4.1.
I can't open the webserver. How do you do that. If i put the webserver adres in the browser i get nothing also nothing with the ip adres i have in the configuration.
If i connect my pc to the wifi webserver i see a logon screen, but can't get in. Not with usr admin-password empty or sonoff and password empty.

define WIFI_IP_ADDRESS        "192.168.1.224"    
#define WIFI_GATEWAY           "192.168.1.1"    
#define WIFI_SUBNETMASK        "255.255.255.0"     
#define WIFI_DNS               "192.168.1.1"     

#define CFG_HOLDER             0x20161209         
#define SAVE_DATA              1                  
#define SAVE_STATE             1  

What do i do wrong?

E_WEBSERVER                            
  #define WEB_SERVER           2              
  #define WEB_PORT             80                
  #define WEB_USERNAME         "admin"            
  #define WEB_PASSWORD         ""                 
  #define FRIENDLY_NAME        "Sonoff"         
  #define USE_EMULATION

njloki30, 2018-04-07:
Just some info. I had the same issue, turned out, the sonoff config didn't like the fact that my Wifi password contained special characters " \ " and " * " even when I escaped them in the code.  After updating my Wifi password to not include those characters it connected fine.

moritz89, 2018-05-06:
So is it a known limitation that non-alphanumerical characters in the SSID and password are not supported? Standalone ESP32 and 8266 SoC support them, deducing that the problem has to be due to the fact that Sonoff 4CH uses the 8285 or somewhere in between.

I verified that it is not the fault of the settings struct after passing the SSID and password as strings directly to the WiFi.begin() function in support.ino on line 644.

lvta0909, 2018-05-06:
I'm having the same problem with special characters on the password on version 5.13.1, but apparently this issue does not affect version 5.9.1. Which supports @chris-008 comment: This issue was introduced in version 5.10.0.

Module 1: sonoff basic
Program Version	5.9.1
Build Date & Time	2017-11-13T21:43:41
Core/SDK Version	2_3_0/1.5.3(aec24ac9)

Module 2: sonoff basic
Program Version	5.13.1
Build Date & Time	2018-05-06T14:10:50
Core/SDK Version	2_4_0/2.1.0(deb1901)

rmoriz, 2018-05-14:
Can't connect to a Wifi with two backslashes in the password. Neither via initial flashing nor setting through the web ui.

Version: 5.13.1b
Device: Sonoff S20

**Workaround**:

Configure a second SSID to make the initial connect, then configure mqtt and use it to re-set the password for ssdi1, for example:

```shell
mosquitto_pub -h mqtt-server-ip -t 'cmnd/sonoff/Backlog' -m 'password1 my\backslash\password'
```

TaRga51170, 2018-05-29:
I am also having the same problem with wifi connectivity.  Two (stubborn :)) sonoffs have no problems connecting with a router named "MAIN" but would not even connect to a router named "TP-Link_C9".  Yesterday I've spent a whole afternoon reflashing, erasing flash, changing CFG__HOLDER value .. you name it .. I did it.  From the above posts now I am seeing something which I did not explore.  Could it be the "-" sign or other special characters in the SSID?  When I'll go home I will try to change SSID name and report back.

aenertia, 2018-06-08:
I have similar issues - however in minimal build it connects just fine. 

Even after stripping out all the defines to have pretty much bare minimal mqtt + bme280/i2c and nothing else I get the same behaviour off the master tree repo.

aenertia, 2018-06-08:
``--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H ---
00:00:07 WIF: Connect failed as AP cannot be reached
00:00:07 WIF: Connecting to AP1 kainga-atawhai in mode 11N as sonoff-5296...
00:00:14 WIF: Connect failed as AP cannot be reached
00:00:14 WIF: WPSConfig active for 3 minutes
00:00:34 APP: Restarting..

The passphrase doesn't have any non alpha , and is 8 chars long for that network.

Like my previous comment it happens on both basics and th16's when building off the master tree ; if I define sonoff-minimal the problem goes away and the unit connects just fine to the wifi.

aenertia, 2018-06-08:
user_config.h attached btw this is off commit 18035506a34cb59de3107e06ad023909b6021dfa
[user_config (copy).h.txt](https://github.com/arendst/Sonoff-Tasmota/files/2084125/user_config.copy.h.txt)

TaRga51170, 2018-06-08:
Remove any special characters from the SSID.  Mine worked after removing the special characters.

aenertia, 2018-06-08:
Hrm - This seems to have something to do with fallback AP's and Static settings. 

After flashing back to minimal and clearing the secondary ap and static fallback strings; and reflashing with head I am able to connect.

hanthecat, 2018-06-14:
@rmoriz Your workaround fixed my issue! Thank you!

ascillato2, 2018-07-10:
Hi,

Seems that your question has been addressed. Closing this issue. Please reopen if needed. Thanks.

closed 2018-07-10 by ascillato2

rmoriz, 2018-07-10:
It's still broken, just a workaround. Or did I miss something?

mixu-, 2018-07-23:
Not sure if my issue was related, but I did see the same error messages. So posting here in case someone else runs in to the issue I had.

When I upgraded from 6.1.0 to 6.1.1 I had two out of my five Sonoff relays crap out on me with the same error messages as mentioned in the OP. They are basic sonoff relays - possibly older HW than the other three I have.

> 0:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_66706C, GroupTopic sonoffs) Versie 5.10.0a
> 00:00:00 WIF: Verbinden met AP2 indebuurt2 in stand 11N als sonoff-4204...
> 00:00:07 WIF: [...]

In addition to the above, with more verbose logging I could see WPSConfig was failing with STATUS 4. This was after "WIF: Connect failed as AP cannot be reached"

Reflashing even with a clean 6.1.1 or 6.1.0 did nothing - Connection to wifi AP as well as WPSConfig were still failing with the same error. I reflashed using Arduino IDE if that makes a difference.

I tried reset1, reset2, button based reset, changed the CFG_HOLDER. None of those helped.

I went back to flashing a clean 5.14 release (no changes in config whatsoever), gave ssid1 and password1 via serial console, and boom - I had connection. After that I upgraded to 6.1.0 and things still work. I dare not touch 6.1.1 because that could mean having to take the sonoffs out from the light fixtures and open them up again.

ascillato, 2018-07-23:
Hi,

@rmoriz, @knd775, The author of this issue said that changing his CFG_HOLDER solves its issue. That is why this issue was closed. From now on, before closing I should make a reference. About the special characters on the SSID or PASSWORD, that has been addressed in several issues before and sorry, those are not supported by now. If you need anything else, just open a new issue with the troubleshooting template complete in order to help you. There are so many issues opened that are stuck or left behind that we need more help in order to solve as many as possible.

@mixu- I found the same issue as you with some of the precompiled bins. Have you tried to compile by yourself. In my case, that worked.

ascillato, 2018-07-23:
Please, also see https://github.com/arendst/Sonoff-Tasmota/issues/3289

Thanks

knd775, 2018-07-24:
@ascillato My issue, and likely many other peoples', was caused by the apparent corruption mentioned in #3289. I flashed the newest 6.1.1 binaries and am good now! Thanks.