arendst/Tasmota #1379: Domoticz Idx

fedele, 2017-12-17:
Hi,
I have this problem configuring my SOnOff:
Domoticz Sensor Idx can't be 5 numbers or letters.
Infact I have in Domoticz a Virtual Device with Idx 1401A but 
SOnOff accept only 1401 and send values for this,
but I can accept only for 1401A.

Can you help me?
Fedele

arendst, 2017-12-17:
You did use the decimal Domoticz Idx field and not the hexadecimal ID field?

[image]

In this case it's 91

fedele, 2017-12-17:
Thank you for the answer.

It works: I don’t know way did not wok first time conducting me in troubles.

Fedele 

 

 <https://user-images.githubusercontent.com/11044339/34078069-50c06428-e313-11e7-9686-515f95f7488d.png>