arendst/Tasmota #1382: sonoff cannot connect to same SSID

sam9s, 2017-12-17:
I have been using sonoff with tasmota for list 3-4 months now without any issue, now recently I factory reset my router and the reconfigured it again. I kept the SSID and password EXACTLY same as it was earlier, all my devices got connected except sonoff on tasmota. The green light keeps blinking trying to connect to the device but fail, and I hear the relay rest after every 1-2 minutes. 

I am lost now as I cannot access, the web interface as the device is not getting connected to the wifi. (and no ip is recieved) Please assist me how to address this issue. I would avoid reflashing as my device is packed up behind the wall switchboard. But I can see the green light blinking trying to connect to wifi ssid. Please assist this has stopped my entire home automation.

Regards
Sammy

reloxx13, 2017-12-17:
build your own bin file an change default wifi tool to wifi retry
```
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```
dont forget to change the CFG_HOLDER in same file.

for now, press the button on the device 4times to start the wifi manager:
https://github.com/arendst/Sonoff-Tasmota/wiki/Button-usage

MatzeJoerling, 2017-12-17:
he cant do that because that means to flash the devices via serial.
did the mac of the ap change? anything in the ap/router log? did you do a power cycle?
In my opinion that should work! i have 2 Aps with the same ssid, sonoff connects to both.

sam9s, 2017-12-17:
ok reloxx solution pointer worked, pressing the button 3 times defaulted the config.h file and I got the web interface back, however, I did have to open the switchboard and that too for all 6 devices spread across my house, which I did not want. Its not feasible to open every switchboard for ever sonof device if this happens again.

So my question is 1) WHY did this happen as all I did provide the same ssid and password, all devices got connected except sonoff, this should not be the intended behavior.

2), How can we rectify this situation externally, OTA of simply saying without opening the switchboard to access the device. Can we somehow bring the functionality of the physical button 

to answer MatzeJoerling no bro ..... mac address did not change, as it the same device. Yes, I did do a power cycle that was the first thing I did. 

Basically, I do not want the switchboard to open again and again and that too for all sonoff devices across the house if this kind of situation comes again.

Please suggest a viable solution.
Sammy

reloxx13, 2017-12-17:
if u did not change the WIFI_CONFIG_TOOL before compiling, the devices start in wps mode as the defaut.

i guess, if u reconnect power to all devices, it should work, too.

they r searching for your wlan, if not found they start wps mode till restart.

sam9s, 2017-12-18:
Yes I understand that, but if anyone can shed some light as to why this happened, the root cause and how to mitigate this without opening the switchboards, because sonofss are ment to hide behind the switchboards, and a problem like this creates a huge issue as you have to open all switchboards and to physically access the button, Whats the alternative or a feasible viable solution to this problem.

Igor122017, 2017-12-20:
hello. i have problem with sonoff T1 2Ch. I reload sonoff with Sonoff-Tasmota. all be, ok... then across 2-3 minutу, I can not see this sonoff on web, lost contact with router.  If I off then on, sonoff woking 2-3 minute and lost wifi. before, about day ago  was all good. help me....

binderth, 2017-12-22:
I'm having difficulties getting freshly flashed Sonoffs to connect with the first Wifi-credentials. They all connect with the second one. At first I thought it is, because my Wifi-Password contains Special characters like &, $, @, ! - but then it could be the the `WIFI_CONFIG_TOOL` attribute.
This one is undocumented, so whats the difference between those six?

Thanks!
```
#define STA_SSID1              "SSID"      // [Ssid1] Wifi SSID
#define STA_PASS1              "PASSWORD-WITH-SPECIAL-CHARACTERS"  // [Password1] Wifi password
#define STA_SSID2              "GUESTSSID"      // [Ssid2] Optional alternate AP Wifi SSID
#define STA_PASS2              "PASSWORT-WO-SPECIAL-CHARACTERS"  // [Password2] Optional alternate AP Wifi password
#define WIFI_CONFIG_TOOL       WIFI_WPSCONFIG    // [WifiConfig] Default tool if wifi fails to connect
                                                 //   (WIFI_RESTART, WIFI_SMARTCONFIG, WIFI_MANAGER, WIFI_WPSCONFIG, WIFI_RETRY, WIFI_WAIT)
```

Igor122017, 2017-12-22:
as I understand, the problem is somewhat different. describe the steps, the download went fine, then the connection was established, (wifi pre-configured), then I go to the web page and configure the MQTT and the type of sonof. and then tricks begin, after saving the settings for 5 minutes, some kind of malfunction occurs and all my settings are lost.