arendst/Tasmota #1386: SonOff 4CH Pro SetOption13

Vobiscus, 2017-12-17:
I set in the console SetOption13 1
But if i double tip the hardware buttons of the sonoff, the channel blocked.
He is now only switchable with software commands, but not with hardware button.
Its reproducible with all four channels.

arendst, 2017-12-18:
Pls provide output of command ``status 0``.

Vobiscus, 2017-12-18:
Here the output:

18:39:51 CMD: Status 0
18:39:51 MQT: stat/sonoff/STATUS = {"Status":{"Module":23,"FriendlyName":"Sonoff","Topic":"sonoff","ButtonTopic":"0","Power":4,"PowerOnState":3,"LedState":1,"SaveData":1,"SaveState":1,"ButtonRetain":0,"PowerRetain":0}}
18:39:51 MQT: stat/sonoff/STATUS1 = {"StatusPRM":{"Baudrate":115200,"GroupTopic":"sonoffs","OtaUrl":"http://domus1:80/api/arduino/sonoff.ino.bin","Laufzeit":10,"Sleep":0,"BootZähler":39,"SaveCount":219,"SaveAddress":"FB000"}}
18:39:51 MQT: stat/sonoff/STATUS2 = {"StatusFWR":{"Version":"5.10.0","BuildDatumUhrzeit":"2017.-357882573.01 1","Boot":6,"Core":"2_3_0","SDK":"1.5.3(aec24ac9)"}}
18:39:51 MQT: stat/sonoff/STATUS3 = {"StatusLOG":{"SerialLog":2,"WebLog":2,"SysLog":0,"LogHost":"domus1","LogPort":514,"SSId1":"Gumprecht","SSId2":"Gumprecht","TelePeriod":300,"SetOption":"5580200B"}}
18:39:51 MQT: stat/sonoff/STATUS4 = {"StatusMEM":{"ProgramSize":477,"frei":524,"Heap-Größe":23,"ProgramFlashSize":1024,"FlashSize":1024,"FlashMode":3}}
18:39:51 MQT: stat/sonoff/STATUS5 = {"StatusNET":{"Hostname":"SonOff4CHpro","IPAddress":"192.168.68.210","Gateway":"192.168.68.1","Subnetmaske":"255.255.255.0","DNSServer":"192.168.68.1","MAC":"60:01:94:98:34:6B","Webserver":2,"WifiConfig":2}}
18:39:51 MQT: stat/sonoff/STATUS6 = {"StatusMQT":{"MqttHost":"192.168.68.220","MqttPort":1883,"MqttClientMaske":"Sonoff4CHpro","MqttClient":"Sonoff4CHpro","MqttUser":"sonoff","MAX_PACKET_SIZE":512,"KEEPALIVE":15}}
18:39:51 MQT: stat/sonoff/STATUS7 = {"StatusTIM":{"UTC":"Mon Dec 18 17:39:51 2017","lokale Zeit":"Mon Dec 18 18:39:51 2017","StartDST":"Sun Mar 26 02:00:00 2017","EndDST":"Sun Oct 29 03:00:00 2017","Timezone":1}}
18:39:51 MQT: stat/sonoff/STATUS10 = {"StatusSNS":{"Zeit":"2017.12.18 18:39:51"}}
18:39:51 MQT: stat/sonoff/STATUS11 = {"StatusSTS":{"Zeit":"2017.12.18 18:39:51","Laufzeit":10,"VCC":3.212,"POWER1":"OFF","POWER2":"OFF","POWER3":"ON","POWER4":"OFF","WLAN":{"AP":1,"SSID":"Gumprecht","RSSI":84,"APMac":"08:96:D7:9D:1D:AB"}}}

Vobiscus, 2017-12-21:
Is there any chance that this will be fixed, respectively is it possible to fix this issue by firmware changes?

arendst, 2017-12-22:
What do you want to gain with SetOption13 1 on a 4CH?

Vobiscus, 2017-12-22:
The problem is, if i douple tap the hardware buttons on 4CH pro, the channel is locked.
Then you can't switch the channel by click the buttons, only by software.
With SetOption13 i thought i could disable this lock feature/bug :-)

arendst, 2017-12-22:
I noticed it once too. Suggest currently not to double tap the button. Will look in to it next year...

Vobiscus, 2017-12-22:
> I noticed it once too. Suggest currently not to double tap the button. Will look in to it next year...

I'll try my best, it's not me, it's my family with their shaky fingers :-)
I wish you merry christmas, lets take a look next year ;-)

Vobiscus, 2018-01-31:
Hello 
Any news for this problem with double tap?

Wessix, 2018-02-03:
Yea would appreciate a solution to.

arendst, 2018-02-04:
Deep dived in it but I'm afraid I cannot solve this one.

The buttons are not connected to the Tasmota controlled ESP8266 but to a second microcontroller. This microcontroller sends single presses almost directly to the ESP8266 for Tasmota to act upon.

Double taps are not transferred to Tasmota and seem to block further relay operation by that button.

Any suggestions are welcome....

RoganDawes, 2018-02-05:
Has anyone looked at modifying the firmware on the second microcontroller? It could open up all sorts of possibilities for interfacing with other RF devices. e.g. weather stations, power meters, etc. For more examples, check rtl_433 project.

closed 2018-06-20 by stale[bot]