arendst/Tasmota #1387: Figuring our learned RfCode in RF 433 Hub

iletunic, 2017-12-17:
I'm a total newcomer into Sonoff and MQTT... I successfully flashed a latest model of RF 433 bridge, and can learn my existing remote I use for some lights and shutters. Keys in the web and via RfKey[x] work fine. However, I would like to integrate the controls into Home assistant via MQTT, and cannot figure how to get the correct codes to send via RfCode. 

If I just add a single MQTT command as e.g. 'sonoffhub/cmnd/RfKey1', that works fine, but I can only switch something on or off with a single entry.

I assume I need to configure my MQTT command_topic as 'sonoffhub/cmnd/RfCode'  and send different payloads for turning different switches on and off. I just cannot figure out the actual codes... I see for example:
sonoffhub/RESULT {"RfReceived":{"Sync":11410,"Low":370,"High":1070,"Data":"151015","RfKey":1}}

Can I manually convert this to the correct RfCode?

Thanks for any help or pointers on where to read more.

ChukXP, 2017-12-18:
I suppose that You'll need to set these parameters first:
"High":1070 with command
RfHigh | 2..32767 | Set high pulse time in microseconds
"Low":370 -
RfLow | 2..32767 | Set low pulse time in microseconds
"Sync":11410 -
RfSync | 2..32767 | Set start sync pulse time in microseconds

Then send payload which You can get from "Data":"151015" 
RfCode | 1..8388607 | Send 24-bit user code using RfSync, RfLow and RfHigh timing

If Your data in hex then use 
RfCode | #1..#7FFFFF

iletunic, 2017-12-18:
Thanks, I understand now... I assume sending all these in a single MQTT command is not possible. Or could I send them all via some kind of JSON object?

jannnfe, 2017-12-22:
I have a old 433Mhz [wireless socket](https://www.amazon.de/uniTEC-48110-Funkfernschalterset-Micro-weiß/dp/B002ZCX6C0/ref=sr_1_3?ie=UTF8&qid=1513958755&sr=8-3&keywords=funksteckdose) at home and want to control that socket with the Sonoff RF Bridge with Tasmota running. Can someone explain exactly with all the steps what i have to do, to control the socket with the RF Bridge. When i click the Remote of the Socket i receive a message. But how can i send a command that the socket will be switch? Thanks!

Jason2866, 2018-01-17:
There is a another problem. What the Rfbridge shows as monitored code is not really the correct code.
For example: sent Intertechno Code #150555 with Rf-Remote.
RfBridge monitored: 
20:35:54 MQT: tele/sonoff-8E8329/RESULT = {"RfReceived":{"Sync":11350,"Low":360,"High":1090,"Data":"15155F","RfKey":"None"}}
Sent this code from RfBridge the poweroutlet didnt react.
Sending the code #150555 with the Rfbridge switches the poweroutlet too.
Maybe this helps to find the "learning bug"

Jason2866, 2018-01-23:
Hi jannnfe,

example for item Switch:

Switch Sideboard   "Sideboard"   (LR,gLight) [ "Lighting" ] 
		{ mqtt=">[broker:cmnd/sonoff-8E8329/Backlog:command:ON:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #150151],
                >[broker:cmnd/sonoff-8E8329/Backlog:command:OFF:RfSync 11320; RfLow 360; RfHigh 1090; RfCode #150154]" }

jannnfe, 2018-01-23:
Thanks but now I've already managed to control my devices. Thanks for your help!

closed 2018-06-07 by stale[bot]