arendst/Tasmota #1389: Decouple relay from switch

dave305, 2017-12-18:
I have been working to to get a switch to decouple from the relay on a sonoff basic and send a message to MQTT. My topics in Tasmota are:
Topic=Yard_LiGhts
Full Topic=LiGhts/Yard_LiGhts/

I have set switchtopic=Rain-Sensor and switchmode=2. I have tried to sub to MQTT topics:
stat/Rain-Sensor/POWER2
stat/Rain-Sensor/POWER1
cmnd/Rain-Sensor/POWER2
cmnd/Rain-Sensor/POWER1
Lights/stat/Rain-Sensor/POWER2
Lights/stat/Rain-Sensor/POWER1
Lights/cmnd/Rain-Sensor/POWER2
Lights/cmnd/Rain-Sensor/POWER1
Lights/Rain-Sensor/stat/POWER2
Lights/Rain-Sensor/stat/POWER1
Lights/Rain-Sensor/cmnd/POWER2
Lights/Rain-Sensor/cmnd/POWER1
#

I am trying to get a rain sensor (PCB with traces as a switch) to turn off my Xmas lights and it works, but if it rains before the time to switch on the lights the relay turns on. I need to send a message to MQTT so I can use node reds logic to keep the lights off when the sensor is wet.

I have searched google and read until my eyes hurt, but i keep on getting the same advice to do the above. I am about ready to give up so I thought I would post to see if anyone might have a clue.

dave305, 2017-12-18:
I have it working :). What I did wrong was My topics in Tasmota where wrong,

Old topics:
Topic=Yard_LiGhts
Full Topic=LiGhts/Yard_LiGhts/

New topics:
Topic = %topic% (sonoff)=Yard_Lights
Full Topic (%prefix%/%topic%/)=Lights/%topic%/

Happy camper :).

MatzeJoerling, 2017-12-18:
the nodered thing could look like this:
[image]

you need: node-red-contrib-bigtimer, node-red-contrib-bool-gate, mqtt and perhaps alexa local to start the light manually. You have to negate the rain sensor in the and gate.

dave305, 2017-12-18:
Dude you rule!!!, I have been trying to get my flow working and seeing that I am new to this it has been a learning experience. Thanks for your help. You gave me a place to start.  I have about the same, but was using traffic cop for the and gate.