arendst/Tasmota #1391: Bricked Sonoff while uploading Tasmota?

joaohess, 2017-12-18:
Hello,

I've already uploaded Tasmota to about 6 Sonoff Basic devices (when I was on MacOs Sierra; now I'm on High Sierra).

Friday I received other 3 Sonoffs and tried to upload Tasmota to them. I did everything I always did:

- Pins are right (they are binded together so I can't have mixed RX/TX)
- FTDI is on 3.3V
- I held the Sonoff button while plugging in to go into firmware rewrite mode.

With all 3 Sonoffs I got an "Error 255"  and if I try to connect them to power, nothing happens. What can be wrong? How can I upload Tasmota now?

I've tried on a Mac on Sierra, on a PC with Windows 10 and still get the same error.

`Building .pioenvs/sonoff-ds18x20/firmware.bin
Looking for upload port...
Auto-detected: /dev/cu.usbserial-A9M9DV3R
Uploading .pioenvs/sonoff-ds18x20/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
`

PianSom, 2017-12-18:
I have had similar issues recently, for no good reason I could ever identify. 

In my case solved by running 

`esptool.py --port=/dev/cu.usbserial-00000000 -b 115200 write_flash 0x0 ~/Downloads/Sonoff-Tasmota-development/.pioenvs/sonoff/firmware.bin`

from a Terminal (after having installed esptool.py, of course).

joaohess, 2017-12-19:
I've tried using esptool but the GUI version. No luck. Still Error 255. How could I possibly have bricked the device?! The FTDI is set to 3.3V and I've never changed that (it uses a jumper connector, not a switch, and I never took out of its place)

`Uploading...
esptool v0.4.8 - (c) 2014 Ch. Klippel <ck@atelier-klippel.de>
	setting board to ck
	setting baudrate from 115200 to 115200
	setting port from /dev/tty.usbserial to /dev/cu.usbserial-A9M9DV3R
	espcomm_upload_file
	espcomm_upload_mem
opening port /dev/cu.usbserial-A9M9DV3R at 115200
	tcgetattr
	tcsetattr
	serial open
opening bootloader
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
resetting board
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
trying to connect
	setting character timeout 0
	done
	setting character timeout 1
	done
	espcomm_send_command: sending command header
	espcomm_send_command: sending command payload
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
 
Finished with exit code 255
`

PianSom, 2017-12-19:
I haven't tried the gui version, only the command line. 

Like you, I was convinced I had bricked 3 new sonofffs (I was using Atom to flash, as I had many times before), but managed to recover them from a 255 error using esptool. I have no idea why it happened. But I got two new S20's yesterday and had a fail on Atom followed by success on esptool. 

[This](https://github.com/arendst/Sonoff-Tasmota/issues/463) thread has more suggestions that you may like to try (perhaps starting with reversing tx/rx in case they changed the board?).

joaohess, 2017-12-21:
Thank you for your help! Great thread; many people with similar problems, But no luck so far... 

I've tried switching pins; 
Keeping the button pressed while running the command;
Using command line esptool with
`esptool.py --port=/dev/cu.usbserial-A9M9DV3R -b 115200 write_flash 0x0 ~/Downloads/sonoff.bin`

No luck... I really don't know what can it be...

k4roshi, 2018-02-28:
`esptool.py --port=/dev/cu.usbserial-00000000 -b 115200 write_flash 0x0 ~/Downloads/Sonoff-Tasmota-development/.pioenvs/sonoff/firmware.bin`

In the same situation (Error 255 and nothing happening after booting afterwards), this solved my problem.