arendst/Tasmota #1392: Sensor polling time

Joeyhza, 2017-12-18:
Hello Again

perhaps I'm missing the obvious - if so please forgive me .

I've notices on my sensors that they are only updating every so often on my Domoticz interface. Usually the sensor reading timestamps are hours behind. Is there any way to check/ change the polling frequency on the console? I've gone through the Tasmota commands and can't find any reference to it.

davidelang, 2017-12-18:
the sensor data is reported every 'teleperiod', by default every 300 seconds. 
This can be set to anything down to 10 seconds.

gajotnt, 2017-12-18:
TelePeriod |   | Show current telemetry period in seconds
-- | -- | --
TelePeriod | 0 / off | Disable telemetry messages
TelePeriod | 1 | Reset telemetry period to user_config.h (TELE_PERIOD)
TelePeriod | <secs> | Set telemetry period between 10 and 3600 seconds

Joeyhza, 2017-12-19:
Thank you very much

closed 2017-12-19 by Joeyhza