arendst/Tasmota #1394: GPIO Pin number of Sonoff Touch T1

charinduedirisuriya, 2017-12-18:
I was wondering if anyone knew the pin numbers of the, 
Button PIN // touch Button
Led //blue led
Relay //relay pins accordingly (1 gang, 2 gang, 3 gang) 

I could not find any information about this anywhere.
Thanks in advance!

edit21, 2017-12-18:
Ben looking myself, though not found anything yet - please do update this if you find the info out, can update the Sonoff Touch page then :)

charinduedirisuriya, 2017-12-18:
@arendst After looking into sonoff_template.h i found the following, 
`{ "Sonoff T1 1CH",   // Sonoff T1 1CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0, 0,
     0, 0, 0,          // Flash connection
     0, 0,
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
  { "Sonoff T1 2CH",   // Sonoff T1 2CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     0,
     GPIO_REL2,        // GPIO05 Blue Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     0,
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },
  { "Sonoff T1 3CH",   // Sonoff T1 3CH (ESP8285)
     GPIO_KEY1,        // GPIO00 Button 1
     GPIO_USER,        // GPIO01 Serial RXD and Optional sensor
     0,
     GPIO_USER,        // GPIO03 Serial TXD and Optional sensor
     GPIO_REL3,        // GPIO04 Blue Led and Relay 3 (0 = Off, 1 = On)
     GPIO_REL2,        // GPIO05 Blue Led and Relay 2 (0 = Off, 1 = On)
     0, 0, 0,          // Flash connection
     GPIO_KEY2,        // GPIO09 Button 2
     GPIO_KEY3,        // GPIO10 Button 3
     0,                // Flash connection
     GPIO_REL1,        // GPIO12 Blue Led and Relay 1 (0 = Off, 1 = On)
     GPIO_LED1_INV,    // GPIO13 Blue Led (0 = On, 1 = Off)
     0, 0, 0, 0
  },`

Thanks!

edit21, 2017-12-18:
Aye they are there, I was (poorly) meaning reference to the  GPIO breakout locations on the touch its self (want to add sensors) :)

charinduedirisuriya, 2017-12-18:
I found the details,

Sonoff Touch T1 (1 Gang)
================
Button  - GPIO01
Relay and LED - GPIO12
LED - GPIO13

Sonoff Touch T1 (2 Gang)
================
Button 1 - GPIO01
Button 2 - GPIO09
Relay 1  and LED - GPIO12
Relay 2  and LED - GPIO05
LED - GPIO13

Sonoff Touch T1 (3 Gang)
================
Button 1 - GPIO01
Button 2 - GPIO09
Button 3 - GPIO10
Relay 1  and LED - GPIO12
Relay 2  and LED - GPIO05
Relay 3  and LED - GPIO04
LED - GPIO13

closed 2017-12-19 by charinduedirisuriya

dwindey1, 2018-04-16:
Thanks for the GPIO layout.  This got me straight to the next problem : GPIO1 = TX...  In full development it's pretty hard to get feedback without serial output.  Anyone has an idea on how to get around this ?

ttenterprises, 2018-04-24:
Just finished uploading my own custom firmware to my 3 gang sonoff after much frustration found the following otherwise all the othe info is correct thanks a million 
#Sonoff Touch T1 (3 Gang)
Button 1 - **GPIO00** not GPIO1
Button 2 - GPIO09
Button 3 - GPIO10
Relay 1 and LED - GPIO12
Relay 2 and LED - GPIO05
Relay 3 and LED - GPIO04
LED - GPIO13

colinl, 2018-11-16:
I was hoping for an answer to this as I think the button led should be inverted, on when the relay is off, so it is visible in the dark before putting the light on. Or just permanently on.

ascillato, 2018-11-16:
@colinl you are right, please do a PR

arendst, 2018-11-16:
This has been answered a year ago. 

The led is connected to the relay. When relay on / led is on. If you want the inverted situation you will have to rewire the led.

For what you want the wifi led has been selected as that's the only one under software control; when all relays are off the led is on to find the switch in the dark.

colinl, 2018-11-16:
Thanks @arendst, I had seen the information earlier that GPIO12 controlled the relay and the LED but I was not certain that meant that this was the _only_ way to control the LED.  I see from the wiki page for the device [1] that the relay and button led are connected to different pins on the EFM8BB1 so there might have been the opportunity to drive them separately.
Thanks again.

[1] https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU

dodavies, 2019-03-15:
Does this mean it is not possible to have the unit completely dark when led power is off.
In a bedroom I cannot handle the brightness of the device when the switches are off.
Is it possible to make the unit dark until switches are on ?

dodavies, 2019-03-16:
i stuck paper over the touch buttons, this darkened it enough :-)