arendst/Tasmota #1395: SSL/TLS 1.2 Support with Certificate Chain

Sunrise17, 2017-12-18:
Dear @arendst ,
It will be so fine to apply TLS 1.2 connection with certificate chain to encrypt auth and data. You can find some application in below link regarding to certificate chain during mqtt TLS connection (for AWS IoT or else), It is only necessary to attach 2 certificate (root.pem & deviceCertAndCACert.pem) and 1 key (deviceCert.key) to establish secure connection (all files in total 7KB).
[https://gist.github.com/eLement87/133cddc5bd0472daf5cb35a20bfd811e](url)

Secure connection is important, otherwise all published datas over 1883 port can be listened as seen in the attached photo which is received from Shodan.io.
[image]
If necessary, you can limit some services or functions to provide enough space for TLS Connection.
Thanks for your best support for Tasmota, looking forward to hear good developments soonest.

thunberg, 2017-12-18:
Seconded, would love to see this.

davidelang, 2017-12-18:
These devices do not have enough storage space to hold full certificates, let 
alone full certificate chains. The most they can do is hold the fingerprint of 
the approved certificate and check that.

Sunrise17, 2017-12-18:
@davidelang thanks for the quick response, during the registration of the device to AWS IoT, it is necessary to use these certificates. Right now, AWS does not accept only Fingerprints. As an alternative, is it not possible to apply on ESP32 ?

Sunrise17, 2017-12-18:
Or Can you minimize the FW (as excluding led light function and else) which only allows switching which can be used with these Certificates?

davidelang, 2017-12-18:
re-writing tasmota to run on the ESP32 is a frequent request, but it's a lot of 
work and currently there are no off-the-shelf devices shipping ith ESP32, so 
it's not a high priority.

AWS has no way of mandating how your device checks certificates (unless they are 
doing something stupid like changing them all the time or presenting different 
certificates with the same name)

What you would need to do is to connect with some tool that can show you the 
certificate fingerprint and then approve that fingerprint in the firmware.

Sunrise17, 2017-12-18:
I have got your point about ESP32. Tasmota is supporting TLS 1.2 connection?

thunberg, 2017-12-18:
This help?

https://github.com/copercini/esp8266-aws_iot

Sunrise17, 2017-12-18:
The problem is when Sonoff connected to AWS IoT first time, AWS is registering the certificate belonging to device automatically based on the certificates and key attached to Sonoff that's why Fingerprint can never solve the issue. If there is some tool, generate Fingerprint from these two cert and key and then extract that Fingerprint to the same certs and key in AWS, may be in that moment it can be possible. But i do not know such tool, which will extract certs and key from Fingerprint.

davidelang, 2017-12-18:
so you are talking about the client certificate on the device, not the 
certificate on AWS?

I'm not sure how the client cert is generated.

Sunrise17, 2017-12-18:
Yes sure, i am talking about the Client certs and keys which is attached to Sonoff (root.cert, client cert and client key). You can find all information in this link how to create these certs in title "Auto-registration of a Device Certificate Signed by Your CA Certificate", [AWS-Just In Time Registration](https://aws.amazon.com/blogs/iot/just-in-time-registration-of-device-certificates-on-aws-iot/)

When Sonoff first connects, AWS automatically recognizing that his certificate was already signed before connection (because Client certificates generated by SampleCA of AWS as seen in the Link) and then it starts automatically registers by itself. During registration mqtt connection breaks and connects again after registration complete. Kindly note that Mqtt Client Id of the device equals certificate Id.

Sunrise17, 2017-12-18:
After you create your certs, you can test your connection on MQTT.FX, it works well.

arendst, 2017-12-19:
As I understand the user has to create his own certs which then have to be installed on the (tasmota) life device.

Tasmota currently does not support FileHandling and SPIFFS needed to let a user simply upload his certs without recompiling. This would add FH code and SPIFFS area to the total amount of code usage.

Considering the current TLS implementation already makes tasmota running on it's toes (no stable webserver support) I think this is just a bridge too far.

Knowing I will be crucified (but soon to be reborn) I think using a local MQTT server makes TLS rather superfluous.

Sunrise17, 2017-12-19:
Dear @arendst , thanks for your response, i think these certs and key can easily be uploaded during the Flash of the Firmware instead adding some fileHanding property inside Tasmota. Right now, even you do not need to create WEB interface for managing cert files.