arendst/Tasmota #1397: Flashing NodeMCU

ameggs, 2017-12-19:
I apologize if this is not the correct forum to ask, however I am trying to flash this firmware to a Node MCU using Atom. Is this possible, if can some one point me in the direction of how to do it? Maybe a tutorial i have missed while searching?

ionciubotaru, 2017-12-19:
read the wiki - https://github.com/arendst/Sonoff-Tasmota/wiki/PlatformIO or https://github.com/arendst/Sonoff-Tasmota/wiki/Arduino-IDE or https://github.com/arendst/Sonoff-Tasmota/wiki/Esptool

Frogmore42, 2017-12-23:
You might get more help if you tell us what you have tried?  Exactly how did you try to flash the firmware?  Exactly what happened?

I found PlatformIO to be workable, but with NodeMCU you need to uncomment out the line in platformio.ini that says:
upload_resetmethod = nodemcu

Otherwise, you need to push the buttons in the right sequence to put the chip into bootloader mode at the right time.

You also need to uncomment out this line:
;env_default = sonoff

That way it will only try and download one of the bin files.

ameggs, 2017-12-23:
Thank You Frogmore42! This information helped. Thank you for taking the time to offer useful and helpful advice. If I may ask another question,  When using an ESP8266-01, it only has 2 GPIO ports available. 2 and 0, can I remapped the tasmota SW to allow me to use these 2 ports?

Frogmore42, 2017-12-23:
If you only have an ESP8266-01 then you do have to use GPIO 0 and 2.  If you select "WeMos D1 mini" in Configure Module --> Module Type, you will have access to both GPIO 0 and 2 (and a whole lot more that you wont actually be able to use on an ESP8266-01)

ameggs, 2017-12-23:
PERFECT!! Thanks again.  What I'm wanting to do is make some senor nodes with Motion and Temp. A few will have RGP LED on them. I dont know anything about arduino coding so using Tasmota to fill in the many blanks i have would be helpful and I can pick up small amounts of HOW TO on the way. Do you see any reason why my idea will not work.?

pepeEL, 2018-07-06:
Hi @ameggs 
Can you write me step by step how you write Tasmote firmware to nodeMCU becasue i can not do it... please help me.