arendst/Tasmota #1399: Flash pin locations of sonoff touch t1

charinduedirisuriya, 2017-12-19:
I bought all 3 types of the sonoff touch (1 gang, 2 gang, 3 gang), and im ready to upload the firmware.
But i couldn't find a picture of the RX, TX GND pins of the touch t1, i only found the method for uploading for the old version of the sonoff touch.

Does anyone have any links or pictures, so i can connect it to the right pins and upload the firmware.
[image]
[image]
[image]
[image]

zibous, 2017-12-20:
Same Problem here with the Sonoff Touch T1 (2 Gang).

Before flashing, I pressed the button S1 on the front, but did not work. 

Now I need help to find the  GPIO 0, so that I can put the device into flash mode.

Does the bottom and top part of the switch have to be mounted for flashing ?

```
Compiling .pioenvs/sonoff/src/sonoff.ino.o
Linking .pioenvs/sonoff/firmware.elf
Checking program size
text       data     bss     dec     hex filename
451920     8260   36088  496268   7928c .pioenvs/sonoff/firmware.elf
Building .pioenvs/sonoff/firmware.bin

Looking for upload port...
Auto-detected: /dev/cu.usbserial-A50285BI
Uploading .pioenvs/sonoff/firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error 255
....
```

[image]

Thanks

Bugzee, 2017-12-21:
Referring to the image of @zibous, the GPIO0 is connected to the resistor circled in red.
[image]

It's pin 15 of the ESP8285. You only need to ground it when you turn it on, I used a jumper cable. I've managed to flash with success and with custom testing sketches as well.

zibous, 2017-12-21:
@Bugzee
Thank you, has worked. 

But it was pretty tricky because I could not solder a wire to the R19 resistor. So I just tried the upper part of the device. With the bottom side on the table (R19 visible) and then before connecting the programmer with a wire the R19 pin to ground.
Connected the RS232 programmer to the serial ports and start flashing with Platform IO (Atom).

It worked, but actually you need 3 hands for it

Bugzee, 2017-12-21:
@zibous 
I'm using a USB-to-serial (FTDI) with a USB hub with switches, so I only need to toggle the switch in the USB while holding the Ground wire to the resistor.

FYI: if you want the touch to be fully functional, power the +5V as well. Powering up with only the 3.3V is not enough for the touch to work. Also the touch pin is 0 (pinMode(0,INPUT);).

zibous, 2017-12-21:
@Bugzee 
Thanks.  

So i do not need to hold ground wire to the resistor **during the hole flashing time** - only
when i connect the FTDI Adapter to the device ?

[image]

I use  3.3V -  no problem. 

Too bad - mirko/SonOTA do not work - , because this would be easier.

Bugzee, 2017-12-21:
@zibous 
Yes, you no need to hold it the whole time, I just hold it for about 5seconds.

jimmyw, 2018-01-06:
Did you manage to figure out how its wired up?  I started a wiki page with my findings here:

https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1

Please help me contribute!

charinduedirisuriya, 2018-01-06:
@jimmyw I actually did not, waiting for a brief explanation on how tot get it into flash mode. They they to ground gpi0, and some people say something else. So yeah. I was doing some other stuff these days was planning to complete this within this work.

jimmyw, 2018-01-06:
Im sure you can flash it by grounding gpio0, but then what? We still have to get the tasmota gpios right, and im not sure how these are wired up right now. I will continue investigate. If you know how, please let me know.

charinduedirisuriya, 2018-01-12:
@Bugzee , where exactly should i connect gpio0 to (resistor R19), i know i should ground it how exaclty should i do this. im using a 3.3v ftid adapter. Could you possibly post a picture on where to connect it from and to where?

charinduedirisuriya, 2018-01-12:
Would connecting GPIO0 and 0v (right hand side) do the trick? @zibous @edit21 @Bugzee @jimmyw @davidelang 

[image]

webtop, 2018-01-12:
Yes, it is GND in this case

webtop, 2018-01-12:
The socket on the right there, also has a GND and makes it easy to plug in a wire and then unplug it after a few seconds of boot. You'll just need to probe it to find which one is GND

zibous, 2018-01-12:
I am using the GND from the FTDI adapter. For flashing, I just hold the pins to the solder joints. I do not have to solder that. After the software is on the device, I can simply flash over OTA.

[image]

charinduedirisuriya, 2018-01-13:
I managed to flash it successfully. Does anyone know what the small button in the bottom does, and the pin number for it. And also i noticed that when long pressing the touch button it makes a beep sound and lights up some extra led's (this happens when pressing the small button on the buttom aswell). It could be nice to use those extra led's attached on the circuit and to use that beep sound only if i knew the pin numbers for it. PS im running my own custom sketch and i did not programe the long press touch button beep either the touch button on the buttom. I guess that the original firmware is still active.

charinduedirisuriya, 2018-01-13:
@zibous @edit21 @Bugzee @jimmyw @webtop Please do let me know if anyone knows about this.

Bugzee, 2018-01-14:
@charinduedirisuriya Read here. https://github.com/arendst/Sonoff-Tasmota/wiki/Sonoff-T1-UK-and-T1-EU
It has all the information.

Kallb123, 2018-01-15:
So I managed to flash an S20 socket and an RF Bridge, but I'm struggling to grasp what's happening in the T1. I didn't realise this was all currently being explored and evolving!

Just to provide some clarity for myself and anyone else that may be struggling, am I correct in this process:

1. Connect GPIO0 resistor to any GND (0v) (could be programmer or EFM8BB1 connector, or is it a specific GND that must be used?),
2. Connect GND, Tx, Rx, 3v on ESP8285 connectors to GND, Rx, Tx, 3v on programmer,
3. Disconnect GPIO0 from GND,
4. Hold touch button 1 and press reset button (labelled 4 on wiki images) to reboot into flash mode,
5. Click upload in PlatformIO,
6. Disconnect programmer

Things I'm not sure about:

1. Have I got step 3 and 4 reverse? There seems to be a reboot involved in 4, so does that mean GPIO0 should still be grounded?
2. In the wiki it states "The front circuit board should be disconnected from the rear relay board to prevent power draw upsetting the flashing process". Is that achieved by connecting GPIO0 to GND?
3. How exactly do these touch buttons work?? They seem quite tempermental with the glass front on, is it light sensative or pressure sensative? Does a small press in the centre trigger it or do I need to cover the white area?

Kallb123, 2018-01-15:
I went through the process earlier and thought I should document it for anyone else at my skill level.

1. Set up software environment (maybe change poweron state to 1 if you want to fallback to external switch, set up wifi),
2. Connect GPIO0 resistor to any GND (0v) (can use the empty EFM8BB1 connector),
3. Connect GND, Tx, Rx, 3v on ESP8285 connectors to GND, Rx, Tx, 3v on programmer,
4. Hold touch button 1 and press reset button (labelled 4 on wiki images) to reboot into flash mode,
5. Disconnect GPIO0 from GND,
6. Click upload in PlatformIO,
7. Disconnect programmer. 

That worked perfectly first time for me and now my setup is working almost how I want it :) 

I found it quite hard to keep GPIO0 connected to GND whilst also plugging the device into USB and then pressing two buttons at once... But I did manage it.

geekypenguin, 2018-01-22:
@Kallb123  Please read the wiki on flashing. There is no need to connect anything to GPIO0 any more as this is achieved via the 1+4 button press method.
The touch buttons are capasitive touch sensors
And in answer to question 2. No. there are two circuit boards inside a T1, the one with the buttons and the esp, and another with the mains connections and relays etc. you must physically separate the two before flashing, and reconnect them and power from the mains when you come to configure them.

Kallb123, 2018-01-22:
@geekypenguin I don't really understand the wiki explanation, so I just gave it a go and everything went perfectly. I guess I was just lucky if I managed to succeed incorrectly? 

I don't know how you would "seperate the two circuits"? I was under the impression that I achieved that by grounding GPIO0. The wiki says nothing about how to disconnect the two circuits, just that it should be done. I guess I was meant to unsolder something? Is it possible to have a diagram like with the pin numbers, showing the process or disconnecting and reconnecting the circuits?

geekypenguin, 2018-01-22:
@Kallb123  the pictures at the top of this thread (specifically the 3rd one) shows the two boards separated. 
Once the front cover is removed the front circuit board just unclips from the rear relay board. the two are connected by the connector labelled J2.
The fact you have had access to the resistor for GPIO0, you must have separated the two already.

Grounding GPIO0 and the 1+4 button method do exactly the same thing, no luck involved as such, you have just doubled up on the steps to put the ESP into flashing mode.

Kallb123, 2018-01-22:
@geekypenguin that makes perfect sense. Yes I had already disconnected the front board and I didn't give the other one any thought. I misunderstood it all as being the front/back of the circuit board.

mattpurnell, 2018-01-25:
Can someone please clarify the LED patterns - I am not sure how to tell whether the device is successfully in the correct flashing mode, or the mode prior to the reset (before pushing the two buttons)

geekypenguin, 2018-01-25:
When it's ready to be flashed, the main led behind the WiFi symbol is off, and the 3 where the touch buttons are, are very dimly lit.
If the main led is on, or is flashing, then the esp has booted and you are not in flashing mode

mattpurnell, 2018-01-26:
Thanks Geek! Maybe "thanks penguin" is more polite.

Yes I can confirm, on my single-pad variant, that the touchpad should be dimly lit, and the main LED should be off (once you have successfully reset with the two buttons).
I personally had trouble until found a way to keep the keep the GPio0 grounded the whole time. I was using breadboard wire pins, and I think the only way these will work, is if you hold the T1 board on an angle so the breadboard wire pins are held tight - it was difficult otherwise, to ensure they were making good contact throughout.

All good after that!

geekypenguin, 2018-01-26:
You don't need to ground gpio 0 any more with the method on the wiki

jeylites, 2018-02-15:
@geekypenguin 

Just making sure I understood the wiki explanation

Should the Arduino IDE set as 74880 baud rate when flashing?

And Button 1 and 4 should be pressed and release the same time? Because on wiki it say just release button 1?

geekypenguin, 2018-02-15:
Flashing instructions are the same as a basic.
The bootloader is communicated with at 74880 baud but you should flash at 112500.

Oh sorry that should be hold 1,  press 4 (the little one) release 4 then release 1. I'll update the wiki if it's wrong

jeylites, 2018-02-15:
@geekypenguin I got it flashed to V.5.12.0b but now I’m on another issue. The setting are not saving and, the touch T1 is rebooting every few seconds. Basically no control of it.

geekypenguin, 2018-02-17:
I've not tried that version myself to test.

There is an option command to make config changes persistent but use this carefully as multiple reboots is a recovery method if you mess something up 

I assume you have read the wiki on how not to power the so off module? More often than not, this will be the cause of rebooting problems. I have only had one out of 15 which kept rebooting when powered correctly but this was a hardware problem and was replaced.

geekypenguin, 2018-02-18:
Yes that's correct @alton76. I hope the wiki is clear enough on this...

KristerBC, 2018-03-25:
Hi felow Tasmotarians,
I just bought a T1 for my roller shutters. However, I can not figure out how to flash it. The layout of my bought doesn't look like any layouts I can find on the internet.

[image]

When I press and hold button4 and one of the white buttons, it just reboots into normal mode, but not into flash mode. Have I done the wiring correctly ?

PlatformIO gives the following : 
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed

geekypenguin, 2018-03-25:
That's the EU version rather than the UK version discussed here and on the wiki. As far as I know there is another thread for the EU version

geekypenguin, 2018-03-25:
Were you pressing the left or the right white button? (In your picture)
I don't have an EU version to try but the UK version only worked when button 1 was held when pressing 4 (this is the left hand button you see).
Failing that, you'll have to revert to the jumper to gpio0 method

KristerBC, 2018-03-25:
On the wifi, the way to flash it doesn't seem to differ, but even when I look at the German videos on youtube, the layout is not the same.

I have tried both buttons. I will try the GPIO0 method, thanks

KristerBC, 2018-03-25:
Ok, so with the GPIO0 method, I do manage to go into flash mode(at least only the two touch buttons are lid slightly and the WIFI led is off.

However I still cannot install the firmware, does this look familiar ?

Uploading .pioenvs\sonoff-NL\firmware.bin
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
*** [upload] Error -1
 [ERROR] Took 24.26 seconds
 
 [SUMMARY]
Environment sonoff              [SKIP]
Environment sonoff-minimal      [SKIP]
Environment sonoff-xxl          [SKIP]
Environment sonoff-DE           [SKIP]
Environment sonoff-ES           [SKIP]
Environment sonoff-FR           [SKIP]
Environment sonoff-HU           [SKIP]
Environment sonoff-IT           [SKIP]
Environment sonoff-PL           [SKIP]
Environment sonoff-PT           [SKIP]
Environment sonoff-RU           [SKIP]
Environment sonoff-CN           [SKIP]
Environment sonoff-NL           [ERROR]
 [ERROR] Took 24.26 seconds

geekypenguin, 2018-03-25:
Try reversing the tx and Rx pins and try again.
Failing that try the Arduino method. I've not used platformio to flash mine so can't really help with that one

geekypenguin, 2018-03-25:
Or try and build Inna different language. It may be an issue with the NL pack

KristerBC, 2018-03-25:
Hi again,
sorry for the late response. I managed to flash it, but didn't want to touch anything on my computer in the meantime.

The trick was to reverse the TX and RX.

Thank you so much.

Little bonus question: Is there anything wrong in using a UK version in France ? I just looked at my T1s and I also have a UK version as well.

geekypenguin, 2018-03-25:
Glad it's sorted.

And I don't see why not, theyre just a different shape to fit the round back boxes found on the continent as opposed to the square ones found in the uk

KristerBC, 2018-03-25:
Ok I see,
If it's only the shape, then it doesn't matter. The EU ones don't look like the normal French shapes anyway.

closed 2018-04-22 by charinduedirisuriya

abbasmzhr, 2018-10-14:
> @Bugzee
> Thank you, has worked.
>
> But it was pretty tricky because I could not solder a wire to the R19 resistor. So I just tried the upper part of the device. With the bottom side on the table (R19 [...]

@zibous Can you please explain about R19, Where's R19 in FTDI & in T1 ?