arendst/Tasmota #1402: Sonoff S20 turns itself on

cyablo, 2017-12-19:
Hi,

every few days my sonoff s20 tuns itself on. I'm using FHEM and Mosquitto with latest Tasmota Firmware on the S20. As soon as is saw it, I turned it of immediately. According to my Access-Point's, there was no WiFi disconnect or something like that. Logs say:

Tasmota:
`20:44:17 MQT: Verbindungsversuch...
20:44:23 MQT: Verbindung fehlgeschlagen aufgrund von 192.168.1.21:1883, rc -2. Wiederversuch in 10 sek
20:44:24 UPP: Multicast (wieder-)verbunden
20:44:34 MQT: Verbindungsversuch...
20:44:44 MQT: verbunden
20:44:49 RSL: tele/tasmota2/LWT = online
20:44:54 RSL: cmnd/tasmota2/POWER = 
20:45:03 UPP: Multicast (wieder-)verbunden
20:45:04 MQT: Verbindungsversuch...
20:45:04 MQT: verbunden
20:45:04 MQT: tele/tasmota2/LWT = online (beibehalten)
20:45:04 MQT: cmnd/tasmota2/POWER = 
20:45:04 MQT: stat/tasmota2/ERGEBNIS = {"POWER":"ON"}
20:45:04 MQT: stat/tasmota2/POWER = ON
20:45:04 MQT: stat/tasmota2/ERGEBNIS = {"POWER":"ON"}
20:45:04 MQT: stat/tasmota2/POWER = ON
20:45:04 UPP: Multicast (wieder-)verbunden
20:45:26 MQT: stat/tasmota2/ERGEBNIS = {"POWER":"OFF"}
20:45:26 MQT: stat/tasmota2/POWER = OFF
20:45:53 MQT: tele/tasmota2/STATUS = {"Zeit":"2017.12.19 20:45:53","Laufzeit":166,"VCC":3.230,"POWER":"OFF","WLAN":{"AP":1,"SSID":"net-2-4","RSSI":66,"APMac":"XX.XX..."}}`

Mosquitto:
`1513712655: Client tasmota2 has exceeded timeout, disconnecting.
1513712655: Socket error on client tasmota2, disconnecting.
1513712678: New connection from 192.168.1.52 on port 1883.
1513712681: New client connected from 192.168.1.52 as tasmota2 (c1, k15, u'DVES_USER').
1513712702: Client tasmota2 has exceeded timeout, disconnecting.`

cyablo, 2017-12-21:
Happened again, turned it off immediately, here are the logs:

Tasmota:
> Dec 21 21:06:06 tasmota2 ESP-WIF: verbunden
Dec 21 21:06:26 tasmota2 ESP-WIF: prüfe Verbindung...
Dec 21 21:06:26 tasmota2 ESP-WIF: verbunden
Dec 21 21:06:46 tasmota2 ESP-WIF: prüfe Verbindung...
Dec 21 21:06:46 tasmota2 ESP-WIF: verbunden
Dec 21 21:07:06 tasmota2 ESP-WIF: prüfe Verbindung...
Dec 21 21:07:06 tasmota2 ESP-WIF: verbunden
Dec 21 21:07:26 tasmota2 ESP-WIF: prüfe Verbindung...
Dec 21 21:07:26 tasmota2 ESP-WIF: verbunden
Dec 21 21:08:28 tasmota2 ESP-UPP: Multicast deaktiviert
Dec 21 21:08:30 tasmota2 ESP-MQT: verbunden
Dec 21 21:08:30 tasmota2 ESP-MQT: tele/tasmota2/LWT = online (beibehalten)
Dec 21 21:08:30 tasmota2 ESP-MQT: cmnd/tasmota2/POWER =
Dec 21 21:08:30 tasmota2 ESP-MQT: subscribe to cmnd/tasmota2/#
Dec 21 21:08:30 tasmota2 ESP-MQT: subscribe to cmnd/sonoffs/#
Dec 21 21:08:30 tasmota2 ESP-MQT: subscribe to cmnd/tasmota2/#
Dec 21 21:08:30 tasmota2 ESP-RSL: empfangenes topic cmnd/tasmota2/POWER, Datengröße 2, Daten on
Dec 21 21:08:30 tasmota2 ESP-RSL: Gruppe 0, Index 1, Befehl POWER, Daten on
Dec 21 21:08:30 tasmota2 ESP-MQT: stat/tasmota2/ERGEBNIS = {"POWER":"ON"}
Dec 21 21:08:30 tasmota2 ESP-MQT: stat/tasmota2/POWER = ON
Dec 21 21:08:30 tasmota2 ESP-RSL: empfangenes topic cmnd/tasmota2/POWER, Datengröße 2, Daten on
Dec 21 21:08:30 tasmota2 ESP-RSL: Gruppe 0, Index 1, Befehl POWER, Daten on
Dec 21 21:08:30 tasmota2 ESP-MQT: stat/tasmota2/ERGEBNIS = {"POWER":"ON"}
Dec 21 21:08:30 tasmota2 ESP-MQT: stat/tasmota2/POWER = ON
Dec 21 21:08:31 tasmota2 ESP-CFG: in Flash gespeichert am F8, zählen 835, Bytes 1648
Dec 21 21:08:31 tasmota2 ESP-UPP: Multicast (wieder-)verbunden
Dec 21 21:08:42 tasmota2 ESP-RSL: empfangenes topic cmnd/tasmota2/POWER, Datengröße 3, Daten OFF
Dec 21 21:08:42 tasmota2 ESP-RSL: Gruppe 0, Index 1, Befehl POWER, Daten OFF
Dec 21 21:08:42 tasmota2 ESP-MQT: stat/tasmota2/ERGEBNIS = {"POWER":"OFF"}
Dec 21 21:08:42 tasmota2 ESP-MQT: stat/tasmota2/POWER = OFF
Dec 21 21:08:43 tasmota2 ESP-CFG: in Flash gespeichert am F7, zählen 836, Bytes 1648
Dec 21 21:08:44 tasmota2 ESP-WIF: prüfe Verbindung...
Dec 21 21:08:44 tasmota2 ESP-WIF: verbunden
Dec 21 21:09:01 tasmota2 ESP-UPP: Multicast deaktiviert
Dec 21 21:09:01 tasmota2 ESP-MQT: Verbindungsversuch...
Dec 21 21:09:01 tasmota2 ESP-MQT: verbunden
Dec 21 21:09:01 tasmota2 ESP-MQT: tele/tasmota2/LWT = online (beibehalten)
Dec 21 21:09:01 tasmota2 ESP-MQT: cmnd/tasmota2/POWER =
Dec 21 21:09:01 tasmota2 ESP-MQT: subscribe to cmnd/tasmota2/#
Dec 21 21:09:01 tasmota2 ESP-MQT: subscribe to cmnd/sonoffs/#
Dec 21 21:09:01 tasmota2 ESP-MQT: subscribe to cmnd/tasmota2/#
Dec 21 21:09:01 tasmota2 ESP-RSL: empfangenes topic cmnd/tasmota2/POWER, Datengröße 3, Daten OFF
Dec 21 21:09:01 tasmota2 ESP-RSL: Gruppe 0, Index 1, Befehl POWER, Daten OFF
Dec 21 21:09:01 tasmota2 ESP-MQT: stat/tasmota2/ERGEBNIS = {"POWER":"OFF"}
Dec 21 21:09:01 tasmota2 ESP-MQT: stat/tasmota2/POWER = OFF
Dec 21 21:09:01 tasmota2 ESP-RSL: empfangenes topic cmnd/tasmota2/POWER, Datengröße 3, Daten OFF
Dec 21 21:09:01 tasmota2 ESP-RSL: Gruppe 0, Index 1, Befehl POWER, Daten OFF
Dec 21 21:09:01 tasmota2 ESP-MQT: stat/tasmota2/ERGEBNIS = {"POWER":"OFF"}
Dec 21 21:09:01 tasmota2 ESP-MQT: stat/tasmota2/POWER = OFF

Mosquitto:
> Dec 21 21:06:55 cybjessie1 mosquitto[24080]: Received PINGREQ from tasmota2
Dec 21 21:06:55 cybjessie1 mosquitto[24080]: Sending PINGRESP to tasmota2
Dec 21 21:07:10 cybjessie1 mosquitto[24080]: Received PINGREQ from tasmota2
Dec 21 21:07:10 cybjessie1 mosquitto[24080]: Sending PINGRESP to tasmota2
Dec 21 21:07:32 cybjessie1 mosquitto[24080]: Client tasmota2 has exceeded timeout, disconnecting.
Dec 21 21:07:32 cybjessie1 mosquitto[24080]: Socket error on client tasmota2, disconnecting.
Dec 21 21:07:32 cybjessie1 mosquitto[24080]: Sending PUBLISH to NetMQTTpm32751 (d0, q0, r0, m0, 'tele/tasmota2                                                                                                                               /LWT', ... (7 bytes))
Dec 21 21:07:46 cybjessie1 mosquitto[24080]: Received PINGREQ from NetMQTTpm32751
Dec 21 21:07:46 cybjessie1 mosquitto[24080]: Sending PINGRESP to NetMQTTpm32751
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: New connection from 192.168.1.52 on port 1883.
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: New client connected from 192.168.1.52 as tasmota2 (c1, k15, u'DV                                                                                                                               ES_USER').
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: Sending CONNACK to tasmota2 (0, 0)
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: Received PUBLISH from tasmota2 (d0, q0, r1, m0, 'tele/tasmota2/LW                                                                                                                               T', ... (6 bytes))
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: Sending PUBLISH to NetMQTTpm32751 (d0, q0, r0, m0, 'tele/tasmota2                                                                                                                               /LWT', ... (6 bytes))
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: Received PUBLISH from tasmota2 (d0, q0, r0, m0, 'cmnd/tasmota2/PO                                                                                                                               WER', ... (0 bytes))
Dec 21 21:08:28 cybjessie1 mosquitto[24080]: Received SUBSCRIBE from tasmota2
[... 2 lines cut ...]
[... cut ...]
Dec 21 21:08:41 cybjessie1 mosquitto[24080]: Received PUBLISH from tasmota2 (d0, q0, r0, m0, 'stat/tasmota2/PO                                                                                                                               WER', ... (3 bytes))
Dec 21 21:08:41 cybjessie1 mosquitto[24080]: Sending PUBLISH to NetMQTTpm32751 (d0, q0, r0, m0, 'stat/tasmota2                                                                                                                               /POWER', ... (3 bytes))
Dec 21 21:08:46 cybjessie1 mosquitto[24080]: Received PINGREQ from NetMQTTpm32751
Dec 21 21:08:46 cybjessie1 mosquitto[24080]: Sending PINGRESP to NetMQTTpm32751

Frogmore42, 2017-12-24:
It looks like your Wi-Fi connection is dropping out and then you have your system configured to replay MQTT commands.  As I recall, there is an option in the wiki for that and it states that you can have issues like this.  It does not appear that there is a ghost, or that your device is possessed, or that someone is hacking you 😀

cyablo, 2017-12-26:
Turned retain off in FHEM to see if that helps. Wifi signal is pretty ok, LEDE tells me -47dbm on the S20 and tasmota itself gives a RSSI of 66.

closed 2018-01-26 by cyablo

HobbytronicsPK, 2019-05-13:
Have a look on to [Ghost / Random Switching on Sonoff Devices (MQTT Retained)](https://blog.hobbytronics.pk/sonoff-tasmota-tips-tricks/#Ghost-Random-Switching)