arendst/Tasmota #1405: Temperature data only?.

dave305, 2017-12-21:
I have a sonoff, tasmota DHT11 and mqtt running. DHT11 is on GPOI 14 and I can get status data with Status 10 but I want just temperature data not a string. Can I query just temp data or do I have to parse the string? If I have to parse the string how do I go about it. I have been trying to understand RegEX, but I don't have clue :). Any help would be a big seeing that I would like to get this running before Friday when we have very cold front coming and I am not going to home (This is for my dogs house).

ionciubotaru, 2017-12-21:
The string is in json format, use a json parser in your HA
There is no way to receive the temperature only.
DHT11 needs 5V, you can get wrong values if you supply 3V3. Better use DHT22

dave305, 2017-12-21:
Thanks for the reply. I will look into the json parser. I will get a DHT22, but for now I only have 3.3v and a DHT11. I have a DHT21 AM2301, but couldn't get a reading with sonoff/tasmota (It is not even listed).

dave305, 2017-12-23:
I was wondering if anyone could help show me how to go about parsing this string to get the temperature data?, I have been trying to get objectcleaner to do it, but I am not having much luck. I will worry about the Humidity later.
{"StatusSNS":{"Time":"2017-12-23T01:35:31","DHT11":{"Temperature":53.6,"Humidity":67.0},"TempUnit":"F"}}
I have been looking into this for a day now and am no farther along than when I started.

Frogmore42, 2017-12-23:
It might help if you told people what other software you are using and what you are trying to do.  Are you using Node-Red or something else?  For Node-Red, I have found it works to use the json converter:
Converts between a JSON string and its JavaScript object representation, in either direction.
And then you can extract the data you want using standard JavaScript object notation, e.g.
msg.payload.DHT11.Temperature

dave305, 2017-12-23:
Sorry about the lack of info :). Yes I am using node-red and I am trying to control a sonnoff that controls my dog house light. I want the light to come on at something like 40F and off at 45F. I don't have time to work on it now, but wanted to thank you for your time. I will get on it after Christmas and reply with the results.  I was not converting to JS in my flow.

Frogmore42, 2017-12-24:
Here is what I am currently using to control temperature (inside a function block in Node-Red)

`var lowLimit = 38; // turn on set point
var hysteresis = 0.5;
var tooHotLimit = 88; // turn off set point
var newMsg = {};
var keys = (Object.keys(msg.payload));
var numKeys = keys.length;
var isOn = context.get('isOn') || false;
var isTooCold = false;
var isTooHot = false;
var isAllOkay = true;

for (var i = 0; i < numKeys; i++) {
    if (keys[i].toLowerCase().startsWith("temp")) {
        if (msg.payload[keys[i]] < lowLimit) {
            isTooCold = true;
            isAllOkay = false;
        }
        else if (msg.payload[keys[i]] < lowLimit + hysteresis) {
            isAllOkay = false;
        }
        else if (msg.payload[keys[i]] > tooHotLimit) {
            isTooHot = true;
            isAllOkay = false;
        }
    }
}

if (isTooCold && !isTooHot) {
    newMsg.payload = "On";
} 
else if (isTooHot) {
    newMsg.payload = "Off";
}
else if (isAllOkay && isOn) {
    newMsg.payload = "Off";
}

if (newMsg.payload) {
    if (newMsg.payload === "On") {
        context.set('isOn', true);
    } else {
        context.set('isOn', false);
    }
    return newMsg;
}
`

GitHub doesn't like to show the code in a code box for some reason (doesn't like the line endings and spaces) but it should give you an idea of how this can be done.  It expects the payload coming into it to contain one or more temperatures.  It then looks at the temps and sees if it should turn on the heater, turn it off or leave it alone.  It does not actually get the state of the heater from the heater, but rather from what it last told it to do.  When it is too cold it will always turn on the heater.  When it is too hot (which should be set to a significantly higher number) it will always turn off the heater.  If it is warmer than the turn on point plus the hysteresis value, it will turn off the heater if it is not on already.  For your settings you should use
`var lowLimit = 40;`
`var hysteresis = 5;`
`var tooHotLimit = 88; // turn off set point`

The output of this node goes to an MQTT output node that is the power topic for the SonOff device controlling the light.

Frogmore42, 2017-12-24:
And here is what that looks like:

[image]

I've got 4 DS18B20 sensors on various parts of a sink in an unheated tack room.  The heater relay turns on a string of incandescent lights 5@5W for 25W, a foot warmer mat (100W or so), a 200W personal heater.  I originally started with just the lights and had the set point at 35.  That was generally enough, at least until the time it got really cold.  You can see from the graph that the control is working well.

I have another one set up to control a pump to circulate hot water through the pipes when they get too cold.

dave305, 2017-12-26:
You gave me more than i could of asked for :). I got a reading on my temp with:

var value = msg.payload.StatusSNS.DHT11.Temperature;
msg.payload = value;
return msg;

Your json converter gave my the msg.payload that I was missing. I am no JS coder (newbie), but I am starting to see patterns in the code like I now see that the msg.payload above was nested three deep. Now I can use this for other sensors I want to read. I am going to play with your code and see what I can do with it :). Again thank you for all your time and effort.

dave305, 2017-12-27:
[image]

Here is my working flow. I did the temp monitoring with the switch node. I need to add a chart, but so far it seems to work fine. Thank you again, couldn't have done it w/o you.

closed 2018-05-10 by stale[bot]

SAM26K, 2018-12-21:
Is there a simple way to parse the data from a sonoff to individual variables?     i.e. one for temp one for humidity etc?   Trying to interface Tasmota with Mycodo and need to return one parameter from Tasmota for each data parameter for Mycodo using python script.  Looking for a specific answer that can parse this:  {"StatusSNS":{"Time":"2018-12-21T13:06:46","AM2301":{"Temperature":69.1,"Humidity":54.1},"TempUnit":"F"}}

SAM26K, 2018-12-22:
Want to us python's json lib to parse the data in the comment above.  Once I  have a working script file I can import it to Mycodo (automation gardening app).