arendst/Tasmota #1406: Software/System restart - what is normal

Paulf007, 2017-12-21:
Hello gents. 

What should be a normal "Software/System restart" for SonOff Basic . One of my switches seems to be restarting quite often ( I actually had to setup a log to see it ) and it seems like it is restarting 2 to 3 times an hour with random intervals.  Is there anything bad to this as well?
Regards
Paul

ionciubotaru, 2017-12-21:
it could be a power supply problem, if other switches works well and only one is restarting is a hardware problem. 
Or a poor wifi signal can cause lots of disconnections followed by restart ....

MatzeJoerling, 2017-12-21:
i have uptime greater than 5 days...

Courty40, 2017-12-21:
just checked on of my remote ones and its reading:
271 Hours
Software/System restart

Shouldn't restart by itself - I had a 4ch unit that kept doing that and it was a fault with the PSU

Courty

Paulf007, 2017-12-21:
Interesting I have a odd 9 units running and none of them has been running for more that 3h , I would need to log all of them but it seems like they are restarting every three hours. Is there a way to debug the info more to see why they are rebooting. I can imagine that all of thier PSU is faulty.

Paulf007, 2017-12-24:
Ok I have runs some logging and one error that I get for rebooting is 

"22:23 {"RestartReason":"Fatal exception:28 flag:2 (EXCEPTION) epc1:0x402077bf epc2:0x00000000 epc3:0x00000000 excvaddr:0x00000008 depc:0x00000000"}"

Where can I check what this error is?

arendst, 2017-12-24:
See https://github.com/esp8266/Arduino/blob/master/doc/exception_causes.rst

Frogmore42, 2017-12-25:
If you are seeing frequent restarts in an area where other devices are working fine, it points to a  hardware issue.  I have multiple sonoffs and also NodeMCU devices most of them work fine.  But, I have one or two that work somewhat okay most of the time, but reset for strange reasons.  I am guessing they are defective.  Since they are not that expensive, and I have spares, I have not spent much time trying to figure exactly what is wrong with them.

closed 2018-05-08 by stale[bot]