arendst/Tasmota #1407: sonoff switch T1 lost wifi... why?

Igor122017, 2017-12-21:
hello. i have problem with sonoff T1 2Ch. I reload sonoff with Sonoff-Tasmota. all be, ok... then across 2-3 minutу, I can not see this sonoff on web, lost contact with router. If I off then on, sonoff woking 2-3 minute and lost wifi. before, about day ago was all good. help me....

Igor122017, 2017-12-21:
Help me please, I do not understand whst to do. where mistake or problem? before sonoff wifi switch in original working perfect. if mistake in  Sonoff-Tasmota, i have another wifi switch T1 1 gang, where all working all right!

reloxx13, 2017-12-21:
which firmware version did you flash?

Try to flash the latest developer version.

Turn on more debug logging for weblog and post it here.

turn off all sensors in the user_config.h enter your wifi data in the file and build the firmware yourself.

how long is the distance between sonoff and wifi router? could it be a bad signal?

Igor122017, 2017-12-21:
Program Version | 5.10.0a
2017-12-19T18:50:19
2_3_0/1.5.3(aec24ac9)
98 at FA000
26

1.version 5.10.0a
2.long about 6 метре, near have anothe switch sonoff wifi and working perfect. and signal 78%
3.i am newer , not proffy.  i am a little do not understand what need to do...

Igor122017, 2017-12-21:
how turn on debug logging?

Igor122017, 2017-12-21:
how build the firmware yourself?

reloxx13, 2017-12-21:
read the [wiki](https://github.com/arendst/Sonoff-Tasmota/wiki) and this [Beginner Guide](https://github.com/arendst/Sonoff-Tasmota/wiki/Beginner-Guide---Create-your-own-Firmware-Build)

Igor122017, 2017-12-21:
Thank you for an answer. I found out a next situation. I today again sewed switch and problem not settled.

a problem is in the following, I did tuning, set the switch of sonoff T1 2ch and influenced MQTT/ after 5-10 minutes all works and spontaneously  there is an upcast of all tuning. what can a problem be in a switch or in the program? the repeated unloading today passed cleanly, errors at loading not discovered.... thanks in advance.

Igor122017, 2017-12-21:
00:00:00 Project sonoff Sonoff (Topic sonoff_T1_2CH, Fallback MD client3, GroupTopic sonoffs) Version 5.10.0b
00:00:00 WIF: Connecting to AP1 MGTS_GPON_3233 in mode 11G as sonoff_T1_2CH-3366...
00:00:05 WIF: Connected
00:00:05 DNS: Initialized
00:00:05 HTP: Web server active on sonoff_T1_2CH-3366.local with IP address 192.168.1.67
00:00:07 MQT: Attempting connection...
00:00:07 MQT: Connected
00:00:07 MQT: tele/sonoff_T1_2CH/LWT = Online (retained)
00:00:07 MQT: cmnd/sonoff_T1_2CH/POWER =
00:00:07 MQT: tele/sonoff_T1_2CH/INFO1 = {"Module":"Sonoff T1 2CH","Version":"5.10.0b","FallbackTopic":"MD client3","GroupTopic":"sonoffs"}
00:00:07 MQT: tele/sonoff_T1_2CH/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff_T1_2CH-3366","IPAddress":"192.168.1.67"}
00:00:07 MQT: tele/sonoff_T1_2CH/INFO3 = {"RestartReason":"Software/System restart"}
02:02:11 MQT: tele/sonoff_T1_2CH/UPTIME = {"Time":"2017-12-22T02:02:11","Uptime":1}
02:02:12 MQT: stat/sonoff_T1_2CH/RESULT = {"POWER1":"OFF"}
02:02:12 MQT: stat/sonoff_T1_2CH/POWER1 = OFF
02:02:12 MQT: stat/sonoff_T1_2CH/RESULT = {"POWER2":"OFF"}
02:02:12 MQT: stat/sonoff_T1_2CH/POWER2 = OFF
02:02:19 MQT: tele/sonoff_T1_2CH/STATE = {"Time":"2017-12-22T02:02:19","Uptime":1,"Vcc":3.152,"POWER1":"OFF","POWER2":"OFF","Wifi":{"AP":1,"SSId":"MGTS_GPON_323
**3","RSSI":88,"APMac":"78:81:02:95:9E:91"}}
02:02:32 MQT: stat/sonoff_T1_2CH/RESULT = {"Reset":"Reset and Restarting"}
02:02:32 CFG: Use defaults
02:02:40 APP: Restarting**

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d

Igor122017, 2017-12-26:
My problem can not be solved at all, I did some sonofs, everyone works. And this sonoff switch 2CH  all the time problem. Help me please, what's the problem, why is it constantly overloaded? permanently lost connection with the MQTT...

02:10:58 MQT: Connect failed to domus1:1883, rc -2. Retry in 10 sec
02:10:58 CMD: mqtthost 192.168.1.65
02:10:58 RSL: stat/sonoff/RESULT = {"MqttHost":"192.168.1.65"}
02:11:00 APP: Restarting

 ets Jan  8 2013,rst cause:1, boot mode:(3,7)

load 0x4010f000, len 1384, room 16
tail 8
chksum 0x2d
csum 0x2d
v09826c6d
~ld
�
00:00:00 Project sonoff Sonoff (Topic sonoff, Fallback DVES_998D26, GroupTopic sonoffs) Version 5.10.0b
00:00:00 WIF: Connecting to AP1 MGTS_GPON_3233 in mode 11N as sonoff-3366...
00:00:06 WIF: Connected
00:00:06 DNS: Initialized
00:00:06 HTP: Web server active on sonoff-3366.local with IP address 192.168.1.67
02:11:09 MQT: Attempting connection...
02:11:09 MQT: Connected
02:11:09 MQT: tele/sonoff/LWT = Online (retained)
02:11:09 MQT: cmnd/sonoff/POWER =
02:11:09 MQT: tele/sonoff/INFO1 = {"Module":"Sonoff Basic","Version":"5.10.0b","FallbackTopic":"DVES_998D26","GroupTopic":"sonoffs"}
02:11:09 MQT: tele/sonoff/INFO2 = {"WebServerMode":"Admin","Hostname":"sonoff-3366","IPAddress":"192.168.1.67"}
02:11:09 MQT: tele/sonoff/INFO3 = {"Restar

SupraJames, 2018-01-06:
Is YOUR mqtt server really called domus1? That't the default name

closed 2018-02-05 by Igor122017